tsduck/tsduck #1046: sp[BUG]

joolzg, 2022-08-24:
#### Bug description:

#### How to reproduce:
sp -v -I hls --live https://d301yv2uyy17su.cloudfront.net/sky-news-live.m3u8 -P regulate -O file a.ts

#### Expected behavior:
I would expect a file with video and audio

#### Errors and logs:

#### Environment:
Ubuntu Linux 20.02
TSDuck 3.32-2769
Version from DEB.

joolzg, 2022-08-24:
Produce should be tsp not sp, for some reason I cannot edit the comment

lelegard, 2022-08-24:
Hi @joolzg 

Why do you use the `regulate`  plugin here? What is the technical purpose?

This plugin is used to regulate a TS which is processed as a higher speed than the video rendering speed (aka playout speed), when the rest of the processing or the output needs to be played at rendering speed. For instance, if you do the exact opposite as your command, reading from a file and sending to HLS, the input file is read at disk speed, much faster than the playout speed. If you directly broadcast it to UDP/IP or HLS or any real time rendering broadcast transmission, the stream will arrive too fast to readers. In that case, you need to regulate the processing speed, to slow it down to the playout speed.

Moreover, the `regulate` plugin alone works well only in the case of a CBR stream, when the bitrate is evaluated ahead. In the case of a VBR stream, the adaptation is always late and the stream is permanently jittered. In the case of a VBR stream, use `-P regulate --pcr-synchronous`.

In your case, the input is already regulated by nature. And the output does not need any regulation because you just save the content in a file. So, at best, with a pure CBR stream, the plugin `regulate` does nothing. At worst, on a VBR stream (as most HLS SPTS are), you break the speed, introduces delays, missing input segments.

I tried your command without `-P regulate` and it works perfectly.

Conclusion: always understand why a plugin should be used, for which reason with which parameters. Otherwise, don't use it or RTFM. Never use a plugin just because you saw it once in an another command which was adapted for a very different environment.

joolzg, 2022-08-24:
Sorry the regulate was from a previous test where i streamed the data over udp.

Looking at VLC and TSP, VLC pulls in 2 streams 04.hd and 09.hd. I guess this is the problem as 04 is the video and 09 is the audio and unless these are muxed together then we will only have video.

joolzg, 2022-08-24:
I haev just run the command without the regulate and the run ffprobe on the output.

Input #0, mpegts, from 'a.ts':
  Duration: 00:01:30.00, start: 7278.483289, bitrate: 933 kb/s
  Program 1 
  Stream #0:0[0x30]: Video: h264 (Main) ([27][0][0][0] / 0x001B), yuv420p(tv, bt709, progressive), 768x432 [SAR 1:1 DAR 16:9], 25 fps, 25 tbr, 90k tbn

lelegard, 2022-08-24:
Your HLS stream is dissociated. There are two distinct TS: one for audio (unique) and one for video (several alternatives with several resolutions). The media player (VLC or your browser for instance) is responsible for remuxing audio and video. The input plugin `hls` reads the main TS, containing video only. In this configuration, there is no audio/video TS, you need to choose or your need to remux.

This is easy to verify:
~~~
$ curl -sL 'https://d301yv2uyy17su.cloudfront.net/sky-news-live.m3u8'
#EXTM3U
#EXT-X-VERSION:4
#EXT-X-INDEPENDENT-SEGMENTS
#EXT-X-MEDIA:TYPE=AUDIO,GROUP-ID="audio2",NAME="ENG",DEFAULT=YES,AUTOSELECT=YES,LANGUAGE="ENG",URI="09_hd.m3u8"
#EXT-X-STREAM-INF:PROGRAM-ID=1,BANDWIDTH=1209781,CODECS="avc1.4D4020,mp4a.40.2",RESOLUTION=768x432,FRAME-RATE=25.000,AUDIO="audio2"
04_hd.m3u8
#EXT-X-STREAM-INF:PROGRAM-ID=1,BANDWIDTH=343244,CODECS="avc1.4D4020,mp4a.40.2",RESOLUTION=256x144,FRAME-RATE=25.000,AUDIO="audio2"
02_hd.m3u8
#EXT-X-STREAM-INF:PROGRAM-ID=1,BANDWIDTH=636585,CODECS="avc1.4D4020,mp4a.40.2",RESOLUTION=512x288,FRAME-RATE=25.000,AUDIO="audio2"
03_hd.m3u8
#EXT-X-STREAM-INF:PROGRAM-ID=1,BANDWIDTH=2322454,CODECS="avc1.4D4020,mp4a.40.2",RESOLUTION=1024x576,FRAME-RATE=25.000,AUDIO="audio2"
05_hd.m3u8
#EXT-X-STREAM-INF:PROGRAM-ID=1,BANDWIDTH=3614954,CODECS="avc1.640020,mp4a.40.2",RESOLUTION=1280x720,FRAME-RATE=25.000,AUDIO="audio2"
06_hd.m3u8
#EXT-X-STREAM-INF:PROGRAM-ID=1,BANDWIDTH=5671715,CODECS="avc1.640020,mp4a.40.2",RESOLUTION=1280x720,FRAME-RATE=25.000,AUDIO="audio2"
07_hd.m3u8
#EXT-X-STREAM-INF:PROGRAM-ID=1,BANDWIDTH=9043454,CODECS="avc1.640029,mp4a.40.2",RESOLUTION=1920x1080,FRAME-RATE=25.000,AUDIO="audio2"
08_hd.m3u8
~~~

This is the master playlist. You see that there is one specific subplaylist with ID="audio2" which contains only audio. Then, there are several subplaylists with a unique URI, various bandwidths and resolutions and the same reference to AUDIO="audio2" for the audio component.

To analyze the audio content, get the audio subplaylist and analyze one segment:
~~~
$ curl -sL 'https://d301yv2uyy17su.cloudfront.net/sky-news-live.m3u8' -D /dev/stdout | grep location:
location: http://linear116-gb-hls1-prd-ak.cdn.skycdp.com/Content/HLS_001_hd/Live/channel(skynewsinternational)/index_hd.m3u8
$ curl -sL 'http://linear116-gb-hls1-prd-ak.cdn.skycdp.com/Content/HLS_001_hd/Live/channel(skynewsinternational)/09_hd.m3u8'
#EXTM3U
#EXT-X-TARGETDURATION:12
#EXT-X-MEDIA-SEQUENCE:166134856
#EXT-X-VERSION:3
#EXT-X-KEY:METHOD=NONE
#EXT-X-PROGRAM-DATE-TIME:2022-08-24T13:42:41.507Z
#EXTINF:9.9840,
20220712T040939-09-166134856.ts
#EXT-X-KEY:METHOD=NONE
#EXTINF:10.0060,
20220712T040939-09-166134857.ts
#EXT-X-KEY:METHOD=NONE
#EXTINF:10.0050,
20220712T040939-09-166134858.ts
#EXT-X-KEY:METHOD=NONE
#EXTINF:10.0050,
20220712T040939-09-166134859.ts
#EXT-X-KEY:METHOD=NONE
#EXTINF:9.9840,
20220712T040939-09-166134860.ts
#EXT-X-KEY:METHOD=NONE
#EXTINF:10.0060,
20220712T040939-09-166134861.ts
$ curl -sL 'http://linear116-gb-hls1-prd-ak.cdn.skycdp.com/Content/HLS_001_hd/Live/channel(skynewsinternational)/20220712T040939-09-166134861.ts' | tsanalyze --service-analysis

===============================================================================
|  SERVICES ANALYSIS REPORT                                                   |
|=============================================================================|
|  Global PID's                                                               |
|  TS packets: 1, PID's: 1 (clear: 1, scrambled: 0)                           |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Global PID's ................................. C          147 b/s  |
|   Subt.  Global PSI/SI PID's (0x00-0x1F) .............. C          147 b/s  |
|  0x0000  PAT .......................................... C          147 b/s  |
|=============================================================================|
|  Service: 0x0001 (1), TS: 0x1111 (4369), Original Netw: 0x0000 (0)          |
|  Service name: (unknown), provider: (unknown)                               |
[... 47 lines cut ...]
|  TS packets: 1, PID's: 1 (clear: 1, scrambled: 0)                           |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Global PID's ................................. C          152 b/s  |
|   Subt.  Global PSI/SI PID's (0x00-0x1F) .............. C          152 b/s  |
|  0x0000  PAT .......................................... C          152 b/s  |
|=============================================================================|
|  Service: 0x0001 (1), TS: 0x1111 (4369), Original Netw: 0x0000 (0)          |
|  Service name: (unknown), provider: (unknown)                               |
|  Service type: 0x00 (Undefined)                                             |
|  TS packets: 53,782, PID's: 2 (clear: 2, scrambled: 0)                      |
|  PMT PID: 0x01E0 (480), PCR PID: 0x0030 (48)                                |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Undefined .................................... C    8,174,427 b/s  |
|  0x0030  AVC video (1920x1080, high profile, level 4.1  C    8,174,275 b/s  |
[... cut ...]
|          (C=Clear, S=Scrambled, +=Shared)                                   |
===============================================================================
~~~
Here, the service in the video TS contains video only.

Consequently, if you wanted to grab the TS containing video + audio, you can't, there is none. You need to separately grab the audio and video and then remux them. This is what media players, including browsers, do.

The `tsp` command processes one TS. It reads one input TS source. The `hls` plugin reads one TS source.

closed 2022-08-24 by joolzg