tsduck/tsduck #1048: [Q] Pipeline for dumping encrypted DVB-S2 channels

Galvo87, 2022-08-30:
Hi, I would like to dump some encrypted channels via satellite.
Does tsp offer plugins for dumping encrypted DVB-S2 channels as of today?
I have pci-e tuners and [this smartcard reader](https://satboy.com/product/smartmouse-phoenix-premium-357-6-8-10-12-mhz-with-dip-switch/) at my disposal.
Thanks

lelegard, 2022-08-30:
Hello @Galvo87 

Since you mention a smartcard reader, I assume you work on CAS-protected signals using a smartcard in the receiver. So, the short answer is "no" because, unless you work for a CAS vendor, that would form the basis of a content-sharing TV piracy system and this is [explicitly banned in the code of conduct of the project](https://github.com/tsduck/tsduck/blob/master/.github/CODE_OF_CONDUCT.md).

Technically speaking, there is a scrambling and a descrambling plugin in TSDuck but they are dedicated to system integration and debug. They either work on known encryption keys you provide or they emulate a fake basic CAS for MUX integration. They are lab tools for professionals.

When a real Conditional Access System (CAS) is used, each CAS has its own way of working, its own ECM and EMM formats, its own smartcards with its own proprietary protocols. All these proprietary information are industrial secrets. And even if you knew these protocols for a given CAS, implementing a dedicated TSDuck plugin for that probably violates some legislation on patent and content protection in most countries. You have been warned...

Galvo87, 2022-08-30:
I am not fond on CAS protected signals, my bad.
So I assume there is no safe and legal way to dump these kind of channels even owning a regular card.
Thanks

lelegard, 2022-08-30:
> So I assume there is no safe and legal way to dump these kind of channels even owning a regular card.

Not without an approved receiver or CI-CAM from the TV operator or its CAS vendor.

Galvo87, 2022-08-31:
Understood, thanks for the clarification.

closed 2022-08-31 by Galvo87