tsduck/tsduck #1051: tsswitch: log stream indexes when changing streams.

Canta, 2022-09-06:
Hi there.

When using `tsswitch`, the `-v` and `--timed-log` add status data to the process output.
One of those status data looks like this: 
`* 2022/09/06 10:12:56 - receive timeout, switching to next plugin`.

I would like to ask for a modification on this notification: to also log the current stream index, and the next stream index.

I've done some modifications to my local tsswitch build, so this log now looks like this:
`* 2022/09/06 16:31:08 - receive timeout, switching to next plugin. Actual: #0. Next: #1`
`* 2022/09/06 16:31:10 - receive timeout, switching to next plugin. Actual: #1. Next: #0`

Here are my experimental code changes for that:
```diff
diff --git a/src/libtsduck/plugins/private/tstsswitchCore.cpp b/src/libtsduck/plugins/private/tstsswitchCore.cpp
index 2766ab14..b7ad18f6 100644
--- a/src/libtsduck/plugins/private/tstsswitchCore.cpp
+++ b/src/libtsduck/plugins/private/tstsswitchCore.cpp
@@ -260,10 +260,15 @@ void ts::tsswitch::Core::setInputLocked(size_t index, bool abortCurrent)
 
 void ts::tsswitch::Core::handleWatchDogTimeout(WatchDog& watchdog)
 {
-    _log.verbose(u"receive timeout, switching to next plugin");
+    size_t next = (_curPlugin + 1) % _inputs.size();
+    _log.verbose(u"receive timeout, switching to next plugin. Actual: #" 
+      + UString::Format(u"%d", {_curPlugin})
+      + u". Next: #"
+      + UString::Format(u"%d", {next})
+    );
 
     GuardMutex lock(_mutex);
-    setInputLocked((_curPlugin + 1) % _inputs.size(), true);
+    setInputLocked(next, true);
 }
```

What do you think about this change?

Best regards.

lelegard, 2022-09-07:
HI @Canta 

> What do you think about this change?

I understand the idea. Two remarks:

First, all loggers use a format string, better use:
~~~
    _log.verbose(u"receive timeout, switching to next plugin. Actual: #%d. Next: #%d", {_curPlugin, next});
~~~

Second, fields such as `_curPlugin` should be used under the protection of the mutex (see above lock and [comments in the header](https://github.com/tsduck/tsduck/blob/905a0fad0391664864b6ef57829f7559e12047aa/src/libtsduck/plugins/private/tstsswitchCore.h#L200)). Even in a read-only manner, they are not volatile. But moving the lock before the message is not nice. By self-discipline, we restrict the usage of log messages in a mutex section to debug messages and internal/fatal errors.

I will look into it.

Canta, 2022-09-08:
Oh, thank you very much! :)

I have several of this little modifications in my agenda. I was originally planning to make pull requests, because I wouldn't want to add extra work to your own agenda, and this kind of changes are actually pretty small.
However, even when I mostly understand what to do, I'm not actually a C++ programmer, so I wasn't really sure about "just doing PRs and then we'll see": dealing with C++ novice code may be even more annoying for you than dealing with small requests.

I'll like to adapt to your way of working. So, let me ask you: would you prefer the creation of several issues from time to time asking for little modifications like this one, or would you rather prefer PRs to take a look at?

Thanks.

closed 2023-04-22 by lelegard