tsduck/tsduck #1052: tsswitch: strange switching order

Canta, 2022-09-09:
Hi there.

I'm curious about some situations where `tsswitch` doesn't seem to switch to the proper stream, or just does it kinda erratically.
This is related to #1051 :  I actually wanted that functionality in order to monitor and track this other issue.

I saw a few times, working with streams received from remote sources, that `tsswitch` may kept stuck in some kind of unintended loop on a single stream. But this was quite cumbersome to diagnose, as the log didn't tell the current and next stream when switching. However, today I built tsswitch again after #1051, and could see this behaviour locally, using test streams:

```
$ bin/release-x86_64-HOSTNAME/tsswitch \
-v --infinite --primary-input 0 --timed-log  \
-I fork 'ffmpeg -hide_banner -loglevel fatal -f lavfi -re -i "testsrc=320x240:r=1" -c:v h264 -preset ultrafast -t 60 -f mpegts -' \
-I fork 'ffmpeg -hide_banner -loglevel fatal -f lavfi -re -i "testsrc=320x240:r=1" -c:v h264 -preset ultrafast -t 120 -f mpegts -' \
-O ip 127.0.0.1:12345
* 2022/09/09 14:04:28 - receive timeout, switching to next plugin (#0 to #1)
* 2022/09/09 14:04:30 - receive timeout, switching to next plugin (#1 to #0)
* 2022/09/09 14:04:38 - receive timeout, switching to next plugin (#0 to #1)
* 2022/09/09 14:04:40 - receive timeout, switching to next plugin (#1 to #0)
* 2022/09/09 14:04:43 - receive timeout, switching to next plugin (#0 to #1)
* 2022/09/09 14:04:45 - receive timeout, switching to next plugin (#0 to #1)
* 2022/09/09 14:04:49 - receive timeout, switching to next plugin (#1 to #0)
* 2022/09/09 14:04:51 - receive timeout, switching to next plugin (#1 to #0)
* 2022/09/09 14:04:54 - receive timeout, switching to next plugin (#0 to #1)
* 2022/09/09 14:04:56 - receive timeout, switching to next plugin (#0 to #1)
* 2022/09/09 14:04:59 - receive timeout, switching to next plugin (#0 to #1)
* 2022/09/09 14:05:01 - receive timeout, switching to next plugin (#0 to #1)
* 2022/09/09 14:05:04 - receive timeout, switching to next plugin (#0 to #1)
* 2022/09/09 14:05:09 - receive timeout, switching to next plugin (#0 to #1)
* 2022/09/09 14:05:11 - receive timeout, switching to next plugin (#0 to #1)
* 2022/09/09 14:05:17 - receive timeout, switching to next plugin (#1 to #0)
* 2022/09/09 14:05:20 - receive timeout, switching to next plugin (#0 to #1)
* 2022/09/09 14:05:22 - receive timeout, switching to next plugin (#0 to #1)
* 2022/09/09 14:05:26 - receive timeout, switching to next plugin (#1 to #0)
* 2022/09/09 14:05:28 - receive timeout, switching to next plugin (#0 to #1)
* 2022/09/09 14:05:30 - receive timeout, switching to next plugin (#0 to #1)
* 2022/09/09 14:05:32 - receive timeout, switching to next plugin (#0 to #1)
* 2022/09/09 14:05:35 - receive timeout, switching to next plugin (#1 to #0)
* 2022/09/09 14:05:37 - receive timeout, switching to next plugin (#1 to #0)
* 2022/09/09 14:05:39 - receive timeout, switching to next plugin (#1 to #0)
* 2022/09/09 14:05:41 - receive timeout, switching to next plugin (#1 to #0)
* 2022/09/09 14:05:43 - receive timeout, switching to next plugin (#1 to #0)
* 2022/09/09 14:05:45 - receive timeout, switching to next plugin (#1 to #0)
* 2022/09/09 14:05:47 - receive timeout, switching to next plugin (#1 to #0)
* 2022/09/09 14:05:50 - receive timeout, switching to next plugin (#1 to #0)
^C
```

Note the following points:
- Sometimes, not always, it keeps "switching to next plugin" while never actually changing the current stream index.
- Sometimes, not always, the switching timing matches the default 2000 ms. But other times that's not the case. 
- The input streams are forks that I don't change. Yet, `tsswitch` behaviour is not predictable/constant/always-the-same.

This was obviously a test. But I saw this kind of strange behaviour working with real streams.

My real use case would be having 3 inputs:
1. Real stream, primary, using either `ip`, `srt`, or `fork` input plugins.
2. Real stream, secondary, using either `ip`, `srt`, or `fork` input plugins.
3. Locally generated black screen stream, most likely using `fork` input plugin, but could also be `ip`.

I did a lot of calibrations working with real streams on that situation (timeouts, priorities, packet sizes, bitrates, input filters, and so on), and eventually felt that `tsswitch`'s behaviour wasn't actually under my control: sometimes it was deciding to correctly change streams as I told it, but other times it kept in weird loops never reconnecting to the proper inputs (usually stated with `--primary-input`). Sometimes it switched to the black screen to never return to main streams (keeping the black screen forever), other times it didn't even connected to the black screen stream. And sometimes it worked great for days, without me changing anything.

Could this be indication of some bug? Perhaps I'm missunderstanding something?
I would appreciate your insights on this.

Thanks.

lelegard, 2022-09-10:
Hi @Canta,

This is not an easy one to troubleshoot indeed.

There were indeed two different problems, a false one and a real one.

The first problem is a false impression from successive verbose messages such as "switching to next plugin (# 0 to # 1)", with always the same switch, here "# 0 to # 1". You use option `--primary-input 0`, meaning that the primary plugin is never stopped and whenever a packet arrives from it, if the current plugin is another one, we immediately switch back to the primary plugin. But there is no verbose message to say that we switched back. So, the two successive switching on timeout "(# 0 to # 1)" were in fact separated by a switch back to the primary plugin # 0 because of a packet arriving from it. I added a verbose message for that and the suite of messages is now more consistent. The absence of verbose message at that point was due to the same reason I explained in #1051: I do not like the idea of logging a verbose message while the mutex is held. Anyway, since it creates trouble in the users mind, I added it. In case of performance issue, just omit the `--verbose` option.

The second problem is due to the fact that, sometimes, not always (why?), the ffmpeg process takes a long time to die when killed, for some reason I ignore. There are two consequences:

- This explains that the interval between two switches is sometimes longer than the 2 seconds timeout. A switch is complete only when the plugin you leave is fully stopped. In the case of the `fork` plugin, this means when the forked process is dead and the pipe is closed. Typically, because of timeouts, when you switch from # 1 to # 0, the forked process in # 1 is killed. If # 0 also times out after 2 seconds, we need to switch back to # 1 again. But you cannot restart # 1 as long as its previous invocation is not terminated.
- The longer than expected switch is correctly performed in the case of timeout. However, it was not fully correctly performed in the case of a switch back to the primary input. In that case, the watchdog was restarted too early, before completion of the previous termination. I fixed this. In fact, I suspect I did that on purpose to improve the reactiveness of the system when many input plugins were present. However, it did not anticipate the situation of only two plugins with a termination time potentially longer than the receive timeout.

I do not know exactly what this type of ffmpeg does (what is the `testsrc` input?) but it produces packets at a very very slow bitrate. The default reception timeout (2 seconds) is too small in that case. Also this ffmpeg command takes a long time to die (why?), creating other issues as described above.

As you have looked into the code of the `tsswitch` core as part of #1051, you have noticed the use of "actions", some kind of rudimentary VM. When I developed `tsswitch`, I realised that the number of possible scenarios was very high, depending on switching modes, primary input or not, etc. A simple state machine would not be sufficient since there was no simple state transition. Some kind of virtual code was necessary. However, this is not yet a miraculous strategy and problems remain such as the long-standing bug #324.

Canta, 2022-09-12:
> The first problem is a false impression from successive verbose messages such as "switching to next plugin (# 0 to # 1)", with always the same switch, here "# 0 to # 1". You use option [...]

I see. 
If that's the case, I believe `--fast-switch` should change that.
But I saw somewhere an open issue regarding `--fast-switch` and `--infinity`, so I decided to use only `--infinity`.

> The absence of verbose message at that point was due to the same reason I explained in https://github.com/tsduck/tsduck/issues/1051: I do not like the idea of logging a verbose message while the [...]

Very well.

> The second problem is due to the fact that, sometimes, not always (why?), the ffmpeg process takes a long time to die when killed, for some reason I ignore.

Not sure, but probably can be handled with `SIGKILL`. Will take a look at it.

> I do not know exactly what this type of ffmpeg does (what is the testsrc input?) but it produces packets at a very very slow bitrate. The default reception timeout (2 seconds) is too small in that case.

I figured that out (the bitrate/timeout situation), but let it fail anyway, so I could reproduce the issue I've seen with real streams. But I can explain that `ffmpeg` command.

- `-f lafvi` as input means "use the lavfi decoder". `lavfi` refers to `Lib AV Filter`, which is the component from the whole "libav" collection of libraries used for audio and video filtering. In the case of the `ffmpeg` command line, it's tipically used when you utilize the `-filter_complex` flag, as well as the `-vf` ("video filter") and `-af` ("audio filter") flags. But it also has the ability to generate streams using the filter library, by means of working as a "decoder" or "demuxer" (not entirely sure which one), which is the case in the example before. So, when you use `-f lavfi`, you can later pass it `-i "filter_syntax_here"`.
- `testsrc` is one of the many lavfi available filters. It creates a testing screen video, with some colours and a counter. The rest of the syntax after `testsrc` are filter parameters. One of them is the screen size.
- But the last parameter for `testsrc` is `r=1`. "r" is for "rate", and that means "1 fps". And that's the reason for the low bitrate. The streams were configured that way in order to avoid unwanted CPU usage during my tests. But if you change that to `r=30`, you'll have a "normal" 30fps stream, with obviously higher bitrate. The and same goes for the screen size: set it to `1920x1080` if you want to.
- You may also add some bitrate output parameters. I didn't needed them for this example, so I kept it simple.

> When I developed tsswitch, I realised that the number of possible scenarios was very high, depending on switching modes, primary input or not, etc. A simple state machine would not be sufficient [...]

Thank you very much for the insights.

I'm building right now the new code, and will deploy it to some server in order to test it with real streams. Please give me some days to see if it somehow fails (I've had a fail yesterday, with pre-1051 code, after days of working fine), so we can check out the new logs, trying to find if something's still going on or it is actually fixed. Is it ok to keep the ticket open until next Friday, so I can test it the whole week?

Best regards.

Canta, 2022-09-12:
> My feeling is that all problems you encounter with that specific configuration come from the fact that you use a reception timeout (2s, the default) which is incompatible with the input sources. The timeout should trigger when there is obviously an error. If a source emits a packet every 2 or 3 seconds, 2s is not a valid timeout. Keep in mind that most TS sources produce megabits per second.

You are, of course, correct about this. I also speculated the timeout and the rate may be related. However, I assure you, I see this behaviour on real streams, and so far I wasn't able to understand why, nor how to trigger it. So, when I saw a reproducible way of triggering the same strange behaviour (at least in the logs), I brought it here.

So, just to let it noted: when I set 1920x1080 and 30fps for the original `tsswitch` command, it doesn't show the loop, either over the same stream nor between streams. The timeout is key.

But let me show you what I see with real streams. Here's a log from yesterday, with pre-1051 code running:

```(...)
Sep 11 23:56:41 live1 tsswitch_realstream2[2455]: * 2022/09/11 23:56:41 - receive timeout, switching to next plugin
Sep 11 23:56:41 live1 tsswitch_realstream1[2455]: * 2022/09/11 23:56:41 - receive timeout, switching to next plugin
Sep 11 23:56:43 live1 tsswitch_realstream2[2455]: * 2022/09/11 23:56:43 - receive timeout, switching to next plugin
Sep 11 23:56:43 live1 tsswitch_realstream1[2455]: * 2022/09/11 23:56:43 - receive timeout, switching to next plugin
Sep 11 23:56:45 live1 tsswitch_realstream2[2455]: * 2022/09/11 23:56:45 - receive timeout, switching to next plugin
Sep 11 23:56:45 live1 tsswitch_realstream1[2455]: * 2022/09/11 23:56:45 - receive timeout, switching to next plugin
Sep 11 23:56:47 live1 tsswitch_realstream2[2455]: * 2022/09/11 23:56:47 - receive timeout, switching to next plugin
Sep 11 23:56:47 live1 tsswitch_realstream1[2455]: * 2022/09/11 23:56:47 - receive timeout, switching to next plugin
Sep 11 23:56:49 live1 tsswitch_realstream2[2455]: * 2022/09/11 23:56:49 - receive timeout, switching to next plugin
Sep 11 23:56:49 live1 tsswitch_realstream1[2455]: * 2022/09/11 23:56:49 - receive timeout, switching to next plugin
Sep 11 23:56:51 live1 tsswitch_realstream2[2455]: * 2022/09/11 23:56:51 - receive timeout, switching to next plugin
Sep 11 23:56:51 live1 tsswitch_realstream1[2455]: * 2022/09/11 23:56:51 - receive timeout, switching to next plugin
Sep 11 23:56:53 live1 tsswitch_realstream2[2455]: * 2022/09/11 23:56:53 - receive timeout, switching to next plugin
Sep 11 23:56:53 live1 tsswitch_realstream1[2455]: * 2022/09/11 23:56:53 - receive timeout, switching to next plugin
Sep 11 23:56:55 live1 tsswitch_realstream2[2455]: * 2022/09/11 23:56:55 - receive timeout, switching to next plugin
Sep 11 23:56:55 live1 tsswitch_realstream1[2455]: * 2022/09/11 23:56:55 - receive timeout, switching to next plugin
Sep 11 23:56:57 live1 tsswitch_realstream2[2455]: * 2022/09/11 23:56:57 - receive timeout, switching to next plugin
Sep 11 23:56:57 live1 tsswitch_realstream1[2455]: * 2022/09/11 23:56:57 - receive timeout, switching to next plugin
Sep 11 23:56:59 live1 tsswitch_realstream2[2455]: * 2022/09/11 23:56:59 - receive timeout, switching to next plugin
Sep 11 23:56:59 live1 tsswitch_realstream1[2455]: * 2022/09/11 23:56:59 - receive timeout, switching to next plugin
Sep 11 23:57:01 live1 tsswitch_realstream2[2455]: * 2022/09/11 23:57:01 - receive timeout, switching to next plugin
Sep 11 23:57:01 live1 tsswitch_realstream1[2455]: * 2022/09/11 23:57:01 - receive timeout, switching to next plugin
Sep 11 23:57:03 live1 tsswitch_realstream2[2455]: * 2022/09/11 23:57:03 - receive timeout, switching to next plugin
Sep 11 23:57:03 live1 tsswitch_realstream1[2455]: * 2022/09/11 23:57:03 - receive timeout, switching to next plugin
Sep 11 23:57:05 live1 tsswitch_realstream2[2455]: * 2022/09/11 23:57:05 - receive timeout, switching to next plugin
Sep 11 23:57:05 live1 tsswitch_realstream1[2455]: * 2022/09/11 23:57:05 - receive timeout, switching to next plugin
Sep 11 23:57:07 live1 tsswitch_realstream2[2455]: * 2022/09/11 23:57:07 - receive timeout, switching to next plugin
(...)
```

I run `tsswitch` as a service, outputing its logs to the common syslog. That's why you see extra logging there.
This is implementing the use case with 3 inputs, described in my original post. And I have multiple of those commands: one per signal. I want to achieve both redundancy between two alternate inputs AND a blackout fallback in case both inputs turn offline for some reason.

Here's the command line for that:
```
  tsswitch -v --timed-log --primary-input 1 --remote 127.0.0.1:$CONTROL_PORT --delayed-switch \
    -I fork "ffmpeg -hide_banner -nostats -y -loglevel error -i 'REDACTED_INPUT_1' -map 0:v -map 0:a -c copy -fflags '+flush_packets +genpts' -avioflags direct -movflags faststart -pcr_period 20 -max_muxing_queue_size 9999 -muxdelay 0 -muxpreload 0 -mpegts_service_id 100 -streamid 0:0x0001 -streamid 1:0x0002 -f mpegts -" \
    -I fork "ffmpeg -hide_banner -nostats -y -loglevel error -i 'REDACTED_INPUT_2' -map 0:v -map 0:a -c copy -fflags '+flush_packets +genpts' -avioflags direct -movflags faststart -pcr_period 20 -max_muxing_queue_size 9999 -muxdelay 0 -muxpreload 0 -mpegts_service_id 100 -streamid 0:0x0001 -streamid 1:0x0002 -f mpegts -" \
[... cut ...] ays ago: just yesterday kept stuck, and not "every time there's some error".

So... it's hard to diagnose. I had hope the "low bitrate" example would lead to an easier to debug situation.

I believe the last logs indicate something like `timeout #1, then recovered before starting #2, then timeout #1 again` in a loop. And this time, we're talking about a real 1080p ~5Mbps stream, sent remotely by a third party.

I'll keep it running for days, and let you know if something new happens.

Thank you.

lelegard, 2022-09-13:
All right. Please submit a new log with all recent traces, including the recovery on primary input, when you get the issue again. The traces are the symptoms of tsp's behaviour. In addition, it would be valuable to explain what happened on the input (which one failed and how), if possible, to correlate the reality with the observed symptoms.

Do not use `--delayed-switch` with `--receive-timeout` because of #324.

Canta, 2022-10-28:
Hi there lelegard.

I finally have some fresh logs, but I'm not able to analyze them right now in detail, so I'll just let it all noted here. I will try to do some deeper analysis next week.

It's a setup with 4 different jobs, all of them set up like the `tsswitch` command stated here: https://github.com/tsduck/tsduck/issues/1052#issuecomment-1244019975

I've traced the server logs to get the first errors reported, and found this:
```
(...)
Oct 28 12:57:21 live1 tsswitch_realstream4[2455]: REDACTED_INPUT_2: Input/output error
Oct 28 12:57:23 live1 tsswitch_realstream3[2455]: REDACTED_INPUT_2: Input/output error
Oct 28 12:57:23 live1 tsswitch_realstream4[2455]: * 2022/10/28 12:57:23 - receive timeout, switching to next plugin (#1 to #2)
Oct 28 12:57:23 live1 tsswitch_realstream2[2455]: REDACTED_INPUT_2: Input/output error
Oct 28 12:57:25 live1 tsswitch_realstream3[2455]: * 2022/10/28 12:57:25 - receive timeout, switching to next plugin (#1 to #2)
Oct 28 12:57:25 live1 tsswitch_realstream4[2455]: * 2022/10/28 12:57:25 - receive timeout, switching to next plugin (#1 to #2)
Oct 28 12:57:25 live1 tsswitch_realstream2[2455]: * 2022/10/28 12:57:25 - receive timeout, switching to next plugin (#1 to #2)
Oct 28 12:57:26 live1 tsswitch_realstream1[2455]: * 2022/10/28 12:57:26 - receive timeout, switching to next plugin (#1 to #2)
Oct 28 12:57:26 live1 tsswitch_realstream1[2455]: REDACTED_INPUT_2: Input/output error
Oct 28 12:57:27 live1 tsswitch_realstream3[2455]: * 2022/10/28 12:57:27 - receive timeout, switching to next plugin (#1 to #2)
(...)
```

And after that, all tsswitch instances (different jobs, with different input streams) stayed in an infinite loop status, reporting this:
```
(...)
Oct 28 13:16:28 live1 tsswitch_realstream1[2455]: * 2022/10/28 13:16:28 - receive timeout, switching to next plugin (#1 to #2)
Oct 28 13:16:29 live1 tsswitch_realstream3[2455]: * 2022/10/28 13:16:29 - receive timeout, switching to next plugin (#1 to #2)
Oct 28 13:16:29 live1 tsswitch_realstream4[2455]: * 2022/10/28 13:16:29 - receive timeout, switching to next plugin (#1 to #2)
Oct 28 13:16:29 live1 tsswitch_realstream2[2455]: * 2022/10/28 13:16:29 - receive timeout, switching to next plugin (#1 to #2)
Oct 28 13:16:30 live1 tsswitch_realstream1[2455]: * 2022/10/28 13:16:30 - receive timeout, switching to next plugin (#1 to #2)
Oct 28 13:16:31 live1 tsswitch_realstream3[2455]: * 2022/10/28 13:16:31 - receive timeout, switching to next plugin (#1 to #2)
Oct 28 13:16:31 live1 tsswitch_realstream4[2455]: * 2022/10/28 13:16:31 - receive timeout, switching to next plugin (#1 to #2)
Oct 28 13:16:31 live1 tsswitch_realstream2[2455]: * 2022/10/28 13:16:31 - receive timeout, switching to next plugin (#1 to #2)
(...)
```

After restarting the tsswitch instances, the streams recovered without further issue, which points to some tsswitch internal problem (instead of, for example, some input stream problem).
And looking at the beginning of the logs, it seems it may have been triggered by some signal or even network glitch.

Sorry it's just a quick note: I know there's not much to do with just this. But I'm afraid it's also hard to catch in real time.
I'm also sorry about the redacted input stuff, but I don't feel confortable sharing publicy details about streams from third parties. But I can tell you it's an rtmp stream with h264 video and aac audio, which ffmpeg turns into a MPEG-TS stream. `INPUT_URL: Input/output error` is a typical ffmpeg error report, and I can't tell if it was something related to the input stream (not rare), or something involving the way tsswitch handles pipes and fork restarts.
Will take another look at this next week.

Thanks.