tsduck/tsduck #1054: [Q] TSDuck  to receive udp, encapsulate TS and then send udp again

FJTor, 2022-09-12:
Hello good afternoon,

I have a GNURadio project that uses DVB-S2 transmitter and needs as input a TS format.

My plan is to use this configuration:

Camera (MP4/H.264 maybe?) ----> UDP Socket sending packets ------> something that encapsulates as TS format ------> another UDP Socket -----> GNURadio transmitter

My question is:

¿Can TSDuck receive MP4 from a camera via UDP, adapt from MP4/H264/etc. to TS, and send it via another UDP socket?

Thanks and regards
F Javier

Canta, 2022-09-12:
> ¿Can TSDuck receive MP4 from a camera via UDP, adapt from MP4/H264/etc. to TS, and send it via another UDP socket?

I believe TSDuck doesn't handle MP4, even when it contains H.264. The point being: TSDuck is for working with TS. So, in your diagram...

> Camera (MP4/H.264 maybe?) ----> UDP Socket sending packets ------> something that encapsulates as TS format ------> another UDP Socket -----> GNURadio transmitter

...` something that encapsulates as TS format` should more likely be another tool: like `ffmpeg` or `gstreamer` for example.
This is how to do it with `ffmpeg`:

```
ffmpeg -i "udp://UDP_CAMERA_IP:PORT" -c copy -f mpegts "udp://OUTPUT_IP:PORT"
```

However, using TSDuck's `fork` input plugin, you can actually make TSDuck's tools (`tsp`, `tsswitch`, and so on) to be able to handle the MP4 stream, by just delegating the translation to MPEG-TS to other forked tool. Like this:

```
tsp -v --timed-log \
-I fork 'ffmpeg -hide_banner -loglevel fatal -nostats -i "udp://UDP_CAMERA_IP:PORT" -c copy -f mpegts -' \
-O ip OUTPUT_IP:PORT
```

That could be useful for several reasons, but I'm not sure if you actually need it for your use case: it looks like your case is fine for using just ffmpeg or any other, without a strong need for TSDuck.

FJTor, 2022-09-13:
Thank you very much, Daniel:

just one more issue... ffmpeg and gstreamer work more or less "Real Time" on a normal PC? I mean... would it be able to process it in a way to have video conference, for instance? (supose my gnuradio app works RT)

Thank you again, I am newbie on this, sorry....

FJTor, 2022-09-13:
Have to add... camera is not much bitrate, it will be less than 600kbps

Canta, 2022-09-13:
> just one more issue... ffmpeg and gstreamer work more or less "Real Time" on a normal PC? I mean... would it be able to process it in a way to have video conference, for instance? (supose my gnuradio app works RT)
> (...)
> Have to add... camera is not much bitrate, it will be less than 600kbps

Yes, they will work in "real time" in a "normal pc".

However, note you're entering a rabbit hole of possible configurations, contextual problems, and lots of stuff to learn: every single tool you'll pick for this tasks will be a whole new world to deal with. So take it easy and don't try to rush it, because it may be complex and slow to gasp. Just keep slowly learning and trying the available tools, and sooner than later you'll get there. I'll recommend you starting with `ffmpeg`, as it has lots of people using it and tips everywhere.

FJTor, 2022-09-14:
thank you so much Daniel. Marking as solved

closed 2022-09-14 by FJTor