tsduck/tsduck #1056: [BUG]

jcekstrom, 2022-09-16:
#### Bug description:
The **spliceinject**  does not fully initialize, i.e. read information from the provided files, before processing packets.

#### How to reproduce:
Using any ts file, create  file with a splice_insert message near the beginning of the ts file.
Run something similar to:
`tsp --debug --verbose --add-input-stuffing 1/2 -I file test.ts -P pmt --service 0001 --add-programinfo-id 0x43554549 --add-pid 600/0x86 -P spliceinject -f scte35.xml --service 0001 --inject-count 2  -P filter -n -p 0x1FFF -O file test_out.ts`

The insert fails to happen.  If instead we run 

#### Expected behavior:
SCTE 35 Messages to be inserted into the output file.

#### Errors and logs:

The first log is the error case... the 2nd is a work around that allows the `spliceinject` processor to initialize before passing data through.

~~~
$ tsp --debug --verbose --add-input-stuffing 1/2 -I file test.ts -P pmt --service 0001 --add-programinfo-id 0x43554549 --add-pid 600/0x86 -P spliceinject -f scte35.xml --service 0001 --inject-count 2  -P filter -n -p 0x1FFF -O file test_out.ts
tsp: Debug: debug level set to 1
tsp: Debug: ====> tsp --debug --verbose --add-input-stuffing 1/2
* Debug: debug level set to 1
* Debug: file: debug level set to 1
* Debug: file: ====> tsp -I file test.ts
* Debug: file: debug level set to 1
* Debug: file: ====> tsp -O file test_x.ts
* Debug: trying to load "/usr/local/Cellar/tsduck/3.31-2761_1/bin/tsplugin_pmt.dylib"
* Debug: dlopen(/usr/local/Cellar/tsduck/3.31-2761_1/bin/tsplugin_pmt.dylib, 0x000A): tried: '/usr/local/Cellar/tsduck/3.31-2761_1/bin/tsplugin_pmt.dylib' (no such file)
* Debug: trying to load "/usr/local/Cellar/tsduck/3.31-2761_1/bin/pmt.dylib"
* Debug: dlopen(/usr/local/Cellar/tsduck/3.31-2761_1/bin/pmt.dylib, 0x000A): tried: '/usr/local/Cellar/tsduck/3.31-2761_1/bin/pmt.dylib' (no such file)
* Debug: trying to load "/usr/local/Cellar/tsduck/3.31-2761_1/lib/tsduck/tsplugin_pmt.dylib"
* Debug: registering processor plugin "pmt", status: ok
* Debug: pmt: debug level set to 1
* Debug: pmt: ====> tsp -P pmt --service 0001 --add-programinfo-id 0x43554549 --add-pid 600/0x86
* Debug: trying to load "/usr/local/Cellar/tsduck/3.31-2761_1/bin/tsplugin_spliceinject.dylib"
* Debug: dlopen(/usr/local/Cellar/tsduck/3.31-2761_1/bin/tsplugin_spliceinject.dylib, 0x000A): tried: '/usr/local/Cellar/tsduck/3.31-2761_1/bin/tsplugin_spliceinject.dylib' (no such file)
* Debug: trying to load "/usr/local/Cellar/tsduck/3.31-2761_1/bin/spliceinject.dylib"
* Debug: dlopen(/usr/local/Cellar/tsduck/3.31-2761_1/bin/spliceinject.dylib, 0x000A): tried: '/usr/local/Cellar/tsduck/3.31-2761_1/bin/spliceinject.dylib' (no such file)
* Debug: trying to load "/usr/local/Cellar/tsduck/3.31-2761_1/lib/tsduck/tsplugin_spliceinject.dylib"
* Debug: registering processor plugin "spliceinject", status: ok
* Debug: loading names file /usr/local/Cellar/tsduck/3.31-2761_1/share/tsduck/tsduck.names
* Debug: spliceinject: adding standards SCTE to none
* Debug: spliceinject: debug level set to 1
* Debug: spliceinject: ====> tsp -P spliceinject -f /Users/jekstr328/scte35.xml --service 0001 --inject-count 2
* Debug: trying to load "/usr/local/Cellar/tsduck/3.31-2761_1/bin/tsplugin_filter.dylib"
* Debug: dlopen(/usr/local/Cellar/tsduck/3.31-2761_1/bin/tsplugin_filter.dylib, 0x000A): tried: '/usr/local/Cellar/tsduck/3.31-2761_1/bin/tsplugin_filter.dylib' (no such file)
* Debug: trying to load "/usr/local/Cellar/tsduck/3.31-2761_1/bin/filter.dylib"
* Debug: dlopen(/usr/local/Cellar/tsduck/3.31-2761_1/bin/filter.dylib, 0x000A): tried: '/usr/local/Cellar/tsduck/3.31-2761_1/bin/filter.dylib' (no such file)
* Debug: trying to load "/usr/local/Cellar/tsduck/3.31-2761_1/lib/tsduck/tsplugin_filter.dylib"
* Debug: registering processor plugin "filter", status: ok
* Debug: filter: debug level set to 1
* Debug: filter: ====> tsp -P filter -n -p 0x1FFF
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* Debug: spliceinject: file server thread started
* Debug: spliceinject: Starting PollFiles on /Users/jekstr328/scte35.xml, poll interval = 500 ms, min stable delay = 500 ms
* Debug: file: detected TS file format TS
* Debug: file: initial buffer load: 13,734 packets, 2,581,992 bytes
* file: initial input bitrate is 2,057,756 b/s
* Debug: file: opening output file test_x.ts
* Debug: file: input thread started
* Debug: file: stopping the input plugin
* Debug: file: input thread terminated after 13,734 packets
* Debug: pmt: packet processing thread started
* Debug: file: output thread started
* Debug: filter: packet processing thread started
* Debug: control commands: debug level set to 1
* Debug: pmt: adding standards DVB to none
* Debug: spliceinject: packet processing thread started
* Debug: control commands: debug level set to 1
* Debug: pmt: adding standards MPEG to DVB
[... 120 lines cut ...]
* Debug: pmt: packet processing thread terminated after 13,734 packets, 13,733 passed, 1 dropped, 0 nullified
* Debug: pmt: stopping the plugin
* spliceinject: injecting SpliceInsert out @0x002959482, start: 0x00292D562, end: 0x002959482, 2 times, current PTS: 0x00292F458
* spliceinject: requeueing SpliceInsert out @0x002959482, start: 0x00293EEA2, end: 0x002959482
* spliceinject: injecting SpliceInsert in @0x0029594DC, start: 0x00292D5BC, end: 0x0029594DC, 2 times, current PTS: 0x00292F458
* spliceinject: requeueing SpliceInsert in @0x0029594DC, start: 0x00293EEFC, end: 0x0029594DC
* spliceinject: injecting SpliceInsert out @0x002959482, start: 0x00293EEA2, end: 0x002959482, current PTS: 0x00294198C
* spliceinject: injecting SpliceInsert in @0x0029594DC, start: 0x00293EEFC, end: 0x0029594DC, current PTS: 0x00294198C
* Debug: spliceinject: packet processing thread terminated after 13,733 packets, 13,733 passed, 0 dropped, 0 nullified
* Debug: spliceinject: stopping the plugin
* Debug: filter: packet processing thread terminated after 13,733 packets, 9,159 passed, 4,574 dropped, 0 nullified
* Debug: filter: stopping the plugin
* Debug: filter: 9,159 / 13,733 filtered packets
* Debug: spliceinject: file server thread completed
* Debug: file: stopping the output plugin
* Debug: file: output thread terminated after 13,734 packets (9,159 output) 
* Debug: Report logging thread terminated
~~~
#### Environment:
 - OS: macOS
 - OS version: latest 
 - TSDuck full version: 3.31-2761
 - Installation type: brew install

#### Additional information:

closed 2022-09-16 by jcekstrom

lelegard, 2022-09-16:
Hi @jcekstrom. It seems that we cross-posted at the same time. Glad that you found the solution by yourself.

jcekstrom, 2022-09-16:
Thanks @lelegard for the prompt response.  You have done amazing work on this project.