tsduck/tsduck #1059: [Q] Please share a simple way of inserting  SCTE-35 ad markers to a  VOD content having multiple mpeg segments file.

ashiskumarsahu, 2022-09-26:
I am completely new to this library and working on a requirement of SCTE markers on my VOD dash and hls contents. I have both DRM and non-drm mpd/hls contents.

I want to add scte-35 markers to all segments in single command. It would be great if you can share an example command by explain each part. Please share an end-to-end flow that will help new learner like me.

As per my understanding using **spliceinject** plugin splice points are created in segments and later these points are identified and replaced with actual advertisement (may be from a VAST tag) with a live transcoder like aws media tailor. 

**My files and folder structure**
master.m3u8
      1080 : index.m3u8  seg1.ts  seg3.ts  seg5.ts  seg7.ts  seg9.ts seg0.ts  seg2.ts  seg4.ts  seg6.ts  seg8.ts
      720   : index.m3u8  seg1.ts  seg3.ts  seg5.ts  seg7.ts  seg9.ts seg0.ts  seg2.ts  seg4.ts  seg6.ts  seg8.ts
      480   : index.m3u8  seg1.ts  seg3.ts  seg5.ts  seg7.ts  seg9.ts seg0.ts  seg2.ts  seg4.ts  seg6.ts  seg8.ts
      320   : index.m3u8  seg1.ts  seg3.ts  seg5.ts  seg7.ts  seg9.ts seg0.ts  seg2.ts  seg4.ts  seg6.ts  seg8.ts

stream.mpd
      audio
            en
                  mp4a.40.2 : init.mp4    seg-1.m4s  seg-2.m4s  seg-3.m4s  seg-4.m4s  seg-5.m4s seg-6.m4s seg-7.m4s seg-8.m4s
      video
            avc1
                  1 : init.mp4    seg-1.m4s  seg-2.m4s  seg-3.m4s  seg-4.m4s  seg-5.m4s seg-6.m4s seg-7.m4s seg-8.m4s
                  2 : init.mp4    seg-1.m4s  seg-2.m4s  seg-3.m4s  seg-4.m4s  seg-5.m4s seg-6.m4s seg-7.m4s seg-8.m4s
                  3 : init.mp4    seg-1.m4s  seg-2.m4s  seg-3.m4s  seg-4.m4s  seg-5.m4s seg-6.m4s seg-7.m4s seg-8.m4s
                  4 : init.mp4    seg-1.m4s  seg-2.m4s  seg-3.m4s  seg-4.m4s  seg-5.m4s seg-6.m4s seg-7.m4s seg-8.m4s

**I have few more questions:**

1. How to calculate time pts dynamically? Is there any plugin available in tsduck?
2. Will it work for multiple resolutions content as in above?
3. Will it work for DRM encrypted content?

Thank You in Advance!

Appreciate your quick response on this.

lelegard, 2022-09-26:
Hello,

You first need to understand which type of SCTE-35 markers you need. For pure TS files and broadcast streams, this is done using splice information tables in the stream. But with HLS it can also be special textual entries in the playlist files (and there are several incompatible forms of markers in playlist files). I am not familiar with DASH but there is maybe some specific information in the playlist files too.

TSDuck can only insert embedded splice information tables. If this is what you need, fine. If you need HLS or DASH markers in playlist files, you need some other tools. TSDuck is a transport stream toolkit for which HLS is just an input / output plugin. TSDuck is not an HLS or DASH specialised tool.

If you want splice information tables in the TS, you need to read and understand:
- The SCTE-35 standard, including the PTS processing.
- The TSDuck user's guide
   - The reference section of the `spliceinject` plugin to understand all possibilities.
   - The example section "5.2.24 Injecting SCTE 35 cue information" and especially "5.2.24.2 Cue insertion in offline files".

The PTS values cannot be automatically calculated. Only the user knows when the event will occur.

TSDuck is free, so you must invest some of your time to use it with full features...

ashiskumarsahu, 2022-09-27:
Hi @lelegard ,
Thanks for the reply. This is so informative.
I am new to this and reading tsduck document gradually. 

Actually I have pre-packaged content as I have shared in folder structure and I want to create splice points. My plan is to use AWS media tailor to replace these points with real-time ads. I have only VOD content.

If tsduck is not helpful in my case, could you please suggest any tools that can insert splice information in HLS and DASH playlist? I am struggling hard to find for a solution from last couple of days. Any help is highly appreciated.

Thank You