tsduck/tsduck #1060: [Q] Is there a way to remove an elementary stream, using the PID value, in a transport stream?

alvarollorente, 2022-09-29:
The idea is, in a transport stream, for example with three different audios, to delete one of them knowing the PID value. And keeping the rest of the values and PIDs the same. 

Is there any command that allows to do this? Maybe with a tsp processor?

Thanks!

cbernander, 2022-09-29:
Have a look at the `filter` plugin for tsp.

lelegard, 2022-09-29:
And also the `pmt` plugin to remove the references of the deleted audio PID.

alvarollgo, 2022-09-30:
Thank you very much for your quick responses @cbernander @lelegard 

I think I solve it with your suggestions, although I don't know if this is the most effective way. I use the `filter` plugin with the `--negate` and `--pid` options to remove the audio packets, and the `pmt` plugin with the `--remove-pid` option to remove the signalling.