tsduck/tsduck #1067: [Q] How do you set LNB power mode for Transport stream capture

xfiles2007, 2022-10-08:
Hi,

I am currently reading about using tsduck to capture a transport stream.

I understand I will have to use tsp command, but I can't see examples on how to set the LNB power to ON or 13v,18v.

The usb tuner I have is : TSB5520SE and the device appears on the system (Linux)

lelegard, 2022-10-08:
Hi @xfiles2007 

You do not need to think in terms of LNB voltage. The user only wants to receive a satellite stream, knowing the signal characteristics, frequency, polarity, modulation, etc. The `dvb` input plugin will take care of the rest.

Specifically, if the signal polarity is vertical, the plugin will set the LNB power to 13V. And if the polarity is horizontal, the LNB power will be set to 18V. You can see the [corresponding Linux-specific code here](https://github.com/tsduck/tsduck/blob/6fa12622cad81aac02506c7444f7ac2721e7e90b/src/libtsduck/dtv/linux/tsTunerDevice.cpp#L1051).

So, if you want a satellite stream as input, just use the `dvb` input plugin, specify all the characteristics of the signal (RTFM for more details) and don't care about LNB power.

xfiles2007, 2022-10-08:
Ok great - good to know!

Thanks :)

closed 2022-10-14 by xfiles2007