tsduck/tsduck #1077: [FEAT] New documentation example TV-->Radio

lars18th, 2022-10-16:
#### Is your feature request related to a problem?
I suggest to add a new example in the documentation: How to transform a TV service into a RADIO service.

#### Proposed solution:
Here an example of the processing chain (the service ID is 300 and the PCR video pid is 401):
```
tsp \
 -I ... \
 -P zap 300 \
 -P filter -p 401 --set-label 1 \
 -P filter --only-label 1 --pcr \
 -P craft --only-label 1 --no-payload --clear-pusi \
 -P continuity -f -p 401 \
 -P pmt -r 401 \
 -P sdt -s 300 -t 2 \
 -P rmorphan \
 -O ...
```
I feel all commands are self explanatory.

#### Alternatives:
Perhaps you found another more simple or compact chain, but this works.

#### Additional information:
I'm using this for some Digital Radio projects.

lelegard, 2022-10-16:
Hi @lars18th 

I would suggest to create a page in the [TSDuck Wiki](https://github.com/tsduck/tsduck/wiki). It was started years ago upon requests from a user who published a few examples in various pages. Since then, not much was added. However, this is the preferred place where users can publish educational example usages.

lars18th, 2022-10-16:
Ok. I'll do it!