tsduck/tsduck #1078: [Q] Best option to reorder audio stream

lars18th, 2022-10-16:
Hi,

I want to change the order of the audio streams for a digital television service. I need to set the default version, and change the order of others. What's the most simple action to achieve this? Changing the PMT table manually?

Thank you.

lelegard, 2022-10-16:
Hi @lars18th 

To my knowledge, there is no standard way of defining the "order of the audio streams" or even the "default version". This is either receiver-specific (hard-coded or through some settings menu) or defined in the private signalisation of an operator.

A receiver may decide to display the list of audio versions in whatever order, hard-coded, in the order of appearance in the PMT, sorted in alphabetical order, according to some private descriptor in the PMT, BAT or NIT, etc.

You need to understand how the languages are sorted in _your_ environment (operator and receiver devices). Then, you will be able to implement the change. Or not. Only if there is a solution at network-level, and not only at receiver level.

lars18th, 2022-10-16:
Hi @lelegard ,

Thank you for the clarification. However, any idea to _reorder_ the audio tracks in the PMT? I feel that the most generic approach is the appearance order, plus the prefered language and country descriptors.

lelegard, 2022-10-16:
Currently, the PMT is implemented as a C++ map of PID -> stream. A map is, by definition, implemented in order of keys, here PID's. When the PMT is serialised (converted to binary), the streams are consequently written in order of PID.  In the plugin `pmt`, the modifications are applied on the PMT C++ object, where the streams cannot be reordered.

So, reordering the streams in the binary PMT would need another approach. I will think about it.

lelegard, 2022-10-16:
OK, I added sorting hints inside the PMT C++ class. Currently, these sorting hints are only set when reading a binary PMT or an XML PMT.

The net result, for now, is that a generated binary PMT now restores the same order of PID's as input, no longer the order of PID values.

More generally, this update gives the possibility to affect the order of PID's in the PMT. Previously, this was not possible.

When I have time, I will add options to the plugin `pmt` such as `--sort-pid 342,43,0x34` and `--sort-languages end,fre,deu`. The latter applies to audio and subtitles.

lars18th, 2022-10-17:
Thank you @lelegard for consider this enhancement! 👍

lelegard, 2022-10-19:
Done, options `--sort-pids` and `--sort-languages` were added.

lars18th, 2022-11-09:
Thank you. Well done. I close this issue.

closed 2022-11-09 by lars18th