tsduck/tsduck #1079: [BUG] Problem with EIT p/f generation

przemo1983, 2022-10-17:
#### Bug description:
There is a problem when using tspacketize tool when using following command:
`tspacketize ./eit_pf.xml -p 18 --eit-normalization -o ./eit.ts`

#### How to reproduce:
Please use an XML as follows:
```xml
<?xml version='1.0' encoding='UTF-8'?>
<tsduck>
	<EIT type="pf" version="8" actual="true" current="true" service_id="1039" transport_stream_id="13" original_network_id="70">
		<event event_id="43119" start_time="2022-10-14 13:00:00" duration="02:00:00" running_status="running" CA_mode="false">
			<short_event_descriptor language_code="nor">
				<event_name>NRK nyheter</event_name>
			</short_event_descriptor>
		</event>
		<event event_id="43120" start_time="2022-10-14 15:00:00" duration="00:30:00" running_status="not-running" CA_mode="false">
			<short_event_descriptor language_code="nor">
				<event_name>NRK nyheter</event_name>
			</short_event_descriptor>
		</event>
	</EIT>
	<EIT type="pf" version="8" actual="true" current="true" service_id="1040" transport_stream_id="13" original_network_id="70">
		<event event_id="54818" start_time="2022-10-14 13:15:00" duration="00:50:00" running_status="running" CA_mode="false">
			<short_event_descriptor language_code="nor">
				<event_name>Drømmeslottet (R)</event_name>
			</short_event_descriptor>
		</event>
		<event event_id="54056" start_time="2022-10-14 14:05:00" duration="00:55:00" running_status="not-running" CA_mode="false">
			<short_event_descriptor language_code="nor">
				<event_name>To liv (R)</event_name>
			</short_event_descriptor>
		</event>
	</EIT>
	<EIT type="pf" version="8" actual="true" current="true" service_id="1041" transport_stream_id="13" original_network_id="70">
		<event event_id="53918" start_time="2022-10-14 13:55:00" duration="00:20:00" running_status="running" CA_mode="false">
			<short_event_descriptor language_code="nor">
				<event_name>Dyrevenn (R)</event_name>
			</short_event_descriptor>
		</event>
		<event event_id="53919" start_time="2022-10-14 14:15:00" duration="00:05:00" running_status="not-running" CA_mode="false">
			<short_event_descriptor language_code="nor">
				<event_name>Bernie</event_name>
			</short_event_descriptor>
		</event>
	</EIT>
</tsduck>
```

#### Expected behavior:
Only EIT pf are generated

#### Errors and logs:
EIT pf and EIT schedule are generated.
```
tspacketize: Debug[3]: debug level set to 3
tspacketize: Debug: ====> tspacketize ./eit_pf.xml -p 18 --eit-normalization -o ./eit.ts -v --debug=3
tspacketize: Debug: loading XML file ./eit_pf.xml
tspacketize: Debug: loading XML file /media/ts_duck/tsduck-3.31-2761/bin/release-x86_64-tnrlocalsetup/tsduck.tables.model.xml
tspacketize: Debug: adding standards DVB to none
tspacketize: Debug: setting EIT generator TS id to 0x000D (13)
tspacketize: Debug: creating EIT segment starting at 2022/10/14 12:00:00.000 for service: 0x040F (1039), TS: 0x000D (13), network: 0x0046 (70)
tspacketize: Debug[2]: loaded event id 0xA86F (43119), service: 0x040F (1039), TS: 0x000D (13), network: 0x0046 (70), starting 2022/10/14 13:00:00.000
tspacketize: Debug: creating EIT segment starting at 2022/10/14 15:00:00.000 for service: 0x040F (1039), TS: 0x000D (13), network: 0x0046 (70)
tspacketize: Debug[2]: loaded event id 0xA870 (43120), service: 0x040F (1039), TS: 0x000D (13), network: 0x0046 (70), starting 2022/10/14 15:00:00.000
tspacketize: Debug: creating EIT segment starting at 2022/10/14 12:00:00.000 for service: 0x0410 (1040), TS: 0x000D (13), network: 0x0046 (70)
tspacketize: Debug[2]: loaded event id 0xD622 (54818), service: 0x0410 (1040), TS: 0x000D (13), network: 0x0046 (70), starting 2022/10/14 13:15:00.000
tspacketize: Debug[2]: loaded event id 0xD328 (54056), service: 0x0410 (1040), TS: 0x000D (13), network: 0x0046 (70), starting 2022/10/14 14:05:00.000
tspacketize: Debug: creating EIT segment starting at 2022/10/14 12:00:00.000 for service: 0x0411 (1041), TS: 0x000D (13), network: 0x0046 (70)
tspacketize: Debug[2]: loaded event id 0xD29E (53918), service: 0x0411 (1041), TS: 0x000D (13), network: 0x0046 (70), starting 2022/10/14 13:55:00.000
tspacketize: Debug[2]: loaded event id 0xD29F (53919), service: 0x0411 (1041), TS: 0x000D (13), network: 0x0046 (70), starting 2022/10/14 14:15:00.000
[... 250 lines cut ...]
      Last provided at cycle: 1
      Last provided at packet: 4
      Due packet: 0
    - EIT schedule Actual
      Repetition rate: 0 ms
      Last provided at cycle: 1
      Last provided at packet: 4
      Due packet: 0
    - EIT schedule Actual
      Repetition rate: 0 ms
      Last provided at cycle: 1
      Last provided at packet: 4
      Due packet: 0
    - EIT schedule Actual
      Repetition rate: 0 ms
      Last provided at cycle: 1
      Last provided at packet: 4
      Due packet: 0
```

#### Environment:
 - OS: Ubuntu
 - OS version: Ubuntu 18.04
 - TSDuck full version: tsduck-3.31-2761
 - Installation type: build from source code

lelegard, 2022-10-17:
> Expected behavior:
> Only EIT pf should be generated

Why? The documentation doesn't say so. To be honest, it doesn't say the opposite either. Maybe the documentation should be more precise.

With `--eit-normalization`, it says _"Reorganize all EIT sections according to ETSI TS 101 211 rules."_ In that case, it uses the same EIT generator as the `eitinject` plugin for which the documentation is more complete: the input EIT's are considered as an "event database" and all EIT's are regenerated for these events according to ETSI TS 101 211 rules.

In the case of the `eitinject` plugin, there are options to select the combination of pf / schedule and actual / other. By default, they are all generated. In the case of `tspacketize --eit-normalization`, there are no selection options and all EIT's are generated. Maybe options should be added to select pf / schedule and actual / other.

> EIT present is missing for service id 1040 and 1041.

They are present but empty (no event). This is normal. Unlike the `eitinject` plugin which regularly updates the EIT database, `tspacketize --eit-normalization` generates a snapshot of EIT's to broadcast at a given "base time". By default, the base time is the oldest date in all EIT sections (see doc for option `--eit-base-date`). In your case, the oldest date is 2022-10-14 13:00:00, on service 1039. It is consequently used as base date. In services 1040 and 1041, the first events start at 13:15:00 and 13:55:00, respectively. So, at the base date, there are no reported events in these services and the EIT present is empty. These events are reported in the EIT following (section 1).

przemo1983, 2022-10-18:
Thank for explanation. Nice to have would be the option to enable generation of EIT pf only. For now I will split the EITs for each service and generate EIT pf and schedule for one service. Will remove schedules from TS files and merge EIT pf TS for all services in the network.

lelegard, 2022-10-21:
Options `--eit-actual`, `--eit-other`, `--eit-pf`, `--eit-schedule` have been added to `tspacketize`, `tstabcomp` and plugin `inject` (same code everywhere to handle the packetization options). By default, all EIT's are generated. When these options are used, only the requested EIT's are generated.

But keep in mind that these options work on an EIT snapshot only. The plugin `eitinject` remains the preferred way to dynamically reorganize EIT's.

closed 2023-01-06 by lelegard