tsduck/tsduck #1081: [Q] Proper way to tune DVB-S2 multistream with tsscan

Galvo87, 2022-10-18:
Hi, what is the proper way to tune DVB-S2 multistream with tsscan?

PLS: Root+8, 16APSK, Stream 17.

`tsscan -d /dev/dvb/adapter9 --nit-scan --delivery-system DVB-S2 --lnb "Astra 19.2E" --frequency 12188000000 --polarity V --modulation 16-APSK --symbol-rate 31400000 --pls-mode ROOT --pls-code 262140 -g --save-channels channels.xml
`
returns:

`tsscan: cannot scan network, no NIT found on specified transponder
`
which tells me I am surely missing something…
The PLS is listed as Root + 8, maybe that’s the issue?
I see support for COMBO, GOLD, ROOT in the docs.
Also, can `--stream-id 17` be used for selecting that particular stream?

Please note that I can tune and record these FTA channels with ease using dvblast, so no hardware issues.

Galvo87, 2022-11-02:
Any hint about this one?
It seems like there's no way to select a specific substream when using PLS.

lelegard, 2022-11-03:
Sorry for not responding before. I no longer have any satellite reception capability and I cannot test or debug this. I don't even think having used S2 physical layer scrambling (PLS) before.

Before trying to analyze the content of the stream, just make sure you receive it using `tsp -v -I dvb ... | dump`.

Galvo87, 2022-11-06:
Understood. It seems it's not even able to receive the stream.