tsduck/tsduck #1083: [BUG] Compile error with long path

AoD314, 2022-10-19:
#### Bug description:
Compile error with long path

#### How to reproduce:
```
cd /tmp
for (( i=0; i < 13; i++ ))
do
	mkdir very_very_long_path
	cd very_very_long_path
done
git clone https://github.com/tsduck/tsduck.git
cd tsduck
git checkout v3.31-2761
mkdir install
install_folder=$(realpath ./install)
make -j$(nproc) NOEDITLINE=1 NODEKTEC=1 NOTEST=1 NOPCSC=1 NOCURL=1 NODTAPI=1 NOTELETEXT=1 SYSPREFIX=${install_folder}
```
#### Errors and logs:
```
...
  [CXX] tsduck.cpp
  [LD] /tmp/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/tsduck/bin/release-x86_64-zion/objs-libtsduck/alltables.o
  [LD] /tmp/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/tsduck/bin/release-x86_64-zion/objs-libtsduck/allcharsets.o
  [LD] /tmp/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/tsduck/bin/release-x86_64-zion/objs-libtsduck/alldescriptors.o
  [LD] /tmp/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/tsduck/bin/release-x86_64-zion/objs-libtsduck/allplugins.o
  [AR] /tmp/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/tsduck/bin/release-x86_64-zion/libtsduck.a
make[2]: /usr/bin/env: Argument list too long
make[2]: *** [Makefile:129: /tmp/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/very_very_long_path/tsduck/bin/release-x86_64-zion/libtsduck.so] Error 127```
#### Environment:
 - OS: Linux
 - OS version: Any
 - TSDuck full version: tag: v3.31-2761
 - Installation type: github repo

lelegard, 2022-10-19:
Hi @AoD314 

Thanks for reporting this. This is a limitation of the environment (shell and operating system). There is not much we can do, except using a shorter path. Maybe you can try to create a short symbolic link such as `/tmp/tsduck` pointing to the long path and `make -C /tmp/tsduck ...`. Then, let's hope the short symlink will be used.

TSDuck has many files and directories. The `ar` command takes all object files at once to store them in the static library, probably ending with a very-very-long command line in your case. Adding them one by one would reduce the command line. However, doing so results in unacceptably long execution time. This was done at the beginning, years ago, and was a killer. So, everything is added in one pass now. We cannot revert to the unacceptably long build time just because one user uses insanely long paths 😄

lelegard, 2022-10-19:
Sorry but I cannot do this for such an unusual configuration. This is the first time I got such a report. So, your paths must be definitely really long. According to [this post](https://serverfault.com/questions/163371/linux-command-line-character-limit), the max command line size on Linux is hardcoded to 131072 (0x20000) characters.

I can propose a test. In file `src/libtsduck/Makefile`, line 190, replace:
~~~
	@echo '  [AR] $@'; $(AR) $(ARFLAGS) $@ $^
~~~
by this:
~~~
	@echo '  [AR] $@'; cd $(OBJDIR); $(AR) $(ARFLAGS) $@ $(notdir $^)
~~~
If this works for you, I will make the update.

In short, since all objects are generated in the same directory, we `cd` to that directory and then use the base names, without directory, of all object files.

Let us know.

lelegard, 2022-10-19:
And if this works, the build of shared library may fail for the same reason. 

Try the same solution, same file, line 145, replace
~~~
	$(CXX) $(CXXFLAGS) $(SOFLAGS) $^ $(LIBTSDUCK_LDLIBS) $(LDLIBS_EXTRA) $(LDLIBS) -shared -o $@
~~~
by this
~~~
	cd $(OBJDIR); $(CXX) $(CXXFLAGS) $(SOFLAGS) $(notdir $^) $(LIBTSDUCK_LDLIBS) $(LDLIBS_EXTRA) $(LDLIBS) -shared -o $@
~~~

AoD314, 2022-10-19:
It's work !

AoD314, 2022-10-19:
Looks like good solution.

lelegard, 2022-10-20:
The main code base has been updated accordingly.

AoD314, 2022-10-25:
Checked on master (HEAD: 1e5707032b948c07f5278cff1d700cdd7e3536df). All works fine.

closed 2022-10-25 by AoD314