tsduck/tsduck #1089: [BUG] Incorrect interpretation of stream_content=0x3 in component_descriptor

paulhiggs, 2022-10-25:
#### Bug description:
tsComponentDescriptor reports any component_type on the range 0x3F00 ~ 03FFF as being 'subtitles' and branches into SubtitlingType() to resolve the value to a name. According to ETSI EN 300 468, only values 0x3F00 ~ 0x3F3F are allocated to subtitles and other values should be reported acording to their definition, i.e.

- ````0x3F40```` - video spatial resolution has been upscaled from lower resolution source material
- ````0x3F41```` - Video is Standard Dynamic Range (SDR) 
- ````0x3F42```` - Video is High Dynamic Range (HDR) remapped from SDR source material
- ````0x3F43```` - Video is HDR up-converted from SDR source material
- ````0x3F44```` - Video is standard frame rate, less than or equal to 60 Hz
- ````0x3F45```` - High frame rate video generated from lower frame rate source material
- ````0x3F80```` -dependent Spatial Audio Object Coding - Dialogue Enhancement (SAOC-DE) data stream

#### How to reproduce:
<component_descriptor 
   stream_content="3" stream_content_ext="0xF" component_type="0x41" language_code="eng"/> --> video spatial resolution has been upscaled from lower resolution source material

<component_descriptor 
   stream_content="3" stream_content_ext="0xF" component_type="0x43" language_code="eng"/> --> Video is HDR up-converted from SDR source material

etc

#### Expected behavior:
Subtitles should be reported for component_type=0x00~0x3f. Other values are stream capabilities.

#### Additional information:
[ETSI EN 300 468](https://www.etsi.org/deliver/etsi_en/300400_300499/300468/01.17.01_60/en_300468v011701p.pdf) clause 6.2.8

lelegard, 2022-10-25:
Thanks Paul. I fixed it. Please close that issue if you agree with the result.

paulhiggs, 2022-10-25:
This looks right for me on signaled SDR content
````
    - Descriptor 2: Component (0x50, 80), 6 bytes
      Content/type: 0x0341 (Video is standard dynamic range (SDR))
      Component tag: 0 (0x00)
      Language: chi
````

closed 2022-10-25 by paulhiggs