tsduck/tsduck #1090: [FEAT] Unicable/SCIF support

abainbridge, 2022-10-25:
#### Is your feature request related to a problem?
My building uses unicable to distribute satellite signals.

#### Proposed solution:
Add support for unicable.

#### Alternatives:
Use a different program or rewire all the world's buildings that use unicable.

#### Additional information:
I've already implemented a working prototype for this. Before I continue, I thought I'd see whether a pull request that implements this feature is something that would interest the project. Also, it would be good to discuss some design details.

#### Command line parameters

Currently, to tune to DiSEqC I might use a command like:
`tsp -I dvb --adapter 0 --delivery-system DVB-S2 --frequency 985000000 --fec-inner 3/4 --polarity vertical --satellite-number 3  --symbol-rate 29950000 -O file`

All of these parameters continue to be needed to tune via Unicable and there are a whole group of extra parameters needed. Adding a new DVB option like the following might be best:
```
--unicable=<version>,<userband slot>,<userband frequency>[,<pin>]
    version: Which Unicable version to use. Must be 1 or 2.
    userband slot: Which userband slot to tell the Unicable switch to output your channel on. For version 1, must be in the range 1 to 4. For version 2, lmust be in the range 1 to 32.
    userband frequency: Frequency the local tuner must tune to in order to receive the specified userband. In MHz.
    pin: Version 2 only. PIN used to protect the userband for installation aid purposes.
```
The man page text for other parameters might need changing too.

`---frequency` was:
```
      Carrier frequency in Hz (all tuners). There is no default.
```
Might become:
```
      Frequency at which the channel is broadcast, in Hz (all tuners). There is
      no default. In the case of DiSEqC and Unicable, the actual frequency the
      tuner tunes to is calculated from this, to take account of hetrodyning.
```

`--satellite-number` was:
```
      Used for satellite tuners only. Satellite/dish number. Must be 0 to 3
      with DiSEqC switches and 0 to 1 fornon-DiSEqC switches. The default is 0.
```
Might become:
```
      Used for satellite tuners only. Satellite/dish number.
      For DiSEqC 1.0, must be 0 to 3.
      For Unicable 1, must be 0 to 1.
      For Unicable 2, must be 0 to 63.
      The default is 0.
```

lelegard, 2022-10-25:
Hi @abainbridge 

Thanks for the contribution. Do you have public reference to Unicable specification and how it is interfaced on Linux and Windows?

About the option `--unicable=<version>,<userband slot>,<userband frequency>[,<pin>]`, we need to differentiate what is specific to a given carrier and what is specific to a physical installation. For instance, is the userband frequency a fully independent data or is it linked to the carrier frequency and the characteristics of the receiving installation? It looks like the concept of high/low band in European satellite installation, in which case this is part of the LNB configuration. In that case, we need to add new LNB descriptions and only specify the exiting `--lnb` option with the name of a new type of LNB.

I do not think that the description of `--frequency` shall be modified. In all cases, the frequency is a carrier frequency "on air". With all satellite receptions, the computation of the intermediate frequency - for actual tuning - is automatically done based on the description of the LNB.

abainbridge, 2022-10-26:
Everything below is true for Unicable 2. I haven't tried Unicable 1 yet.

> Do you have public reference to Unicable specification

No sorry. This is a very annoying situation. It would certainly be discouraged to search for EN 50607 pdf on baidu.com. (And EN 50494 pdf).

> how it is interfaced on Linux and Windows?

I haven't thought about Windows. On Linux, the interfaces needed for Unicable already exist. We only need `ioctl_fe_set_tone()`, `ioctl_fe_set_voltage()` and `::ioctl(_frontend_fd, ioctl_request_t(FE_DISEQC_SEND_MASTER_CMD)...`

> is the userband frequency a fully independent data or is it linked to the carrier frequency and the characteristics of the receiving installation?

The userband slot number is chosen by the user and is independent of the carrier frequency. The frequency of that slot is up to the Unicable switch. The switch's manual will have to table showing the mapping of slots to frequencies. There is a command to allow the receiver to query that mapping, but many (most?) Unicable devices don't support it (or any of the bi-directional commands).

> It looks like the concept of high/low band in European satellite installation, in which case this is part of the LNB configuration.

No, I don't think so. The receiver sends the `ODU_Channel_change` command to the switch saying which channel frequency+bank it wants, and which userband slot it wants it on.  As a result, the user has to give both the userband slot and frequency to their receiver.

> I do not think that the description of --frequency shall be modified. In all cases, the frequency is a carrier frequency "on air".

I think the description of `--frequency` as `Carrier frequency in Hz (all tuners)` will be confusing because in a Unicable use case because the user knows about two frequencies: their userband frequency and the channel's broadcast frequency. There's nothing in the current description to tell the user which frequency to pass for this parameter. The word "carrier" does not help. ie, it is perfectly correct for me to say something like, "my userband carrier frequency is 985 MHz". Would using the phrase "channel broadcast frequency" be bad for some reason?

lelegard, 2022-10-26:
> On Linux, the interfaces needed for Unicable already exist. We only need ioctl_fe_set_tone(), ioctl_fe_set_voltage() and ::ioctl(_frontend_fd, ioctl_request_t(FE_DISEQC_SEND_MASTER_CMD)...

That is what is used to drive the LNB. This is why I think that some of the Unicable parameters could be handled as an LNB description. Maybe the LNB configuration shall be extended, as I did when adding the specificity of the Japanese ISDB-S standard equipment.

When I look for Unicable on the net, I see many "Unicable LNB" for sale, meaning that you can formalize this form of reception equipment as an LNB.

If you do not have a full specification of Unicable, could you please provide a complete technical explanation of its principles, the various parameters, where they come from and how they are used? I do not want to add a specific option without understanding all the details, especially if the same objective can be achieved using more generic mechanisms which are already implemented.

abainbridge, 2022-10-26:
> That is what is used to drive the LNB.
That is true if your receiver is directly connected to an LNB or to several via a DiSEqC switch. However, if you are one of several receivers sharing an LNB, it is not true. For example, if you've got a DiSEqC multiswitch (my building has a Whyte WM1732) connected to a quattro LNB, then the signalling is going to the DiSEqC switch, not the LNB. The quattro LNB hetrodynes all four bands at once down four coax cables all going to the multiswitch. The receivers send signals to the multiswitch to say which band they want. 

The same is true with our Unicable switch (one of these - https://johansson.be/en/products/digital-scr-multiswitch-4-outputs-9774). The receivers talk to it, not the LNB(s).

> If you do not have a full specification of Unicable, could you please provide a complete technical explanation of its principles, the various parameters, where they come from and how they are used?
I can try. I'll start with an overview of Unicable 2. I should have something for you tomorrow.

abainbridge, 2022-11-03:
Sorry for the delay, I was diverted onto work that was a higher priority for my employer. Having had a chat with my boss, we have decided that adding Unicable support well enough to be acceptable quality to merge is probably going to be too much work.

I've got it working well enough for our purposes. The Unicable optional parameter was added in a very hacky way - sorry I didn't have time to do a better job. I will submit a pull request anyway. Hopefully the code will be of help to someone.

lars18th, 2022-11-09:
Hi @lelegard ,

IMHO the pull request #1101 for minimal JESS (Unicable II) support is very useful. In fact the unicable support is not more that a DiseqC "hack". Therefore, if the code is implemented as a "in_case_of_unicable_change_the_diseqc_cmd_to_this()_and_after_tune_to_the_client_freq". Then it will be simple to support it.

I hope you want to accept to merge it.
Regards.

stulluk, 2024-07-15:
Hi @abainbridge  @lelegard  for this PR, is there a possibility to fix merge conflicts with master and accept it as a "minimally supported feature" ?

Or , do you suggest to use it just for our own purposes only for now?

lars18th, 2024-07-15:
Hi,

I just want to share my opinion: In my view, Unicable support is a must. And even if this PR is not fully complete, it provides a minimum support. Therefore, I suggest to add it in some way.

Thank you all for this very useful and impressive project.

lelegard, 2024-07-15:
Hi,

> is there a possibility to fix merge conflicts with master and accept it as a "minimally supported feature" ?

Merge conflicts are just technical matter, there are always solutions.

On the other hand, I have a problem with "minimally supported feature" (or "minimal hacky changes" as #1101). When we add a feature, we need to do it well and consistently on all platforms (if technically possible), or we don't do it at all. At least not in the main project. Over the years, I tried to preserve the quality, portability, genericity of the project. This has been recognized as a quality of the project. I don't want to spoil it.

I developed TSDuck because I needed it. I have added features when I needed them. However, in all cases, I tried to generalize my instant need to some broader feature to anticipate my - and other's - future needs, with consistent support, test and documentation. If I did it, others can do it too.

I understand that someone may develop a quick modification to match his instant need. However, this remains a personal hack for a personal need. If I merge it in the project, it won't be long until someone else asks for a complete implementation, a port on another OS, etc.

Do you have a complete spec of Unicable, in order to understand the feasibility and complexity of what needs to be implemented?

lars18th, 2024-07-15:
Hi @lelegard ,

The standard involved is **EN 50607:2015**. Its operation is very simple: the request command is sent using the same technology as for addressable switches. And you obtain in a _client_ (fixed) frequency the muxer that you want to tune. Therefore, the only difference at tuner level from a DiseqC switch is that you need to:
- Calculate the target frequency.
- Compute the request command.
- Send the diseqc command (aka unicable cmd).
- Tune to the _client_ frequency (the same for each user).

So perhaps you can share your point of view of the _architecture_ of the code. And after others can complete the code. What you think?

FYI; I'm in the process to update my current sat infrastructure to a multi-sat unicable access.

lelegard, 2024-07-15:
> The standard involved is EN 50607:2015.

For sale, online, from 40€ to 230€, depending on the site (what a wide range...) CENELEC is like ISO, not open source friendly. Unlike DVB which is fully open.

I cannot recommend an architecture without knowing the spec. Usually the problem precedes the solution, not the opposite...

It depends on how the reception differs from the LNB structure. Can it be implemented as a generalization of the current LNB structure? Using just another entry in the [XML description of LNB](https://tsduck.io/download/docs/tsduck.html#lnb-config)? Or with some more code in class LNB? Or need a completely different class? The way command line options will be defined will depend on this (just another LNB name or dedicated option).

konrad-dabek, 2024-07-15:
It's worth to add here, that the SCR/Unicable support will be unavailable on Windows.
BDA at the beginning didn't allow to send RAW DiSEqC commands (needed in this case). Later IBDA_DiseqCommand.put_DiseqSendCommand was added, but practically this method is rarely available in the device drivers (and untested at best), so it will fail for most of the cases. Private extensions would have to be used per device (e.g. TBS provides such access), which doesn't seems to be a way to go for tsduck (hacky solutons).

Moreover - there are two standards, besides EN50607, there's also older EN50494 and EN50607 is not just an extension - ways for counting IF Freq are significantly different. Supporting only EN50607 will rule out support of -so called- Unicable I. (EN50494 = Unicable I, EN50607 - Unicable II, using term "Unicable" is kinda improper - this is or at least it was an Inverto trademark).

lelegard, 2024-07-16:
@konrad-dabek, thank you for the clarification and perspective over EN50494 and EN50607. So, to implement this correctly and thoroughly, we need someone who understand the two levels of Unicable standard and find a generic way to integrate them, preferably, in the LNB description scheme of TSDuck.

lars18th, 2024-07-17:
Hi,

Only as a reference for the standard:

- The original protocol developed before the standard is from the company JULTEC. You can read a small description of JESS here: http://www.jultec.de/gloss_JESS.html
- To read (not download) the standards you can check:
  - EN 50607: https://www.doc88.com/p-6532363610790.html
  - EN 50494: https://www.doc88.com/p-66816249098865.html
- The Linux libdvbv5 library has native support for SCR/Unicable/JESS: https://www.linuxtv.org/docs/libdvbv5/index.html

I hope this helps.

lelegard, 2024-07-17:
> To read (not download) the standards you can check:

@lars18th, call me paranoid if you like, but what about a site embedding scripts containing that kind of stuff:

~~~
function downloaddkclient(_Lu) {
    var _dF = "https://download.daokeyuedu.com/docbox/稻壳阅读器安装程序.exe";
    if (isMacOs()) {
        _dF = "https://download.daokeyuedu.com/docbox/稻壳阅读器安装程序.dmg"
    }
    if (_Lu == 3 || _Lu == 5) {
        var _LI = new Object();
        _LI.a = "download_dk_" + _Lu;
        _LI.btn = _Lu;
        Core.api._DN(_LI);
        saveDkDownloadInfo();
        window.location.href = _dF;
        return
    }
    var _Dv = 0;
    if (_Lu == 1) {
        _Dv = 1
    }
    if (_Lu == 0 || _Lu == 4) {
        var dkinstalled = getCookie("dkinstalled");
        if (dkinstalled == 1) {
~~~

Yeah, probably some client software which "improves your experience". Yeah, of course...

lars18th, 2024-07-17:
Hi @lelegard ,

The doc88.com is just a _last option_ for downloading PDF documents. Use a sandbox for safe viewing.
I don't have a copy of the original documents. And I assumed that your goal was just to understand how the standard works.

massweet, 2025-07-15:
I'd like to refresh this issue.

Starting with the code that my ex-colleague Andy wrote (Hi Andy!) I have resolved conflicts and added EN50494 (Unicable I) support, which I have been able to test because I now have access to a Unicable switch that supports both standards without reconfiguring the hardware.

The standards are still behind a paywall, so I cannot help you to access them.

Addressing some of your questions from earlier:

- Unicable is intended for use in large installations such as hotels, apartment buildings and offices. It permits several receivers to share a single line of COAX up to the switch. This massively reduces the cabling in the building.
- Unicable uses the same hardware bus as DiSEqC, but in a different way. As such no new low-level driver support is required over that which the linux driver already provides.
- Unicable has both unidirectional (receiver to switch using the DiSEqC 1.x hardware) and bidirectional (receiver to switch with reply using DiSEqC 2.x hardware) commands.
- It is perfectly possible to tune to a new channel with unidirectional commands, using either a Unicable I or Unicable II command.
- The channel change command includes parameters for: "user band", "bank" and "tuning word".
- The receiver does not control or setup the LNB, instead it sends Unicable commands via the COAX to the Unicable switch. The Unicable switch then heterodynes *just* the requested bank and frequency onto the coax segment at the user-band frequency, which the receiver then tunes to. Note that this is the only frequency that the receiver needs to tune to.
- The standard does not define the mapping between user-band and user-band frequency. As unidirectional receivers do not have any way of finding the frequency associated with the user band that would not disrupt other users of the COAX segment, every unicable switch I have ever seen comes with a table documenting the mapping.
- The `--unicable` parameters "version", "user-band" and "user-band frequency" are therefore local between the receiver and the Unicable switch.
- "bank" is a bitfield combining "satellite position", "H/V polarity" and "High/Low".

| | Unicable I | Unicable II |
| satellite position | 0-1 | 0-63 |
| user band | 1-8 | 1-32 |

closed 2026-01-11 by github-actions[bot]