tsduck/tsduck #1091: [Q] Error: inject: input bitrate unknown or too low, specify --inter-packet

Litrona, 2022-10-25:
TSP version: 3.32-2973 (over x86_64 GNU/Linux 5.10.136-1) 
I have a UDP multicast MPTS input with 2 services. 
I can send it to a Vatek modulator:

tsp -I ip -b 26214400 224.10.10.7:5000 -O vatek --constellation 64-QAM -r 2/3 -f 474000000 -g 1/4 

and no issue. However when I try to insert new tables (i have enough room):
tsp -I ip -b 26214400 224.10.10.7:5000 -P sdt --service-id 0x0001 --eit-pf 1 --eit-schedule 1 -P sdt --service-id 0x0002 --eit-pf 1 --eit-schedule 1 -P inject -p 20 -b 3008 --stuffing tdt.xml -P timeref --system-synchronous -P eitinject --terrestrial -f eit.xml -O vatek --constellation 64-QAM -r 2/3 -f 474000000 -g 1/4

And I get the message: 

Error: inject: input bitrate unknown or too low, specify --inter-packet

I could also capture the UDP in a dump.ts file: tsp -I ip -b 26214400 224.10.10.7:5000 -O file dump.ts
And also could send it to the Vatek inserting the tables with no issue:

tsp -I file dump.ts -P regulate --pcr-synchronous -P sdt --service-id 0x0001 --eit-pf 1 --eit-schedule 1 -P sdt --service-id 0x0002 --eit-pf 1 --eit-schedule 1 -P inject -p 20 -b 3008 --stuffing tdt.xml -P timeref --system-synchronous -P eitinject --terrestrial -f eit.xml -O vatek --constellation 64-QAM -r 2/3 -f 474000000 -g 1/4

Only when I try to send the UDP MPTS directly to the -P inject seems to not being able to find the bitrate, that in fact it is:
Estimated based on PCR's: ............... 19,906,000 b/s   21,600,128 b/s (tsanalyze dump)

Thanks in advance.

lelegard, 2022-10-25:
Hi @Litrona 

The reason is the `ip` input reading very few packets at the beginning of the stream. This is the same problem, in principle, as #1070. See my explanations here about why the bitrate is not initially known with the `ip` input plugin.

In your case, the plugin `inject`, when starting, wants to evaluates the bitrate of the TS in order to know every which number of packets it should insert a packet from the tables to inject. With the `ip` input plugin, there is not enough packets at the beginning to evaluate the bitrate. That's why `inject` complains.

Soon after the beginning, the bitrate becomes known, when there are enough packets with PCR to evaluate it. That's why you see it in the analysis. But the `inject` plugin wants this at the very beginning. Other input plugins read more initial packets and the bitrate is known from the beginning.

I will try to find a general solution for this. In the meantime, you may specify the option `--inter-packet` as suggested in the error message. With this option, `inject` does not need to know the TS bitrate. Since you inject a TDT template, there is no need to be very precise. You just need to know an approximate value of the expected bitrate.

Let's say BR = approximate bitrate in b/s (e.g. 5,000,000 b/s) and D = cycle duration of your table in seconds (e.g. 20 s. for a TDT):

- inter-packet = (D * BR) / (8 * 188)

Litrona, 2022-10-25:
OK, done thx

closed 2022-10-25 by Litrona