tsduck/tsduck #1093: Not able to install tsduck on CentOS 7

jatinderkumargupta, 2022-10-26:
#### Bug description:
Installation of TSDuck on CentOS 7 is not working.

#### How to reproduce:
Download TSDuck from the below links:
[3.12-781](https://objects.githubusercontent.com/github-production-release-asset-2e65be/94976922/51c78822-6b6e-11e8-884a-36bc31148212?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20221026%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20221026T052932Z&X-Amz-Expires=300&X-Amz-Signature=f1cc99eb885bd326b424e292ad726b3a515ffee2918d40e4a677124157d73180&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=94976922&response-content-disposition=attachment%3B%20filename%3Dtsduck-3.12-781.fc28.x86_64.rpm&response-content-type=application%2Foctet-stream)
Also tried below version:
[3.31-2761](https://github.com/tsduck/tsduck/releases/download/v3.31-2761/tsduck-3.31-2761.el8.x86_64.rpm)
Installing that into CentOS 7.

#### Errors and logs:
$ sudo yum localinstall tsduck-3.12-781.fc28.x86_64.rpm --skip-broken
[sudo] password for jatinder:
Loaded plugins: fastestmirror
Examining tsduck-3.12-781.fc28.x86_64.rpm: tsduck-3.12-781.fc28.x86_64
Marking tsduck-3.12-781.fc28.x86_64.rpm to be installed
Resolving Dependencies
--> Running transaction check
---> Package tsduck.x86_64 0:3.12-781.fc28 will be installed
--> Processing Dependency: libstdc++.so.6(CXXABI_1.3.8)(64bit) for package: tsduck-3.12-781.fc28.x86_64
Loading mirror speeds from cached hostfile
 * base: [repo.extreme-ix.org](http://repo.extreme-ix.org/)
 * epel: [repo.extreme-ix.org](http://repo.extreme-ix.org/)
 * extras: [repo.extreme-ix.org](http://repo.extreme-ix.org/)
 * rpmfusion-free-updates: [ftp.nluug.nl](http://ftp.nluug.nl/)
 * updates: [centos.excellmedia.net](http://centos.excellmedia.net/)
--> Processing Dependency: libstdc++.so.6(GLIBCXX_3.4.20)(64bit) for package: tsduck-3.12-781.fc28.x86_64
--> Processing Dependency: libstdc++.so.6(GLIBCXX_3.4.21)(64bit) for package: tsduck-3.12-781.fc28.x86_64Packages skipped because of dependency problems:
    tsduck-3.12-781.fc28.x86_64 from /tsduck-3.12-781.fc28.x86_64
~~~
[be sure to surround verbatim log lines with "~~~" lines]
~~~

#### Environment:
 - OS: CentOS
 - OS version: 7
 - TSDuck full version: 3.12-781

lelegard, 2022-10-26:
Hi @jatinderkumargupta,

This is not a bug, you use a package which is not compatible with your distro. The package you use is named `.fc28`. Following the usual naming conventions, this means that this a package for Fedora 28. Packages for CentOS 7 are named `.el7`. 

If you do not find an `el7` package, you need to rebuild TSDuck from sources. 

Note that an `.el7` package is built for a given CentOS 7.x and may not work either on a distinct 7.y.

TSDuck is an open-source project with limited resources. Each release is provided with binary packages for the latest version of a few distros. With other distros or other versions, you need to rebuild from source. The process is straightforward and documented in the programming guide (see tsduck.io).

jatinderkumargupta, 2022-11-09:
Thanks @lelegard , the package with .el7 worked for me.

closed 2022-11-09 by lelegard