tsduck/tsduck #1094: [Q] How to pass stdin commands?

lars18th, 2022-10-27:
Hi @lelegard ,

I want to do something like this example:
- ` CONSUMER process | tsp [...] | PRODUCER process`

Therefore:
- In this scenario the PRODUCER is a binary that generates a MPEG-TS and outputs it over the PIPE of the stdout. Think on this like the FFmpeg tool.
- Then the CONSUMER is a process that consumes the MPEG-TS stream from the producer filtered by the TSDuck `tsp` tool. In my case this consumer will be an streamer, but it can be a player like VLC.

So the question is:
- How I can receive _messages_ from the CONSUMER to the PRODUCER using the PIPE?

My idea is to _write_ lines to the PIPE from the consumer and receive the same line in the PIPE with the producer. So make that _transparent_: from  `consumer.exe | producer.exe` to `consumer.exe | tsp.exe | producer.exe`.

Solutions?

lelegard, 2022-10-27:
The `tsp` input can be made of TS packets only. A text line, or anything which is not a TS packet, will generate an error. However, if you have control over the producer and consumer, you can encapsulate your command in TS packets, for instance in an unused PID such as 0x1FFE. Just make sure to not remove orphan PIDs in the `tsp` chain and don't send too many of them to avoid altering the bitrate (if this is a concern).

lars18th, 2022-10-27:
Hi @lelegard ,

Thank you for the response. But this to pass "encapsulated" commands from the PRODUCER to the CONSUMER. But what I want to do is the other way: from the consumer to the producer. See this:
```

        <-----<<  MPEG-TS <----<<
CONSUMER            TSP          PRODUCER
        >>-----> user cmds >>--->
```

The consumer is a PLAYER and the producer is a CAPTURE process. So I want to control the producer _inline_ sending some commands over the PIPE in the reverse flow.

lelegard, 2022-10-27:
Sorry, I did not read carefully enough. In that case, you need a dedicated application-specific return channel (named pipe or other interprocess communication). The input and output pipes of `tsp` cannot be used in the reverse direction. The opposite file descriptor (for instance the read fd of the output pipe) is closed. This is required to be notified of partner termination.

lars18th, 2022-11-09:
Ok, thanks for the answer. Now I understand why this is not possible. I'll try to find another solution. Regards.

closed 2022-11-09 by lars18th