tsduck/tsduck #1095: [BUG] Plugin SVRESYNC fails when the PCR pid is independent

lars18th, 2022-10-27:
#### Bug description:

When the PCR is included in an independent pid of a service (for example, when removing the video of a tv service to convert it to radio, leaving the original pid with only the PCR timestamps). In this case, applying the `svresync` plugin to this service, the result is buggy.

#### How to reproduce:

Source is `test-001.ts` from https://github.com/tsduck/tsduck-test/blob/master/input/test-001.ts

Test 1: OK (only zap DECALE {8802} & CNEWS {8810} and analyze)
```
tsp.exe -v \
 -I file test-001.ts \
 -P zap -s --eit 8810 8802 \
 -P analyze -w \
 -O drop
```

Test 2: OK (Resync DECALE to CNEWS)
```
tsp.exe -v \
 -I file test-001.ts \
 -P zap -s --eit 8810 8802 \
 -P svresync -s 8810 8802 \
 -O drop
```
Output: `* svresync: adjusted 96 PCR, 126 PTS, 72 DTS`

Test 3: FAIL (Resync DECALE to CNEWS after converting DECALE to a radio service) (video pid in DECALE is 210)
```
tsp.exe -v \
 -I file test-001.ts \
 -P zap -s --eit 8810 8802 \
 -P filter -p 210 --set-label 1 \
 -P filter --only-label 1 --pcr \
 -P craft  --only-label 1 --no-payload --clear-pusi \
 -P continuity -f -p 210 \
 -P pmt -s 8802 -r 210 \
 -P sdt -s 8802 -t 2 \
 -P rmorphan \
 -P svresync -s 8810 8802 \
 -O drop
```
Output: `* svresync: adjusted 0 PCR, 43 PTS, 0 DTS`

#### Expected behavior:

The expected behaviour is that the changes of the plugin `svresync` will be same, as the original PCR timestamps are un touched, like the rest of PTS and DTS values. But `adjusted 0 PCR, 43 PTS, 0 DTS` instead of `adjusted 96 PCR, 126 PTS, 72 DTS`.

#### Errors and logs:

~~~
[be sure to surround verbatim log lines with "~~~" lines]
~~~
I feel all is well expressed before.

#### Environment:
 - OS:  Windows
 - OS version:  Windows 10 64bit
 - TSDuck full version:  3.29-2651
 - Installation type:  PORTABLE

#### Additional information:

Identical behaviour in Linux.

lars18th, 2022-11-09:
Hi @lelegard ,

Any reason not to comment on this bug?
Here more information using the last version with the timestamps values in the analyze plugin (#1096):

Test A: OK (initial analysis)
```
tsp.exe -v \
 -I file test-001.ts \
 -P zap -s --eit 8810 8802 \
 -P analyze -w \
 -O drop
```
```
|============================================================================================|
|  PID: 0x00D2 (210)                                                              AVC video  |
|  PES stream id: 0xE0 (Video 0)                                                             |
|  1920x1080, high profile, level 4.0, 4:2:0                                                 |
|  Service: 0x2262 (8802) CANAL+ DECALE                                                      |
|--------------------------------------------------------------------------------------------|
|  Single Service PID             Transport:                     Discontinuities:            |
|  Bitrate: ...... 3,072,606 b/s  Packets: .............. 7,173  Expected: .............. 0  |
|  Access: ............... Clear  Adapt.F.: ............... 185  Unexpect: .............. 1  |
|                                 Duplicated: ............... 0  PES:                        |
|                                 PCR: ..................... 97  Packets: .............. 84  |
|                                 TSrate: ...... 47,782,780 b/s  Inv.Start: ............. 0  |
|  Clock values range:                                                                       |
|  PCR: ...... 1,722,694,546,453  PTS: .......... 5,742,376,801  DTS: ....... 5,742,369,601  |
|  to: ....... 1,722,785,272,472  to: ........... 5,742,686,401  to: ........ 5,742,668,401  |
|============================================================================================|
|  PID: 0x00DD (221)                                                       AC-3 Audio (fra)  |
|  PES stream id: 0xBD (Private stream 1)                                                    |
|  AC-3, stereo (L,R), @48,000 Hz, complete main, bsid 6                                     |
|  Service: 0x2262 (8802) CANAL+ DECALE                                                      |
|--------------------------------------------------------------------------------------------|
|  Single Service PID             Transport:                     Discontinuities:            |
|  Bitrate: ........ 191,476 b/s  Packets: ................ 447  Expected: .............. 0  |
|  Access: ............... Clear  Adapt.F.: ................ 43  Unexpect: .............. 0  |
|                                 Duplicated: ............... 0  PES:                        |
|                                 PCR: ...................... 0  Packets: .............. 22  |
|                                                                Inv.Start: ............. 0  |
|  Clock values range:                                                                       |
|                                 PTS: .......... 5,742,318,223                              |
|                                 to: ........... 5,742,620,623                              |
|============================================================================================
```

-----
Test B: **ERROR** (troubles with some timestamps)
```
tsp.exe -v \
 -I file test-001.ts \
 -P zap -s --eit 8810 8802 \
 -P svresync -s 8810 8802 \
 -P analyze -w \
 -O drop
```
```
|============================================================================================|
|  PID: 0x00D2 (210)                                                              AVC video  |
|  PES stream id: 0xE0 (Video 0)                                                             |
|  1920x1080, high profile, level 4.0, 4:2:0                                                 |
|  Service: 0x2262 (8802) CANAL+ DECALE                                                      |
|--------------------------------------------------------------------------------------------|
|  Single Service PID             Transport:                     Discontinuities:            |
|  Bitrate: ...... 3,072,318 b/s  Packets: .............. 7,173  Expected: .............. 0  |
|  Access: ............... Clear  Adapt.F.: ............... 185  Unexpect: .............. 1  |
|                                 Duplicated: ............... 0  PES:                        |
|                                 PCR: ..................... 97  Packets: .............. 84  |
|                                 TSrate: ...... 47,773,759 b/s  Inv.Start: ............. 0  |
[... 109 lines cut ...]
|  PCR: ...... 1,722,694,546,453                                                             |
|  to: ....... 1,722,785,272,472                                                             |
|============================================================================================|
|  PID: 0x00DD (221)                                                       AC-3 Audio (fra)  |
|  PES stream id: 0xBD (Private stream 1)                                                    |
|  AC-3, stereo (L,R), @48,000 Hz, complete main, bsid 6                                     |
|  Service: 0x2262 (8802) CANAL+ DECALE                                                      |
|--------------------------------------------------------------------------------------------|
|  Single Service PID             Transport:                     Discontinuities:            |
[... cut ...]
|                                 PTS: .......... 5,742,318,223                              |
|                                 to: ........... 5,742,620,623                              |
|============================================================================================|
```
Problem: None of the timestamps marks have been changed. Neither those of PCR nor those of PTS. So the plugin fails to apply it when the PCR is in a independent pid with only PCR values.

-----

Please can you check this?
Thank you.

lars18th, 2022-11-10:
Hi @lelegard ,

Regarding the _current_ troubles with the `svresync` plugin, Iet to explain two topics:
1. I know that the TSDuck is a _realtime_ tool. So it **can't** change the time (someone says: _back to the future?_). Then the outcomes from the current implementation regarding the initial _not changed_ timestamps could be considered correct (Test C). That's because **until the plugin doesn't have the timestamps of the reference pid then it's impossible to calculate any _new_ values**. And I agree with this. However, the current implementation lacks a method to discard these _uncheangable_ timestamps. Let me to explain more: if you need to "resync" some service that after you want to remux, then the "non-recomputed" timestamps generate troubles in the muxer. So my suggestion is to add one new parameter to the plugin like `-P svresync --mute-pid value ...` which indicates that the plugin will remove the pid _value_ until you have the required information to change the timestamps of _this_ pid. Because, yes the plugin can't know the pids of the services until it reads them, and it can't know the target time value to recalculate the timestamps after have all the information. But until this point of the time, if the user knows the involved pids and can pass them as a parameters, then you could mute these pids until this time. Don't you think so? 
2. The other issue is created by a simple edge case: when the PCR timestamps are inserted in a PID that doesn't includes any PES data (check the `PCR (not otherwise referenced)` message). The plugin seems that in this case it refuses to change the timestamps because it doesn't agree with the PMT information. However, this is a valid stream. Therefore I suggest to remove this limitation. You agree?

I hope you want to target this issue soon.
Regards.

lars18th, 2022-11-10:
Hi @lelegard ,

Regarding the second issue (failure on Test D because the empty PCR pid) this patch resolves the issue:
```
diff --git a/src/tsplugins/tsplugin_svresync.cpp b/src/tsplugins/tsplugin_svresync.cpp
index 0212034..2493a52 100644
--- a/src/tsplugins/tsplugin_svresync.cpp
+++ b/src/tsplugins/tsplugin_svresync.cpp
@@ -229,7 +229,7 @@ ts::ProcessorPlugin::Status ts::SVResyncPlugin::processPacket(TSPacket& pkt, TSP
     }

     // Adjust time stamps in the target service (if we have a reference).
-    if (_last_ref_pcr != INVALID_PCR && _target_pids.test(pkt.getPID())) {
+    if (_last_ref_pcr != INVALID_PCR && (_target_pids.test(pkt.getPID()) || _target_pcr_pid == pkt.getPID())) {

         // If the target packet contains a PCR, adjust the time difference between the two services.
         if (pkt.hasPCR()) {
```

So, please merge it. However, I'm unsure if this edge case could be present on other parts of the toolkit. So perhaps it will be necessary to review the `PIDSet::test()` function to consider this case. What you think?

lars18th, 2022-11-10:
Hi @lelegard ,

Regarding the edge case related to pids with only PCR timestamps and no PES content, the filter `rmorphan` has the same behaviour. I feel the `PMT` class requires a review to include such pids. You agree with that?

lars18th, 2022-11-12:
Hi @lelegard ,

Regarding the enhancement for the _non_changed_timestamps_ I've another proposal different to the `--mute-pid value`. I feel this suggestion will be more close to your idea of the KISS model and the current implementation:

- Add to the `svresync` plugin the parameter `--touch-label-pid 1:100,101,...` : the concept is very easy. When the changing of the timestamps starts to apply inside a pid (i.e. 100 or 101 or etc.), the label flag indicated (i.e 1) is then set for all the rest of packets in this pid.

I believe you will quickly understand the utility of this. Does this solution look simple and clean? What do you think?

lelegard, 2022-11-12:
Hi

I am going to merge #1116, thank you.

> Regarding the edge case related to pids with only PCR timestamps and no PES content, the filter `rmorphan` has the same behaviour.

I don't think so. The PCR PID of each service is marked as referenced in `rmorphan`.

Referencing an independent PID as PCR PID in a PMT is well-known and usually handled properly. Except in `svresync` where it was forgotten. It is possible that other omissions exist but this is not a general problem and must be investigated case by case.

> I feel the `PMT` class requires a review to include such pids.

The `PMT` class simply describes a PMT. Properly using its `pcr_pid` is an application issue and does not impact the `PMT` class itself.

closed 2022-11-12 by lelegard

lars18th, 2022-11-13:
> I am going to merge #1116, thank you.

You're welcome.

> > Regarding the edge case related to pids with only PCR timestamps and no PES content, the filter `rmorphan` has the same behaviour.
> 
> I don't think so. The PCR PID of each service is marked as referenced in `rmorphan`.

Right. I've check it after commenting it. So don't worry at all.

> > I feel the `PMT` class requires a review to include such pids.
> 
> The `PMT` class simply describes a PMT. Properly using its `pcr_pid` is an application issue and does not impact the `PMT` class itself.

Yes, another time. Handling the `PMT::pcr_pid` it's sufficient to overcome this edge case in any circunstance. If I encounter a similar problem, then I'll report it. Thank you for your support.

commit 8050de35 (2022-11-12): Fix svresync plugin with independent PCR pid

Add support for the edge case when the PCR is inside an independent pid. Fixes #1095.
PR #1116 (2022-11-12): Fix svresync plugin with independent PCR pid
Add support for the edge case when the PCR is inside an independent pid. Fixes #1095.