tsduck/tsduck #1096: [FEAT] Include in the plugin "analyze" Timestamps marks

lars18th, 2022-10-27:
#### Is your feature request related to a problem?

When using the `analyze` plugin (even with the -w wide output) the clock timestamps of the pids are missing. This information is useful to check the outcomes from some plugins (like svresync, pcrcopy, etc.).

#### Proposed solution:

Enhance the data printed by the `analyze` plugin including almost the STARTING CLOCK and END CLOCK values for PCR, PTS and DTS when they exist.

#### Alternatives:

The alternative is to open the output in another testing tool (like the good DVB inspector), but it's a slow process and difficult to automate.

#### Additional information:

The support for changing timestamps, and not only PCR but PTS and DTS are very useful to do complex processing actions without remuxing. Futhermore, for a reliable remuxing it's very important that the incoming TS is CBR and clean in respect to the timestamps. So, please add this necessary information.

lelegard, 2022-11-05:
Done. Added first and last PCR, PTS, DTS in grid, normalised and json formats.

lars18th, 2022-11-09:
Hi @lelegard ,

Thank you for the improvement `tsp -I file test-001.ts -P analyze -w  -O drop`:
```
|============================================================================================|
|  PID: 0x00D2 (210)                                                              AVC video  |
|  PES stream id: 0xE0 (Video 0)                                                             |
|  1920x1080, high profile, level 4.0, 4:2:0                                                 |
|  Service: 0x2262 (8802) CANAL+ DECALE                                                      |
|--------------------------------------------------------------------------------------------|
|  Single Service PID             Transport:                     Discontinuities:            |
|  Bitrate: ...... 3,072,606 b/s  Packets: .............. 7,173  Expected: .............. 0  |
|  Access: ............... Clear  Adapt.F.: ............... 185  Unexpect: .............. 1  |
|                                 Duplicated: ............... 0  PES:                        |
|                                 PCR: ..................... 97  Packets: .............. 84  |
|                                 TSrate: ...... 47,782,780 b/s  Inv.Start: ............. 0  |
|  Clock values range:                                                                       |
|  PCR: ...... 1,722,694,546,453  PTS: .......... 5,742,376,801  DTS: ....... 5,742,369,601  |
|  to: ....... 1,722,785,272,472  to: ........... 5,742,686,401  to: ........ 5,742,668,401  |
|============================================================================================|
```
However, I'm missing three things:
1. The _number_ of PTS and DTS timestamps. In the initial summary is the number of packets with PCR, but not with PTS nor DTS. So it's difficult to check if some output processing timestamps is correct or not. You can add this too, please?
2. The printed values are in ticks, but is more common to use time representation. Any option to enable this alternative format?
3. No info about turn back of the clock. When some reset of the clock appears will be useful to notice it.

Regards.

lars18th, 2022-11-15:
Hi @lelegard ,

Please add almost this too...

```
|                                 Duplicated: ............... 0  PES:                        |
|                                 PCR: ..................... 97  Packets: .............. 84  |
|                                 TSrate: ...... 47,782,780 b/s  Inv.Start: ............. 0  |
|                                                                PTS: .................. 16  |
|                                                                DTS: .................. 10  |
|  Clock values range:                                                                       |
```

Thank you.

lelegard, 2022-11-15:
I will add total counts of PCR, PTS, DTS, as well as number of "leaps", typically more than a few seconds between adjacent values. The idea is to detect values wrapping up. But "discontinuities" can also be ahead.

lars18th, 2022-11-15:
> I will add total counts of PCR, PTS, DTS, as well as number of "leaps", typically more than a few seconds between adjacent values. The idea is to detect values wrapping up. But "discontinuities" can also be ahead.

That's perfect! ❤️

lelegard, 2022-11-17:
Done...

lars18th, 2022-11-18:
Thank you @lelegard .

```
$ tsanalyze.exe -w test-001.ts
...
|============================================================================================|
|  PID: 0x00D2 (210)                                                              AVC video  |
|  PES stream id: 0xE0 (Video 0)                                                             |
|  1920x1080, high profile, level 4.0, 4:2:0                                                 |
|  Service: 0x2262 (8802) CANAL+ DECALE                                                      |
|--------------------------------------------------------------------------------------------|
|  Single Service PID             Transport:                     Discontinuities:            |
|  Bitrate: ...... 3,171,148 b/s  Packets: .............. 7,402  Expected: .............. 0  |
|  Access: ............... Clear  Adapt.F.: ............... 189  Unexpect: .............. 2  |
|                                 Duplicated: ............... 0  PES:                        |
|                                 TSrate: ...... 47,789,188 b/s  Packets: .............. 85  |
|                                                                Inv.Start: ............. 0  |
|  Clock values range:                                                                       |
|  PCR: .................... 100  PTS: ..................... 85  DTS: .................. 74  |
|  from ...... 1,722,691,708,887  from .......... 5,742,384,001  from ....... 5,742,366,001  |
|  to ........ 1,722,785,272,472  to ............ 5,742,686,401  to ......... 5,742,668,401  |
|  Leaps: .................... 0  Leaps: .................... 0  Leaps: ................. 0  |
|============================================================================================|
```

closed 2022-11-18 by lars18th