tsduck/tsduck #1097: [BUG] Playing a TS File with TSDuck. The continuity plugin throws there are missing packets, while in reality the TS File does not contain a single error

moving-digital, 2022-10-30:
#### Bug description:
I'm using TSDuck to play a TS File from both a Windows Machine and a Debian one, and the `continuity` plugin throws there are missing packets, while in reality the TS File does not contain a single error:

File played from Windows with TSDuck 3.31-2761
[image]

File played from Debian11 (Bulleseye) with TSDuck 3.32-2978
[image]

I'm able to assess the TS File is 100% clean because I have analysed it with 3 different tools:
- TSDuck with tsanalyse
- Sencore CMA1820
- TSReader Pro
- 
All the above tools show that there is not a single TS packet with invalid sync nor transport error. You will find the screenshots attached below.

#### How to reproduce:
Download the TS file here:
https://wetransfer.com/downloads/33e87c5a4fb87eda15b4a7ab8dc5fc7020221030084417/c0f7ae1cb27bfcd9d99611570644bc7120221030084426/ccbdd8
--
1 item
Channel_NTV.ts130 MB

Use the following command:
`tsp.exe --verbose --initial-input-packets 7 --max-input-packets 7 --max-flushed-packets 7 --max-output-packets 7 -I file ..\video\Channel_NTV.ts --infinite -P regulate -P continuity --fix -O ip 224.1.1.3:10003 -l 192.168.1.118 --packet-burst 7 --enforce-burst`

I also tried a simpler command, with the same result:
`tsp.exe --verbose -I file ..\video\Channel_NTV.ts --infinite -P regulate -P continuity --fix -O ip 224.1.1.3:10003 -l 192.168.1.118 --packet-burst 7 --enforce-burst`

#### Expected behavior:

#### Errors and logs:
Sencore CMA1820:
[image]

TSReader Pro:
[image]

TSDuck tsanalyze:
[image]

#### Environment:
Machine1
 - OS: Windows
 - OS version: 10
 - TSDuck full version: 3.31-2761
 - Installation type: Official binary

Machine2
 - OS: Debian
 - OS version: Bullseye
 - TSDuck full version: 3.32-2978
 - Installation type: Official binary

#### Additional information:

lelegard, 2022-10-30:
I can't analyse the file now, I am away and have nothing more than my phone. 

However, I am pretty sure that the errors appear when the file wraps up. At the end of the file, there is no reason that the continuity counters are adjacent to the one at the beginning. The `continuity` plugin analyzes a continuous stream and has no information on the nature of the input. When the file wraps up, everything is broken, CC, PCR, frames, etc. And one CC error appears once on each PID. 

Errors occur in bursts at the same place in the file, exactly one error per PID, at 700k packets, 1.4 packets, etc. So, your file must have 700k packets.

You say that your file has no CC error. If you analyse the file once, there is no error. But if you append the file to itself, the CC errors will appear at 700k packets.

moving-digital, 2022-10-30:
Thanks a lot, Thierry. 

I think I got confused because the CC errors don't happen all at once when looping: as you can see, the PID 0x02A4 is experiencing 12 packets loss in the middle of the file, but not right after the file has looped, which is weird to me.

It would be great to get your confirmation that this is a normal behaviour.

lelegard, 2022-10-30:
The plugin `continuity` can realize that there is a discontinuity only when it encounters the first packet of each PID after the loop back. For each PID, the discontinuity appears at index file size + index of first packet of that PID, then with a constant offset equal to the file size. 

You have to check what is this PID 0x02A4. If this is a "sparse" PID which appears at index 500k packets for the first time, the discontinuity is discovered at packet 700k (initial round) + 500k (first packet in that PID in second round) = 1.2 packets. Please check.

moving-digital, 2022-10-30:
The PIS is the SCTE-35 DPI one.

lelegard, 2022-10-30:
So it seems normal that nothing happens on that PID until there is an event to signal. Again, I bet that its first packet is at offset ~500k. Everything is normal. 

If you have a tool that does not report any error on the _file_, then try to double the file, ie copy it twice in a new file, with double size. Then reuse the same tool on the new file with double size and it will report the same errors as the plugin `continuity`.

moving-digital, 2022-10-31:
Hi @lelegard 

You were right: the CC error on the SCTE-35 PID matches the time where the the cue is fired in the PID...

Sorry for the unnecessary ticket.

closed 2022-10-31 by moving-digital