tsduck/tsduck #1099: timeshift usage questions[Q]

rebATgtv, 2022-11-01:
Hi

First of all, I want to give my appreciation for the tsduck tools and the developer/s.

I have some usage questions regarding the timeshift plugin.

When using the timeshift plugin with input and output as ip unicast or multicast, we experiences some odd issues.

Our setup is as following.

The teleste mux sends an cbr transport stream over multicast to the server with tsduck installed.
Then tsduck recieves the stream and timeshifts the transport stream for 4 hours and then sends it back to the teleste mux also over multicast.

example command.
/usr/bin/tsp -I ip 239.6.3.4:6201 -P timeshift -t 14400000 -O ip 239.6.3.50:6301 -l 192.168.3.191

When we check the stream from tsduck on the teleste mux, there is allot of cc errors occurring on many of the services in the transport stream.

We simply cant figure out where the issue is, we have tried se if we could se it with debug enabled. every thing seems to be in order.

lelegard, 2022-11-03:
Concerning the CC issue, I would recommend to add a `continuity` plugin before and after the `timeshift` plugin, to check where the CC errors appear. Something like;
~~~
tsp -I ip 239.6.3.4:6201 \
    -P continuity --tag before \
    -P timeshift -t 14400000 \
    -P continuity --tag after \
    -O ip 239.6.3.50:6301 -l 192.168.3.191
~~~
The `--tag` option is used to identify which instance of the plugin issues errors.

Second point, using `-P timeshift --time NN` assumes that you have a precise knowledge of the bitrate from the beginning. 
~~~
  -t milliseconds
  --time milliseconds
      Specify the size of the time-shift buffer in milliseconds. The initial
      bitrate is used to convert this duration in number of packets and this
      value is used as fixed-size for the buffer. This is convenient only for
      constant bitrate (CBR) streams. There is no default, the size of the
      buffer shall be specified either using --packets or --time.
~~~
In practice, the time-shift plugin needs a size in packets for its buffer. If the initial bitrate is correctly estimated, the plugin computes the corresponding number of packets. This is fine with most input plugins, except the `ip` plugin which receives too few initial packets (usually 7) to have a precise estimation of the bitrate. If you have a prior knowledge of the expected CBR, try to use `-P timeshift --packets NN` to size the time-shift buffer.

rebATgtv, 2022-11-03:
Thanks for the reply, I have tried the suggested command to identify where the cc errors occur.
Here is the weird part,  there isn't any tag prints on before or after. I also tried with debug on.

I haven't tried to specified the delay in number of packets,  so that's my next step.

lelegard, 2022-11-03:
So, the problem is after `tsp`, in the reception of TS packets, in the tool which reports the CC errors.

Maybe the output of the time-shift plugin is too bursty. Try to add a `regulate` plugin just before the `ip` output. RTFM for use cases of the `regulate` plugin.

rebATgtv, 2022-11-03:
Hi

We have tried with number of packets instead of time with the timeshift plugin and added the regulate just before ip output.
It seems that I am unable to manually set the --packet-burst value if i use the option.

regulate sets the new regulation burst to 337 packets regardless what we define the --packet-burst value to with the regulate plugin.

lelegard, 2022-11-03:
> regulate sets the new regulation burst to 337 packets regardless what we define the --packet-burst value 

How do you see this? What are the exact messages?

rebATgtv, 2022-11-03:
here is the command we tried:
/home/transkodning# tsp -d -I ip 239.6.3.4:6201 -P timeshift -p 2022048 -P regulate --packet-burst 100 -O ip 239.6.3.50:6301 -l 192.168.3.191

and the regulate output is:
* regulate: regulated at bitrate 50,685,743 b/s
* regulate: regulated at bitrate 50,685,743 b/s
* Debug: regulate: new regulation, burst: 10,000,000 nano-seconds, 337 packets
* Debug: ip: input: got bitrate 50,685,727 b/s
* regulate: regulated at bitrate 50,685,727 b/s
* Debug: regulate: new regulation, burst: 10,000,000 nano-seconds, 337 packets

lelegard, 2022-11-03:
The responses are in the messages. TSDuck has determined that 10 ms is the lowest application clock resolution for your operating system. An application cannot wait a shorter delay. Your bitrate is high, 50 Mb/s. Meaning that 10 ms of content represents 337 TS packets. You cannot go below for that bitrate on that machine.

Specialized NIC's may probably regulate with a finer hardware delay (without `regulate` plugin). But the `ip` plugin uses a generic UDP/IP socket interface which sends packets when `send()` is called. Meaning that your cannot use shorter delays between calls to `send()` than allowed by the kernel for applications.

rebATgtv, 2022-11-03:
Okay, so our limitations maybe the hardware we are using?

rebATgtv, 2022-11-07:
Hi

We have made some progress regarding limiting the packets flow to ip output.

Here is our current command.

/usr/bin/tsp --max-output-packets 200 --max-flushed-packets 400 -d -I ip 239.6.3.4:6201 -P timeshift -p 2022048 -P regulate -O ip 239.6.3.50:6301

After implementing the "--max-output-packets 200" and "--max-flushed-packets 400" the cc error rate dropped significantly, the count is lower that 200 now.

Is there a rule of thumb for how the values for --max-output-packets and --max-flushed-packets should be for our setup.