tsduck/tsduck #1100: how to convert a movie mp4 to hls in tsduck using tsp toolkit

udhayNuk, 2022-11-03:
tsp -I hls Idiot.mp4 -O hls manifest.m3u8
 
how to convert for  VOD playlist

comment the full tsp mp4 to hls code 
 
Thanking you in advance

lelegard, 2022-11-03:
TSDuck, as expressed in its name, manipulates transport streams (TS), not MP4 files.
Have a look at ffmpeg for conversion of container format.

udhayNuk, 2022-11-04:
@lelegard 
i have another doubt please clear my doubt please help me

how to convert tsp udp to hls with multiresolution for OTT streaming in tsduck toolkit

tsp -I ip 233.1.1.1:1001 -O hls --duration 3 -p master.m3u8 stream.ts

how to convert hls with multiresolution for ott streaming in tsduck

and how to delete the hls segment file after 10 segment file automatically for live streaming

tell me a full script how to delete the segment file automatically and for multiple resolution OTT streaming udp to hls

lelegard, 2022-11-04:
HLS with multiresolution means:
- Create N distinct TS with transcoded video (distinct resolution and/or codec) with sliding segments
- Create and regularly update N distinct media playlists, one per TS
- Create one static master playlist, containing references to all media playlists

TSDuck is a transport stream toolkit, not an HLS system, and `tsp` uses one input TS and produces one output TS. For ` tsp`, `hls`  is just an input and output plugin, reading one TS, producing one TS.

Additionally, TSDuck has no audio/video transcoding capability. This is the domain of ffmpeg and other specialised tools.

Consequently, one instance of `tsp` cannot create multiple output streams. You may perhaps build an HLS system using several instances of `tsp` and `ffmpeg`. Open source tools such as TSDuck and FFmpeg provide the building blocks but this is your own job to design the system. Nobody will design it for you for free.

Concerning your last point, the automatic deletion of segments, please read the TSDuck user's guide, output plugin `hls`:
~~~
  --live value
      Specify that the output is a live stream. The specified value indicates
      the number of simultaneously available media segments. Obsolete media
      segment files are automatically deleted. By default, the output stream is
      considered as VoD and all created media segments are preserved.
~~~

udhayNuk, 2022-11-05:
@lelegard 
thanking you dude

udhayNuk, 2022-11-05:
@lelegard 
another one more doubt please help me dude

tsp -I ip 233.1.1.81:1001 -P zap 2 -P regulate --pcr-synchronous -P pcrbitrate -P analyze -i 10 -o report.txt -O ip -l 127.0.0.1 235.1.1.2:1004 --enforce-burst

how to encrypt and decrypt tsp tsduck upd to udp conversion

thank you in advance

lelegard, 2022-11-05:
It depends on what you mean by "encrypt". Which cipher? Which key? Static or dynamic, using a fixed CW or rotating crypto-periods using a DVB SimulCrypt-compliant ECMG? First, define precisely what you want to implement and then RTFM, plugins `scrambler` and `descrambler`.

udhayNuk, 2022-11-05:
@lelegard 
Thank u soo much dude

udhayNuk, 2022-11-07:
@lelegard 
another one doubt dude please reply me

tsp -I ip -l 192.168.20.30 233.1.1.1:1001 -P zap "NEWS" -P analyze -i 30 -o report.txt \
        -P regulate --pcr-synchronous \
        -O ip --ttl 64 -l 10.10.10.10 233.1.1.1:1111

iam using this script to stream channels, above 20 streams i have runned and played the video the video and audio was little slow motion strucking, what is the problem please solve this issue and i have runned one streams no video was struked

thank u in advance

M-Bala-2612, 2022-11-07:
Same dout me also !!!

udhayNuk, 2022-11-12:
@lelegard 
how take only one audio from udp
my input udp receives 2 audio and 1 video 
how to take only one audio to output stream is there is any possible command
please reply me thanking you

lelegard, 2022-11-12:
> how to take only one audio to output stream is there is any possible command using tsp

Please read the user's guide, reference section for the `zap` plugin, options `--audio` and `--audio-pid`.

Please note that this forum is not a replacement for reading the documentation.

udhayNuk, 2022-11-14:
@lelegard Thank you for your reply
i used --audio-pid but the audio was not played

tsp -I ip $INPUT_URL -P zap "0401 NDTV GT" --audio-pid 0x05A0 -P analyze -i 30 -o report.txt \
 -P regulate --pcr-synchronous -O hls --live 15 --duration 3 -p master.m3u8
udp was receiving one with and four audio i want take only one audio but i used audio pid 
video was playing but the audio was not playing
my udp report i have attached

  SERVICES ANALYSIS REPORT                                                   |
|=============================================================================|
|  Global PID's                                                               |
|  TS packets: 8, PID's: 3 (clear: 3, scrambled: 0)                           |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Global PID's ................................. C       12,070 b/s  |
|   Subt.  Global PSI/SI PID's (0x00-0x1F) .............. C       12,070 b/s  |
|  0x0000  PAT .......................................... C        7,544 b/s  |
|  0x0001  CAT .......................................... C        3,017 b/s  |
|  0x0011  SDT/BAT ...................................... C        1,509 b/s  |
|=============================================================================|
|  Service: 0x0004 (4), TS: 0x0087 (135), Original Netw: 0x0001 (1)           |
|  Service name: 0401 NDTV GT, provider: NDTV                                 |
|  Service type: 0x01 (Digital television service)                            |
|  TS packets: 4,361, PID's: 6 (clear: 6, scrambled: 0)                       |
|  PMT PID: 0x0054 (84), PCR PID: 0x05B4 (1460)                               |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Digital television service ................... C    6,579,451 b/s  |
|  0x0054  PMT .......................................... C        7,544 b/s  |
|  0x058C  MPEG-2 Audio (eng, Audio layer II, 128 kb/s,   C      137,292 b/s  |
|  0x05A0  MPEG-2 Audio (eng, Audio layer II, 128 kb/s,   C      137,292 b/s  |
|  0x05B4  AVC video (720x576, high profile, level 3.0,   C    6,027,266 b/s  |
|  0x05BE  MPEG-2 Audio (eng, Audio layer II, 128 kb/s,   C      135,783 b/s  |
|  0x05C8  MPEG-2 Audio (eng, Audio layer II, 128 kb/s,   C      134,275 b/s  |
|          (C=Clear, S=Scrambled, +=Shared)                                   |
=========================================================================

please reply me thank you dude

closed 2023-04-22 by lelegard