tsduck/tsduck #1104: [Q] new user here, tsp seems to create CC errors in multicast of clean TS file

geib, 2022-11-03:
I am trying to output a transport recording, which has no CC errors, as a multicast on a mac. I am trying commands like:
```
tsp --max-input-packets 7 -I file PTC-426.ts -P regulate --bitrate 41221632 -O ip 224.10.11.12:5000
```

However, the multicast stream seems to now have many CC errors. When I analyze the TS file I see no CC errors. But analyzing the multicast I see lots of CC errors.

I would appreciate any help, and suggestions.

Thanks,
Mark.

lelegard, 2022-11-04:
Hi @geib 

TSDuck does not _create_ discontinuities. If discontinuities exist in the input file, `tsp` simply propagates them, unchanged. Without input discontinuity, output packets do not contain any either. If the receiver sees discontinuities, the reason is that the receiver is not able to receive all packets. 

In your case, it is probable that the output is bursty and the receiver expects a smoother reception, without sufficiently large reception buffers to absorb the bursts.

The reason is probably similar to issue #1099. Your bitrate is high, 41 Mb/s, meaning 3915 UDP datagrams per second. If you want a smooth output, a UDP datagram must be sent every 255 microseconds. Without specific tuning, the kernel of Linux or Windows cannot suspend an application for such a small amount of time. Usually, the minimum achievable wait time is 10 ms at application level. In that case, with your bitrate, each output burst contains 39 UDP datagrams or 273 TS packets. Probably too bursty for your receiver.

To broadcast a smooth output TS at that speed, your need a specialised NIC such as those from Dektec. In that case, the application provides the TS with bursts but the hardware NIC outputs the packets at regular small intervals.

If you want to know the highest possible bitrate for applications without specialised hardware NIC and UDP output, consider that you may send one UDP datagram every 10 ms, the smallest wait interval. This means 1.05 Mb/s. Above that, bursts appear. Small bursts (2, 3, maybe 10 datagrams) are easily absorbed by most receivers. So, you may be reliable up to 10 Mb/s. Above that, you need specialised hardware (or applications performing active waits and eat 100% CPU).

geib, 2022-11-04:
So in this case the receiver is a software application which I am developing. I increased the network buffers and problem is gone. Thanks for information and detailed explanation.

closed 2022-11-04 by geib