tsduck/tsduck #1109: [Q] "stack-based buffer overrun" in sample code; on new installation

zbensal, 2022-11-08:
Hi, 

I'm new to TSDuck. Installed tsduck-3.32-2983 from binaries on Windows 11.
I am trying to run the sample-app in Visual Studio 2022. The code compiles, but running it causes the following error:
"Unhandled exception at 0x00007FFDAE9522CD (tsplugin_until.dll) in sample-app.exe: Stack cookie instrumentation code detected a stack-based buffer overrun."

This issue happens only if any of the processing plugins are present. No issue if processing plugins are commented out. Program stops at the if statement in the code shown below:
`// Start the TS processing.`
 `   if (!tsproc.start(opt)) {`
`       return EXIT_FAILURE;`
`   }`
I would appreciate your help.
Ben.

lelegard, 2022-11-09:
Hi @zbensal 

Does the application run without error in release mode (non-debug build)?

If yes, then this is a known problem with Windows where release and debug builds are sometimes incompatible. For some reason, when you build an application in release vs. debug mode, you need to use the release vs. debug version of some DLL's. It does not happen with all DLL's but this is quite frequent. This is why the "development environment" option of TSDuck installs release and debug versions of tsduck.dll, the main TSDuck library.

On the other hand, the TSDuck "tools and plugins" section is produced in release mode only. When you activate an external plugin in a debug application, you dynamically load a release DLL inside the debug exe, causing various binary inconsistencies.

Note that some frequently used or small plugins are not implemented as DLL. They are directly included into tsduck.dll and you should have no problem with them since you use a debug version of the DLL.

If you really want to debug an application which activates external plugins, you have to rebuild TSDuck in debug mode to collect debug version of all plugin DLLs. I could also ship debug versions of those DLL's in the TSDuck installer? But this installer is already so huge, 6 times larger than Linux equivalent, that this may annoy most users.

zbensal, 2022-11-09:
Thank you, @lelegard for your response. 
It is working in release mode.

closed 2022-11-09 by zbensal