tsduck/tsduck #1111: [BUG] Compilation fails with python3 with missing python

lars18th, 2022-11-09:
#### Bug description:
When compiling TSDuck with command `python3` accessible but not `python` the compilation fails at one point.

#### How to reproduce:
Inside a recent Debian Linux, install the build-essentials and python3, and not more packages.

After try to compile with
 `make NOTEST=1 NOPCSC=1 NODEKTEC=1 NOHIDES=1 NOVATEK=1 NOPCSC=1 NOSRT=1 NORIST=1 NOEDITLINE=1` 
to not require additional libraries.

Then the compilation fails when executing one `python` command. But the Python 3 package is installed. Futhermore, if you install Python 2.5, and the `python` command will be accessible, even in this case the compilation fails. All because the environment requires to use version 3 and not version 2.5.

#### Expected behavior:
It will be more useful if the compilation will call directly to `python3`. And if it's not available then failback to `python`.

#### Environment:
 - OS: Linux
 - OS version: `cat /etc/debian_version`  > 11.1
 - TSDuck full version: 3.33-3006
 - Installation type: git clone

#### Additional information:
This is not a full bug, but only a trouble. However, I feel it will be interesting to improve the compilation environment to overcome this case. You agree with that @lelegard ?

lelegard, 2022-11-09:
Could you be more specific? Where does it fail? Which command?

There are some Python scripts in the build process. In the makefiles, I tried to always use commands such as
~~~
$(PYTHON) $(SCRIPTSDIR)/cleanup.py
~~~
where `$(PYTHON)` is defined either as `python3` when available and `python` otherwise. Which is exactly the requested behaviour. If there is one missing `$(PYTHON)`  somewhere, please specify.

More generally, the sheebang of all Python scripts is `#!/usr/bin/env python` because `python3` is not available as a command on some Linux distros. They consider that Python is `python` regardless of the version. On the other hand, all distros have a way to explicitly redirect the `python` command to either Python 2 or 3. On Debian, use `update-alternatives` to do so.

If you do not want to set Python 3 as default Python interpreter, then you must run each Python scripts using `python3 script.py`. This is what the makefiles do, just in case. Please specify where the error is.

barsnick, 2022-11-09:
I have referenced the commit causing the regression in my PR #1113. I don't know whether the `#!/usr/bin/env python` needs to be fixed as well though.

By the way, does `tsduck.dektec.names` also need to be built if NODEKTEC is specified?

lelegard, 2022-11-09:
> I have referenced the commit causing the regression in my PR https://github.com/tsduck/tsduck/pull/1113.

Thanks. That is what I suspected in my previous post, one missing `$(PYTHON)` in a makefile rule.

> I don't know whether the #!/usr/bin/env python needs to be fixed as well though.

No, as explained in my previous post, this must not be changed. The sheebang is a shortcut to run the script without explicitly specifying the interpreter. There are some Linux distros where `python3` never work while there is always a way to make `python` work as Python version 3. That is why the sheebang will not change.

> By the way, does tsduck.dektec.names also need to be built if NODEKTEC is specified?

Yes, the file is always built but contains an empty section when Dektec is not supported.

barsnick, 2022-11-09:
> No, as explained in my previous post, this must not be changed. The sheebang is a shortcut to run the script without explicitly specifying the interpreter. There are some Linux distros where [...]

I personally was under the impression that every installation of python always brings a `python2` or `python3` counterpart. I don't know about Windows, but this is what I've seen on Ubuntu and Fedora for ages. On the other hand, Ubuntu dropped the default `python` alternative, and put it in a package called `python-is-pythonX`. Fedora did the same as `python-unversioned-command`.

But that said: It should not matter! As far as I understand, python(3) is used only at build time for some file generation (fixed for non-PS1 scripts), and module (`ts.py`, `tsduck.py`) are provided, where a shebang is not required.

By the way, the modules should probably be in site-packages. I'm fixing that in my local packaging right now. (I'll provide my thoughts on that when I get around to explaining the motivation.)

> > By the way, does tsduck.dektec.names also need to be built if NODEKTEC is specified?
> 
> Yes, the file is always built but contains an empty section when Dektec is not supported.

Thanks, I get it!

lelegard, 2022-11-09:
> I personally was under the impression that every installation of python always brings a python2 or python3 counterpart. I don't know about Windows, but this is what I've seen on Ubuntu and Fedora [...]

You are right, the mainstream distros do that properly; commands `python2` and/or  `python3` and a clean way to assign the default  `python` command to one of them.

But not all distros do that. To support various exotic issues, compilation or configuration, from users, I have setup 10 distinct Linux VM: Fedora, Alma (RHEL clone), Ubuntu, Debian, Mint, openSUSE Leap, openSUSE Tumbleweed, Alpine, Arch, Gentoo. At least two by the end of the list (the exotic ones with hordes of zealots) do not provide the command `python3` when installing Python 3.

Windows does not use the sheebang, `python` is always available as a command and `python3` is never defined.

So, there is no universal solution. I came up with using `python` everywhere in sheebangs because `python3` is impossible to use in some places while `python` is always possible with the right setup. The makefiles are supposed to be resistant to users not willing to setup `python` as `python3`, except the bug you found.

lelegard, 2022-11-09:
> By the way, the modules should probably be in site-packages. 

If someone can provide a _universal_ definition of "site-packages", that could be a good idea...

Depending on the distro and the type of installation, the "site-packages" is virtually anywhere on the file system, somtimes with a precise Python version in the path, system-wide, user-specific, etc. So, the installation of TSDuck installs them in a deterministic place, at `/usr/share/tsduck/python`. It is up to the user to add it in its `$PYTHONPATH`. On Windows, the TSDuck installer updates `$PYTHONPATH` at system level and the user has nothing to do.

When using 10 distinct Linux distros, 2 versions of Windows and, recently, FreeBSD, I realize that Python is the most ubiquitous language with the least universal configuration.

The same type of mess applies to native extensions. While TSDuck can implement a clean Java native interface (JNI) extension which works with all Java versions without recompilation, this was impossible with Python. You need to rebuild your extension for each micro-sub-version of Python. This is a delivery nightmare. This is why I had to develop an ugly bare-metal interface using `ctypes` and encapsulate the dirt into `tsduck.py`.

I heard once someone saying that Python was a "dirty slut in a lovely dress". Not completely wrong...

closed 2022-11-09 by lelegard

lars18th, 2022-11-09:
Hi @lelegard ,

Thank you! Now it compiles without troubles. 👍 

However, only FYI: In recent Debian Linux, if you install Python3 (`apt-get install python3`) and not Python (`apt-get install python`) then the command `python` is not available. But `python3` is available. That's because `update-alternatives` is not configured by default when you install a specific version without installing the default name.

Regards.

lelegard, 2022-11-09:
> That's because update-alternatives is not configured by default when you install a specific version without installing the default name.

This is done in `install-prerequisites.sh`. The idea of that script is to run it on a fresh Unix installation and have everything ready to build TSDuck.

lars18th, 2022-11-09:
> This is done in `install-prerequisites.sh`. The idea of that script is to run it on a fresh Unix installation and have everything ready to build TSDuck.

Yes, I know it! But my idea is to compile TSDuck _without_ using the `install-prerequisites.sh` scripts. This script is very useful, and necessary with containers. However, not every time we can use it. So for this reason I'm testing to compile without dependencies by hand. And in this way I've discovery this bug.

So in any case, thank you for your fast response. 😉