tsduck/tsduck #1114: how to create master playlist for multiple resolution in ffmpeg

udhayNuk, 2022-11-10:
how to create one master playlist for all resolutions please reply 

#!/bin/bash

if [ ! -d playlist ]; then
  mkdir playlist
fi

if [ ! -d playlist/360p ]; then
  mkdir playlist/360p
fi

if [ ! -d playlist/480p ]; then
  mkdir playlist/480p
fi

if [ ! -d playlist/720p ]; then
  mkdir playlist/720p
fi

if [ ! -d playlist/1080p ]; then
  mkdir playlist/1080p
fi

RStreamer_engine_1.5 -i "$INPUT_URL?overrun_nonfatal=1&fifo_size=50000000" \
   -s:v 640x360 -c:v libx264 -movflags +faststart -preset ultrafast -c:a aac -b:a 128k -ac 2 -b:v:0 1000k \
   -f hls -hls_time 2 -hls_wrap 15 \
   "playlist/360p/manifest.m3u8" \
   -s:v 720×480 -c:v libx264 -movflags +faststart -preset ultrafast -c:a aac -b:a 128k -ac 2 -b:v:0 2000k \
   -f hls -hls_time 2 -hls_wrap 15 \
   "playlist/480p/manifest.m3u8" \
   -s:v 1280x720 -c:v libx264 -movflags +faststart -preset ultrafast -c:a aac -b:a 128k -ac 2 -b:v:0 3000k \
   -f hls -hls_time 2 -hls_wrap 15 \
   "playlist/720p/manifest.m3u8" \
   -s:v 1920x1080 -c:v libx264 -movflags +faststart -preset ultrafast -c:a aac -b:a 128k -ac 2 -b:v:0 4000k \
   -f hls -hls_time 2 -hls_wrap 15 \
   "playlist/1080p/manifest.m3u8" 
#</dev/null > /dev/null 2>&1 & echo $! > $STREAM_PID_PATH
 
thank u in advance

lelegard, 2022-11-10:
Hi @udhayNuk 

For the benefit of all users, it is important to keep this discussion forum clean, clear and well organized. 

This issue is not clear enough. Is this a question? A response? To which question? It seems redundant with multiple issues you opened recently such as #1100, #1102, #1108 and maybe #1110 from a different user but with similar concern and style. 

Redundant posts shall be avoided, especially from the same user at the same period. Supplement the discussion of existing topics instead.

Please read the [Github documentation on how to write topics using markkdown](https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github), especially to properly embed code and commands. Mixing descriptive text and code and commands make the post extremely difficult to read and understand.

Support is provided here on a best effort basis. Developing and supporting TSDuck requires a substantial amount of effort with no compensation. Users benefits from these efforts _for free_. In return, users are expected to spend some effort in reading the documentation, understanding the product and the related standards, understand the way GitHub discussions work. If users appear to provide no effort and expect extensive support for free, there will be no response.

closed 2022-11-10 by lelegard

udhayNuk, 2022-11-10:
@ ok dude

udhayNuk, 2022-11-10:
@lelegard 
ok dude