tsduck/tsduck #1119: [FEAT] Plugin "svresync" new parameter "--touch-label-pid"

lars18th, 2022-11-13:
Hi @lelegard ,

Regarding the enhancement for the _non_changed_timestamps_ I've this suggestion:

- Add to the `svresync` plugin the parameter `--touch-label-pid 1:100,101,...` : the concept is very easy. When the changing of the timestamps starts to apply inside a pid (i.e. 100 or 101 or etc.), the label flag indicated (i.e 1) is then set for all the rest of packets in this pid.

I believe you will quickly understand the utility of this. Does this solution look simple and clean? What do you think?

_Originally posted by @lars18th in https://github.com/tsduck/tsduck/issues/1095#issuecomment-1312389418_

lelegard, 2022-11-13:
OK but the option will only take the label number, no PID. The value of the PID's to update is not under control of the user or the command line. They are discovered during execution, when the PMT of the target service is read.

For each target PID, starting from the first time a PCR, PTS or DTS is modified on that PID, all packets of that PID will get the specified label.

lars18th, 2022-11-13:
Hi @lelegard ,

I'm happy that you accept to consider this suggestion. Thank you.

And it's true, the pids to process are not necessary. However, for the processing that I want to do (that is to filter out the content of the service until it will be resynchronized) my strategy will be:
1. Mark with label 1 _all_ pids of the service that I want to process (note that although the code does not know them, I know them because they're under my control).
2. Leave the svresync plugin to apply the mark of label 2 to processed timestamps.
3. At the end check for packets with label 1 and not label 2 and remove them.

Regards.

lelegard, 2022-11-13:
Option `--set-label` has been added:
~~~
  --set-label label1[-label2]
      Set the specified labels on the modified PID's. On each PID, the label is
      first set on the first modified packet, and then on all packets of the
      PID. Several --set-label options may be specified.
~~~

lars18th, 2022-11-15:
Hi @lelegard ,

Nice! Thank you.

closed 2022-11-15 by lars18th