tsduck/tsduck #1122: [FEAT] Partial command line redirection from a file with comments

lars18th, 2022-11-15:
#### Is your feature request related to a problem?
I need to support `“@filename”` parameters (documentation section 3.1.5) with a more flexible content. Example:
```
## this is a comment
-I
 dvb
  --frequency
    12,169,000,000
  --symbol-rate
    27,500,000
  --fec-inner
    3/4
  --polarity
    horizontal
  --delivery-system
  --modulation
    8-PSK
## another comment
```
What I need is that lines starting with `#` are omitted. And initial spaces are omitted too.
I don't need anything more. With this it's sufficient to write these files as a simple sources that a human can read and edit.

#### Proposed solution:

After a short review of the code, it could be implemented without effort editing the function `ts::UString::LoadAppend()` that is called from `UString::Load()` that is used inside `ts::Args::processArgsRedirection()`.

However, I don't want to do a dirty change in the code. So I request if you @lelegard want to enhance this. Or if not, then let me to know you preference about the change:
- You accept to add to `ts::UString::LoadAppend()` a third boolean parameter to indicate if you want to omit "#" and spaces?
- Or you want to pass a list of characters to control the filtering?
- Or you have another idea?

#### Alternatives:

The only alternative is continue using the current file format that is quite difficult to edit for humans.

#### Additional information:

I hope you want/accept to improve this.

lelegard, 2022-11-15:
I must admit I do not like the idea very much. The principle of those files is to avoid all escaping issues, special characters, spaces, and all command interpreter syntax issues. Everything on each line literally belong to the parameter, including leading spaces if I remember correctly. One parameter = one line. No syntax, nothing.

By adding comments, you introduce syntactic concepts again. Leading to questions such as: how do you specify a parameter, a service name for instance, such as "#mychannel"? Do not tell me to add quotes because that would be yet another level of syntax. Double it? Yet another syntax?

Comments are nice in general and I do insist to have meaningful comments in source code. But, here, I would rather not.

lars18th, 2022-11-15:
Hi @lelegard ,

The idea is VERY SIMPLE. Only two enhancements:
1. Don't consider any line that starts with `#` (that's a comment)
2. If the line starts with one space, then remove all spaces until the first non-space character. After continue reading as normal.

So this enables "comments" and "tabulation" in such files. Nothing more.

And to implementing it, the question is this: you prefer to add a simple (optional) `bool` parameter to enable it (default false); or you want something else?

lelegard, 2022-11-16:
This is not a complex vs. easy code modification. As I explained, that would break existing command files. The net result is potential annoyance for some users vs. a small benefit for one user. So, the answer is no. Sorry.

lars18th, 2022-11-16:
Hi @lelegard ,

Perhaps I'm not expresed very well. Sorry, but my suggestion **CAN'T** generate any side effect. I'll try to explain more clearly:
- I want to modify the function `ts::UString::LoadAppend()` with a third _optional_ parameter called `enhanced_mode`. It's a bool with a default value of "false".
- Only when `enhanced_mode` is "true" the code doens't consider lines starting with "#" and initial spaces.
- Then all the current code continues compiling and working as now.
- After I want to modify the chain that process the command line parameters using "@" inside `ts::Args::processArgsRedirection()` to _only_ trigger the parsing with `enhanced_mode` as "true" when importing external files. In any other case the parsing will be done using `enhanced_mode` as "false" following the current behaviour.
- To support the previous point I need to enhance some related functions in the chain between the previous function calls. This could be done adding the parameter as optional or adding new function names with this support. Here I need to know your preferance.

So, after these changes... could you please explain which side effect could be created?
Let me to explain my use case, that perhaps could be useful to understanding the advantage of this _minimal_ change:
- Using scripts for TSDuck is a must. Every experienced users needs to use them. However, scripts are not portable. Them could be if you use the same interpreter (for example, Bash in Windows or WSL; or PowerShell in Linux). However, it could be more easy to "encapsulate" your scripts inside "include files" (the file with parameters called with "@" in the command line). However, the current strict syntax is quite difficult to read and maintain for the user. Only adding the option to comment lines and use tabs is sufficient to improve the speed as you can do tests and research. For example, compare the sharing of a list of different command lines using "tsp" with the  ending ` \` char compatible only with Bash-like shells and not Windows, with sharing a text file that you can call with a simple `tsp.exe @test-1.txt` or `tsp -v --debug @test-1.txt`.

So, are you sure that this is only a benefit for one user, when it does not disturb any other user and cannot generate unknown side effects? Also, my idea is to implement it by myself and provide a PR for the merge. So my request is just to do the coding using your recommendations to follow your coding style and not break anything.

Therefore, with all due respect, could you please reconsider your position?

lelegard, 2022-11-16:
> could you please explain which side effect could be created?

Consider a service named "#mychannel" (in the Twitter era, many marketing guys consider that renaming something as "#something" is clever). A production script contains this:
~~~
-P
zap
#mychannel
~~~
The production is now broken...

lars18th, 2022-11-16:
Hi @lelegard ,

You're right (another time). And for this reason I prefer to discuss any question _before_ implementing anything. So, I'll try to think on alternatives. I hope you agree to almost continue the discussion.

Thank you.

lelegard, 2022-11-16:
> What you think?

Too awkward, error-prone, creating two types of parameter files. Users already have problems with the double @@ (and there is no other obvious solution), see some issues here. No need to create additional confusion. For such a little benefit.

@lars18th, over the last few years, you have requested many, many enhancements. You are, by far, the most demanding user in terms of new features. Most of the time, your demands have been implemented, sometimes exceeding the request in terms of features. In return, you have to accept that, sometimes, after debate and explanations, the answer can be no. And, here, this is a final "no".

lars18th, 2022-11-16:
Hi @lelegard ,

> And, here, this is a final "no".

OK. Don't worry. I feel that a _constructive_ discussion is not useless. So I can accept your opinion. In fact it's really true that my last proposal is error-prone. So perhaps I need to use a completly different approach. I'm think on do this externally: that's creating my personal files, and then applying a pre-processing to convert to the TSDuck format, for finally calling to tsp. In fact, is not very complex: with simple shell commands I can do the work (awk?). So finally the KISS model is the winner.

Thank you and regards. 😉

closed 2022-11-16 by lars18th