tsduck/tsduck #1124: [Q] Is this possible for TSDuck to use RAM instead of Virtual Memory?

moving-digital, 2022-11-18:
Hi Thierry @lelegard 

I got two questions out of curiosity:
- Is this possible for TSDuck to use RAM instead of Virtual Memory?
- Regardless to the answer, what would be the pros and cons of TSDuck using RAM?

lars18th, 2022-11-18:
Hi @moving-digital ,

I'm not sure about this, but a fast review of the source code reveals that TSDuck has the implementation of the `tsResidentBufferTemplate` class that uses pined memory. With implemenations for Unix and Windows:
https://github.com/tsduck/tsduck/blob/3f37174f26da9d336f3ac0846a071e6189d10bd7/src/libtsduck/base/system/tsResidentBufferTemplate.h#L30-L31

What a I don't know is until which point this is used. Because this class seems to not be used at all. Perhaps the author (@lelegard) could explain something about this.

moving-digital, 2022-11-18:
Thanks @lars18th 

From my little analysis, TSDuck is indeed fully utilizing Virtual Memory only indeed:

[image]

I would prefer using the RAM though since I got enough, and could benefit from better performances.

lelegard, 2022-11-18:
I would say that this is an old fashion discussion.

First, you never use RAM _instead_ of VM. VM is virtual and mapped over RAM and page/swap file or partition. So, RAM is part of VM. In the old days, when RAM was limited to a few MB per system, the right adjustment of the working set (the part of VM which is in RAM at a given moment) could be crucial since the page fault rate could kill the performances of some applications. I played a lot with this kind of system tuning in the 80's and 90's, mostly on VAX/VMS and Alpha/VMS. And by the way, if you see the same vocabulary, "working set" et al,, in Windows, this is simply because the Windows NT kernel was redesigned from scratch by the original VMS developers, namely Dave Cutler and his team from DEC.

Nowadays, most of the time, this is an obsolete debate. Systems have GB of RAM and sometime no page/swap file at all. On Linux system, it is always recommended to have some swap partition to avoid stressing the memory management in the kernel. But it is rarely used in practice. We should consider this as obsolete memory management in the Linux kernel. This should be redesigned somehow.

In 2005, when I needed to play with TS transformation, my old system tuning reflexes prevailed and I locked the TS packet global buffer in physical memory, when possible or allowed. This later became the C++ class `ResidentBuffer` in TSDuck that @lars18th  mentioned in a post above. If I had to rewrite it from scratch today, I would simply ignore it.

Also note that, on Linux, a user application is not allowed to lock memory pages in physical memory. So, the mechanism is not used. You need to be root to lock in physical memory. But running a complex application using the root account, especially when it loads plugins and shared libraries from various origins, is strongly discouraged for security reasons. If you run `tsp` on Linux in debug mode, in a regular user account, you will see the error message saying that memory locking failed. Previously, this message was displayed in normal mode and users always complained about the error message without understanding the reasons. So, I simply hid it in debug mode.

The fact that memory locking still exists in Linux but is accessible only in conditions that modern security considerations would forbid is another indication that this is mostly an obsolete feature.

> From my little analysis, TSDuck is indeed fully utilizing Virtual Memory only indeed, 830MB * 4, but only 423 MB of RAM is used. I would prefer using the RAM though since I got enough, and could benefit from better performances.

This is an incorrect analysis. You can see from the screenshot that swap is not used at all, meaning that TSDuck runs 100% in RAM. And by the way, your display exactly illustrates what I just explained above: You have 8 GB of RAM and only 100 MB of swap. This is probably the default configuration. The swap has a ridiculously small size compared to RAM (100 MB vs. 8 GB) and exists only to avoid stressing the kernel. But it is not used _at all_ in practice ("0K/100.0M"). In the old days, when _virtual_ memory meant something, the ratio was the opposite. The page/swap space was 10 times or even 100 time _larger_ than RAM. When the swap/page space is smaller than RAM, especially in the proportion you see, this is a obvious indication that it it useless.

So, my conclusion is : just ignore it.

lelegard, 2022-11-18:
> Virtual Memory only indeed, 830MB * 4, but only 423 MB of RAM is used.

To be precise, the VM size is not "used". A part of it is unused and backed by nothing. This also typically includes code pages from executable or shared object files which are backed by these exe files.

moving-digital, 2022-11-18:
> > Virtual Memory only indeed, 830MB * 4, but only 423 MB of RAM is used.
> 
> To be precise, the VM size is not "used". A part of it is unused and backed by nothing. This also typically includes code pages from executable or shared object files which are backed by these exe files.

Thank you so much, Thierry, once again for this insightful answer! Noted.

closed 2022-11-18 by moving-digital