tsduck/tsduck #1125: [FEAT] HLS output "--no-align"

lars18th, 2022-11-23:
#### Is your feature request related to a problem?
I want to produce HLS segment files that will be a 1:to:1 representation of a CBR stream. The HLS output plugin can write to files, and the HLS input plugin can read from a file URI. So we can produce these segments to process them off-line. However, the current HLS output plugin is inserting a copy of PAT and PMT tables at the start of each segment:

[ Documentation "hls (output)" section ]
_**-a | --align-first-segment**
Note that all subsequent output segments always start with a copy of the last PAT and PMT,
on a video PES packet boundary, with or without this option._

#### Proposed solution:
Add to the `hls (output)` plugin a new option `--no-align` that will produce *clean* segments. These segments are a bit perfect representation of the TS packets, without any reordering or filtering. The generated segments could them be joined (`cat foo-000000.ts foo-000001.ts foo-000003.ts > stream-foo.ts`) to reconstruct the original stream. Or they can be used with the HLS input plugin to read them using the standard playlist `foo.m3u8`.

As an addition, it will be useful to provide an alternative mode like `--no-align-extended`. In this case the segments are the same, but for each of them another file with the missing starting PAT and PMT tables are created. Therefore for the file `foo-000000.ts` another `foo-000000.psi` is created. And this file constains the "missing" packets. So with a command like `cat foo-000000.psi foo-000000.ts > foo-000000.ts` the _standard_ chunk is obtained. This alternative mode could be used in case the standard HLS segments are necessary.

#### Alternatives:
The only alternative that I think it would be done is to post-process all segments to remove the PSI tables. However, this is error-prone because you don't know the number of TS packets occupied by these tables. Futhermore, if the original position of the tables are coincident with the starting of the segment, then you can't remove them. So the only solution is that the output plugin (segmenter) will do this task.

#### Additional information:
Because the work necessary to complete this action is small I think this functionality will be interesting. Then we can use HLS as a "chunk base fileformat". And we can use then tools for analyze and process only some parts of the full stream very easy. In fact all resulting segments are valid TS streams, because they're only splices of the original stream. So a very flexible and powerful technique with a minimal cost.

I hope then you (@lelegard) want to implement this. You agree?
Thank you.

lelegard, 2022-11-23:
This is no longer HLS, this is just TS slicing, cutting a raw TS file in chunks of multiples of 188 bytes. There are tools for that, not specifically for TS files.

HLS mandates that each segment file starts with a PAT and PMT. This is in the standard. An HLS client is supposed to play each segment as independent files. If the vast majority of segments do not start with PAT/PMT, there will be many holes in the playback.

lars18th, 2022-11-23:
Hi @lelegard ,

I know that my request is not related to the _standard_ HLS. It's only a medium to make more easy to work with TS slices. And I'm sure that you can agree that if TSDuck adds support for working with slices (or chunks, or segments, or whatever you want to call it) then it opens the oportunity to do some types of (parallel) processing that is not feasible without it.

Having said that, let me comment on these issues:
- The current implementation of the "hls input plugin" can accept segments without starting with PAT and PMT? True, right? And this is because in fact without the option of the output plugin `--align-first-segment` to enforcing the PAT and PMT at start, then the first segment _is not starting_ with these tables. So in practice the _current_ implementation of the "hls input plugin" is transparent to this. And this part of the toolkit will not be changed at all.
- What I'm requesting is only a new option to _override_ the default behaviour of including a copy of the last PAT and PMT tables at the starting of a segment. Therefore is not more work, actually less work for the segmenter.
- For sure perhaps you have another approach to achieve the same objective: work with slices. However, IMHO the KISS model dictates the most simple solution. And in this case, the reutilization of the HLS implementation seems to be an easy alternative. However, perhaps you have another different view, yes?

Anyway, as you know I don't want to enforce anything in your development. I'm only requesting what I think has sense to achieve my goals. That perhaps will be also useful for other users. So the question now is: *What about TS slices support for TSDuck?*

Please can you comment on this? I will keep an open mind.
Regards.

lelegard, 2022-11-23:
There is no need to use TSDuck for this:
~~~
$ split --verbose -b 18800000 -d -a 4 --additional-suffix=.ts in.ts out.
creating file 'out.0000.ts'
creating file 'out.0001.ts'
creating file 'out.0002.ts'
creating file 'out.0003.ts'
creating file 'out.0004.ts'
creating file 'out.0005.ts'
creating file 'out.0006.ts'
creating file 'out.0007.ts'
creating file 'out.0008.ts'
creating file 'out.0009.ts'
$
~~~
Of course, make sure that the chunk size, here 18800000, is a multiple of 188.

You can verify that the result is correct as follow:
~~~
$ cat out.* >foo.ts
$ cmp in.ts foo.ts
$ 
~~~
The last command also demonstrates that you do not need a `tsconcat` command to concatenate TS files, `cat` does it right.

It is sometimes important to remember that TS files are just files and can be manipulated by any standard system command. TSDuck is only useful when some transport stream specific manipulation is required.

So, in application to the KISS principle, don't modify the code when a standard command can do the job :)

lars18th, 2022-11-23:
Hi @lelegard ,

What you propose will work _only_ offline. Please, do this test (I've do it):

- Get any TS file (`test1.ts`) and execute: `split --verbose -b 18800000 -d -a 4 --additional-suffix .ts test1.ts out.ts`
- Then execute `tsp -I file test1.ts -O hls -f 18800000 -p master.m3u out.0000.ts` in another directory.
- And after copy the `master.m3u` to the initial directory, then execute `tsp -I hls master.m3u -O file test1-new.ts`
- Finally compare the outputs with `tscmp -v test1.ts test1-new.ts`

The result is that:
1. The HLS input reads without troubles the segments without PAT and PMT tables at starting of the segments (obviously).
2. The output using this technique is fully transparent, so 1:to:1 bit perfect copy.

And what I want to do, that I can't without this "hack", is this:
- Inject a CBR from the multicast to a TSP process to split them in chunks to a temporal directory, and with a master playlist. The idea is to have there a _ring buffer_ of various minutes.
- Process these segments with several processes.
- And using a TSP process read the master playlist (with some time of difference) to feed the full TS to a new multicast address.

As you can see this architecture has some advantages:
- The ring buffer in the temporal directory will be automatically rewrited.
- If some processing fails, the _sender_ tsp process that sends to the multicast will have all the data to continue.

I understand that perhaps you're thinking that this proposal seems to be quite complex. But it isn't at all. For sure the processes that will handle the work to process the segments will need the data of the PAT and PMT tables. But they can handle it. Futhermore, it's mandatory that the stream and bitrate is not changed at all. But it's easy to orchrestate and very robust.

And the only request that I've to put all together is one small enhacement of the HLS output plugin to _not add_ the tables. I can't do in this way: `tsp -I ip ... | split` because I need to have the updated playlist, and the ring buffer overwriting.
So please, even if you think this is stupid, it is really necessary.

I hope you want to consider it.
Regards.

lars18th, 2022-11-23:
> TSDuck is only useful when some transport stream specific manipulation is required.

Or you can view this from another point of view:
- The `timeshift` plugin requires that all the buffering is done in the same server (and using the same process). Using the "HLS timeshift" proposal you can distribute the producer and consumer in different servers/locations. Futhermore, with some full tolerance, if the filesystem is a shared storage with redundancy and the producer/consumer processes are redundant too.

lars18th, 2022-11-24:
Hi @lelegard ,

I've implemented by myself in PR #1127.
I hope you will agree to merge it.
Regards.

lars18th, 2022-11-25:
Hi @lelegard ,

Regarding this discussion, I've reviewed the RFC 8216 "HTTP Live Streaming", section 4.3.5.1 "EXT-X-INDEPENDENT-SEGMENTS":
https://www.rfc-editor.org/rfc/rfc8216#section-4.3.5.1

And what I understand from this is:
- When _all_ segments are independent, this tag can be inserted.
- When this tag is not used, then the segment could not be independent.

Futhermore, the section 3.2 "MPEG-2 Transport Streams" indicates:
https://www.rfc-editor.org/rfc/rfc8216#section-3.2
_Transport Stream Segments MUST contain a single MPEG-2 Program; playback of Multi-Program Transport Streams is not defined._

Therefore, when the output is a CBR of a MPTS, the definition of the protocol is _not_ defined. In this case, we aren't broking anything, because we're using an _undefined mode_.

And to be consistent then in the HLS output of TSDuck:
- When not using the slice mode, the playlist REQUIRES to include the `EXT-X-INDEPENDENT-SEGMENTS` tag.
- And when it's used, then this label is not included.

You agree with that?

lelegard, 2022-11-25:
Merged #1127. Added `EXT-X-INDEPENDENT-SEGMENTS`  when ` --slice-only`  is not specified.

lars18th, 2022-11-25:
Thank you @lelegard !

Only to understand your changes over my code, and not be wrong the next time:
- The overhead of the initialization of the demux class is lightweight? I ask this because you have removed the block to not do it when using the --slice-only mode.
- When the demux class is not receiving packets, because `_demux.feedPacket(*pkt);` is not called, then the function `handleTable()` is never called?

I feel I need to read the full documentation of the classes! 😉

lelegard, 2022-11-25:
> When the demux class is not receiving packets, because _demux.feedPacket(*pkt); is not called, then the function handleTable() is never called?

A demux does not work by magic. If it never receives any packet, it won't do anything.

closed 2022-11-25 by lars18th