tsduck/tsduck #1128: [FEAT] HLS output "--purge-obsolete-delay"

lars18th, 2022-11-24:
#### Is your feature request related to a problem?

This is a suggestion based on this commet in the source code:
https://github.com/tsduck/tsduck/blob/master/src/libtsduck/plugins/plugins/tshlsOutputPlugin.cpp#L389-L392

#### Proposed solution:

The idea is quite simple:
- The parameter `--purge-obsolete-delay` indicates the number of segments to maintain in the filesystem after removing them from the playlist.

Therefore, when the segment X is removed from the playlist, it will be not physically deleted until the segment X+n is purged from the playlist. Where n is the value of the parameter. In fact, this will "hide" the segments until a safe time to remove them.

#### Alternatives:

A feel this approach is the most simple to overcome the problem described in the source code. Any other automatically method could be more complex to implement it. And with this strategy the user could control when delete the files.

#### Additional information:

This is only a suggestion based on the comment, and not a personal requirement.

lars18th, 2022-11-25:
Hi @lelegard ,

This is the solution implemented in the HLS segmenter of the FFmpeg. From the documentation of this tool:

_**'hls_delete_threshold size'**
Set the number of unreferenced segments to keep on disk before hls_flags delete_segments deletes them. Increase this to allow continue clients to download segments which were recently referenced in the playlist. Default value is 1, meaning segments older than hls_list_size+1 will be deleted._

So perhaps you want to use the parameter `--delete-threshold` with a default value of `0` that is the current behaviour.
You agree?

lelegard, 2022-11-25:
Added option `--live-extra-segments`.

lars18th, 2022-11-25:
> Added option `--live-extra-segments`.

Nice! Thank you.
However, why not leave the default value `DEFAULT_LIVE_EXTRA_DEPTH` to 0 ?
I prefer 1 but the current (previous) behaviour is 0. Perhaps other users will not see the change.

lelegard, 2022-11-25:
The only visible change is that clients trying to read the oldest segment immediately after the playlist is updated will now work while they previously failed to get the segment. So, we can see this as an improvement, maybe even a bug fix.

closed 2022-11-25 by lars18th