tsduck/tsduck #1130: [FEAT] tsp: regularly overwrite the same output file

basicmaster, 2022-11-25:
#### Is your feature request related to a problem?
A longer TS pipeline with multiple plugins might run into a problem sometime, e.g. after running error-free for some days. At this point in time, it would be helpful to be able to look into a part of the TS which was processed by the pipeline, just before the problem occured.

#### Proposed solution:
To aid debugging in such case, it would be helpful if the `file` output/processing plugin was able to overwrite the *same* output file again and again (something like a circular buffer would be more elegant, but possibly more difficult). So when an issue occurs and is realized, one could have a look into the dump and e.g. use it to replay the last minutes of the TS pipeline.

In the end this would be kind of a counterpart to the `-i` option of the `file` input plugin, but for output it had always to be used together with either the `--max-duration` or the `--max-size` option.

#### Alternatives:
Use the `fork` output/processing plugin to forward the TS to a different application that splits the output in regular size/time intervals and always writes the packets to the same file. But as the necessary changes for this proposal seem to be quite small, it would be better if this could be done without the need of a different application.

lelegard, 2022-11-25:
> So when an issue occurs and is realized, one could have a look into the dump and e.g. use it to replay the last minutes of the TS pipeline.

I understand the need. However, I do not think that the proposed implementations are practical solutions.

> it would be helpful if the file output/processing plugin was able to overwrite the same output file again and again (something like a circular buffer would be more elegant, but possibly more difficult). 

Each time you rewrite the , either you recreate the file (restart with an empty file) or rewrite the same file, not truncating it, overwriting the packets "in place". In the former case, there is a risk that you crash right after recreating the file and only get a few packets. In the latter case, you get a fixed size file, let's say 10 MB (rounded to a multiple of 188). Then, what? How do you use that file? How do you know where to start? You do not know where the write pointer was.

I propose an alternative. First, the `file` packet processing plugin only supports a subset of the options of the `file` output plugin. We need to mutualize the code and support the same set of options in both. This brings the options `--max-size` and `--max-duration` in the packet processing plugin. These options create multiple files, in chunks of the stream. Second, add an option to automatically delete the oldest files and only keep N files. This way you would always get an identifiable stream corresponding to some duration you chose in the command line (file size and max number of files to retain).

lelegard, 2022-11-25:
I implemented the proposition above.

- All options in the `file` output plugin are now available in the `file` packet processing plugin.
- The option `--max-files` has been added to the two plugins.

Sample command:
~~~
tsp -I ... -P file foo.ts --max-size 8000000 --max-files 4 .... -O ...
~~~
The packets are saved in files `fooXXXXXX.ts` with 8MB or more per file and only the last 4 files are kept. Thus, you have a reliable snapshot of the current state of the stream.

Keep in mind that all plugins execute asynchronously in distinct thread. If the `-P file` is before the plugin which crashes the system, it may have processed much more packets than `-P file`. This means that the cause of the problem may be well behind the end of the last file. For this reason, you must keep more than the size of the global buffer (16 MB by default). Conversely, if you place `-P file` after the plugin which causes the problem, you won't see the culprit sequence in the file.

lars18th, 2022-11-26:
Hi @lelegard ,

Regarding the last refactoring of the output and input plugins, I've a suggestion. Note this:

_OUTPUT_:
```
    args.help(u"max-size",
              u"Specify a maximum size in bytes for the output files. "
              u"When an output file grows beyond the specified limit, it is closed and another one is created. "
              u"A number is automatically added to the name part so that successive output files receive distinct names. "
              u"Example: if the specified file name is foo.ts, the various files are named foo-000000.ts, foo-000001.ts, etc.\n\n"
              u"If the specified template already contains trailing digits, this unmodified name is used for the first file. "
              u"Then, the integer part is incremented. "
              u"Example: if the specified file name is foo-027.ts, the various files are named foo-027.ts, foo-028.ts, etc.\n\n"
              u"The options --max-duration and --max-size are mutually exclusive.");
```

_INPUT_:
```
    args.help(u"",
              u"Names of the input files. If no file is specified, the standard input is used. "
              u"When several files are specified, use '-' as file name to specify the standard input. "
              u"The files are read in sequence, unless --interleave is specified.");
```

Now, we can generate using the output plugin the same slices as with the HLS --slice-only option. However, with the new input implementation is not easy to describe all the files. We can do a join with the cat command. However, this implies to generate a new large file. Therefore, in case of not using HLS encapsulation, and only file spliting, then in this case a simple method to load _consecutive_ files, in the form: foo-027.ts, foo-028.ts, etc. is not available at time.

I suggest then to add a new option like `--filenames-are-ranges` to use the filename `foo-027.ts` as the stating file of the files and increment the 027 for each file until no more to read. Or indicate an end filename with `--filename-sequence-ends 032`.

I feel this will be more complete and consistent.
What you think?

lelegard, 2022-11-26:
On output, the files are created in strict alphanumerical order, either with sequence numbers (with zero padding) or dates. This is done on purpose so that shell wildcards apply. Using `tsp -I file foo-*.ts`is the same as concatenating the files first.

lars18th, 2022-11-27:
> Using `tsp -I file foo-*.ts` is the same as concatenating the files first.

Nice! I hope you want to document this is a some form (for example in the _output_ args help of `--max-size`, and/or in the _input_ args help).

Regards.

lelegard, 2022-11-27:
> >     Using tsp -I file foo-*.ts is the same as concatenating the files first.
> Nice! I hope you want to document this is a some form

Come on ! The wildcards have been a shell feature for 50 years, no need to document this.

lars18th, 2022-11-27:
Hi @lelegard ,

> Come on ! The wildcards have been a shell feature for 50 years, no need to document this.

You know it, but IMHO not all people got it. And I'm including it. So...

```
    args.help(u"",
              u"Names of the input files. If no file is specified, the standard input is used. "
              u"When several files are specified, use '-' as file name to specify the standard input. "
              u"Use wilcards (foo-*.ts) to read chunk files with sequences numbers. "
              u"The files are read in sequence, unless --interleave is specified.");
```

Thank you for this impresive and robust project!

basicmaster, 2022-12-11:
@lelegard Thank you for harmonising the packet/output plugin and for adding the `--max-files` parameter. This is definitely a better and more flexible solution for the described issue.

I tested with `tsp -v (...) -O file --max-duration 10 --max-files 3 test.ts` and noticed that during runtime up to 4 (instead of 3) files are present in the folder. Just when I abort via Ctrl+C, the oldest file gets deleted, so that there are the requested 3 files at the end (I would assume that a non-elegant exit would leave 4 files though):

```
* file: creating file test-20221211-223644.ts
* file: creating file test-20221211-223654.ts
* file: creating file test-20221211-223704.ts
* file: creating file test-20221211-223714.ts
* file: deleting obsolete file test-20221211-223644.ts
* file: creating file test-20221211-223724.ts
* file: deleting obsolete file test-20221211-223654.ts
* file: creating file test-20221211-223734.ts
* file: deleting obsolete file test-20221211-223704.ts
* file: creating file test-20221211-223744.ts
* file: deleting obsolete file test-20221211-223714.ts
* file: creating file test-20221211-223754.ts
* file: deleting obsolete file test-20221211-223724.ts
* file: creating file test-20221211-223804.ts
* file: deleting obsolete file test-20221211-223734.ts
* file: creating file test-20221211-223814.ts
^C* tsp: user interrupt, terminating...
* file: deleting obsolete file test-20221211-223744.ts
```

Of course the more TS history you have, the better. But nevertheless I would have expected that the limit of 3 is respected all the time, e.g. if chosen to not exceed space. Is this behaviour intended?

lelegard, 2022-12-12:
This is the intended behaviour. When you ask for "3 files", you mean you ask for the amount of history corresponding to 3 files. If we delete the 1st file at the time the 4th one is created and then the process crashes, we will only get the content of 2 history files (2d and 3rd), the 4th one just being created and containing almost nothing. This is why the 1st file is deleted only when the 4th file is properly closed (Ctrl-C is a user's request which properly closes everything).

basicmaster, 2022-12-19:
Ah, I see; this makes sense to me.

The option documentation here says "When the number of created files exceeds the specified number, the oldest files are deleted.", which led to my assumption that there would never be more than three files anytime.

---
To not accumulate any (small) delays over time, I propose to change

https://github.com/tsduck/tsduck/blob/dfe8b2c4bddd2e1de15aa8c0ec41554cf80ac722/src/libtsduck/dtv/transport/tsTSFileOutputArgs.cpp#L242

to:

```
            	if (_next_open_time == Time::Epoch)
            	    _next_open_time = Time::CurrentUTC();
                _next_open_time += _max_duration * MilliSecPerSec;
```

(or to initialize `_next_open_time` with `Time::CurrentUTC()` at an appropriate location, instead of the `if`)

---
Regarding the split upon a certain amount of time elapsed, it would also be interesting to have the split upon certain regular points in time e.g. every full hour. So e.g. if you started the workflow at 17:12:34, the first file would start at 17:12:34 as well, but the following files would start at 18:00:00, 19:00:00, 20:00:00 and so on. This would allow to navigate in one of the files and to easily translate the relative timecode within the file to an absolute point in time (doesn't apply to the first file, of course).

Technically this would mean that a split would always happen when `$seconds_since_midnight % $interval == 0` with `$interval = 3600` in this example.

lelegard, 2022-12-20:
> The option documentation here says "When the number of created files exceeds the specified number, the oldest files are deleted.", which led to my assumption that there would never be more than three files anytime.

The exact wording could be "When the number of _closed_ files exceeds the specified number" but it sounds too weird. This is a user's guide, not an ISO standard. Therefore, I will leave it as it is.

> To not accumulate any (small) delays over time, I propose to change

Correct. This is what I usually do in similar situations. I don't know why I did it this way in that specific plugin.

> This would allow to navigate in one of the files and to easily translate the relative timecode within the file to an absolute point in time

Not really. The buffering system in `tsp` is done to decorrelate the processing of each plugin and absorb the delay which could be incurred in one of them (especially in multi-core systems). To also reduce the multi-threading overhead, the amount of buffered packets can be high before they are passed to the next plugin. Depending on whether you want high throughput or low latency (the two being contradictory), you need to play with the buffering tuning options. Also add the fact that `tsp` can work indifferently in offline or realtime modes. When you combine all of these, in the general case, there is no precise correlation between the time ("wall-clock time") an output file is created and the corresponding time code ("program reference clock") in the stream. Giving the false impression that a file named `out-18-00-00.ts` exactly starts at time code "18:00:00" will generate a flow of questions here when people realize that this is not exactly true.

closed 2023-05-21 by github-actions[bot]