tsduck/tsduck #1131: [FEAT] HLS output "--hide-bitrate"

lars18th, 2022-11-25:
#### Is your feature request related to a problem?
When using the parameter `--no-bitrate` with the HLS output plugin the playlist don't include the duration of the segments.

Without the parameter:
```
#EXTINF:10.000,
#EXT-X-BITRATE:22000
out.0000.ts
```

With the parameter:
```
#EXTINF:0.000,
out.0000.ts
```

And what we want is:
```
#EXTINF:10.000,
out.0000.ts
```

#### Proposed solution:

The solution is a parameter to only "hide" the tag `#EXT-X-BITRATE`, but continue calculating the bitrate.

#### Alternatives:

The only alternative to consider is a post-processing of the playlist. But this is complex when using live events.

#### Additional information:

The tag `#EXT-X-BITRATE` is optional. But duration is widely used.

lelegard, 2022-11-25:
No need to create a new option. I consider that setting the duration to zero when the bitrate is not set is a bug. I fixed it in commit 50718ab8af84681219576469162f8f9a0bc86a6e.

lars18th, 2022-11-26:
OK, thank you. I'll check it.

Regards.

closed 2023-01-06 by lelegard