tsduck/tsduck #1133: [Q] how to get only the PCR SYNCHRONOUS output log in a file

udhayNuk, 2022-11-28:
@lelegard 
tsp -v -m --synchronous-log -I ip 233.1.1.1:1001 -P regulate --pcr-synchronous -P history --suspend-packet-threshold 60 -o file -O ip 233.1.1.1:0134 2>log

0: PID 8191 (0x1FFF) first packet, clear
7: PID 5001 (0x1389) first packet, clear
21: PID 6001 (0x1771) first packet, clear
82: PID 5001 (0x1389), PES stream_id is 0xE0 (Video 0)
175: PID 6001 (0x1771), PES stream_id is 0xC0 (Audio 0)
244: PID 0 (0x0000) first packet, clear
244: PAT v1, TS 0x0000
246: PID 17 (0x0011) first packet, clear
246: SDT Actual v1, TS 0x0000
247: PID 3001 (0x0BB9) first packet, clear
247: PMT v0, service 0x07D1
175: PID 6001 (0x1771) suspended, clear, service 0x07D1
251: PID 6001 (0x1771) restarted, clear, service 0x07D1
332: PID 6001 (0x1771) suspended, clear, service 0x07D1
411: PID 6001 (0x1771) restarted, clear, service 0x07D1
244: PID 0 (0x0000) suspended, clear, service 0x0000
581: PID 0 (0x0000) restarted, clear, service 0x0000
246: PID 17 (0x0011) suspended, clear, service 0x0000
582: PID 17 (0x0011) restarted, clear, service 0x0000

iam getting this type of log in output

i want only the PCR SYNCHRONOUS output log in a file
tell me which command to use in that command line
please help me 
thank you in advance

lelegard, 2022-11-28:
Two initial comments:
- You do not get these lines in the log, you get them in the file named `file` because you specified `-P history ... -o file`.
- Using `--suspend-packet-threshold 60` is not recommended. Please RTFM. Here, 60 means 60 TS packets, not 60 seconds. Considering that a PID is suspended after 60 packets is not realistic. Do not specify this option, keep the default value, unless you know exactly what you want.

Then regarding your question,  what does _"i want only the PCR SYNCHRONOUS output log in a file"_ mean?

udhayNuk, 2022-11-29:
@lelegard 
thank your for your reply
(Then regarding your question, what does "i want only the PCR SYNCHRONOUS output log in a file" mean?)
this means i want the output log 
 i want output log only the error occured when my command line was running 
audio and video synchronous error occured

please clear my doubt
thank you in advance

lelegard, 2022-11-29:
> i want output log only the error was occured when my command line was running audio and video synchronous error occured

I am sorry but this is not much more clear. I still have no idea of what you are looking for.

Also note that using `regulate` with `ip` input is useless at best, counter-productive at worst. The `regulate` plugin is used to regulate an unregulated input such as `file` (where packets arrive at the speed of disk). The `ip` input is a regulated input, the packets arrive at the playout speed (as well as input plugins `dvb`, `dektec` and most of the time `srt` and `rist`).

lelegard, 2022-11-29:
It seems to be a shell question, not a TSDuck question. When you run the command, `tsp ... 2>log`, all errors and logs are redirected to the file named `log`. If you want to see them on the terminal, just remove `2>log`.

udhayNuk, 2022-11-29:
@lelegard 
Thank you for your kind response
tsp -v -m --synchronous-log --verbose --debug -I ip -r 233.1.1.1:1001 -P pcrbitrate -O ip 233.1.1.1:0134 2>log
when i using this script no error showing in log like this (PES packet mismatch,Non-monotonous DTS in output stream,Timestamps are unset in a packet for stream) only showing this errors 

* [MON] 2022/11/29 15:39, resource monitoring started
* ip: joining multicast group 233.1.1.1:1001 from local address 192.168.0.103
* ip: joining multicast group 233.1.1.1:1001 from local address 192.168.5.36
* ip: unknown initial input bitrate
* pcrbitrate: new bitrate from PCR analysis: 5,000,369 b/s
* [MON] 2022/11/29 15:39, VM: 457 MB (+210 MB), CPU:8.20% (average:8.20%)
* [MON] 2022/11/29 15:40, VM: 457 MB (stabilizing), CPU:8.20% (average:8.20%)
* [MON] 2022/11/29 15:40, VM: 457 MB (stabilizing), CPU:8.10% (average:8.16%)
* [MON] 2022/11/29 15:40, VM: 457 MB (stabilizing), CPU:8.20% (average:8.17%)
* [MON] 2022/11/29 15:40, VM: 457 MB (stabilizing), CPU:8.10% (average:8.15%)
* [MON] 2022/11/29 15:40, VM: 457 MB (stabilizing), CPU:8.30% (average:8.18%)

i want Async and vsync Report or error when running this script no error was showing

please reply me and clear my doubt

lelegard, 2022-11-29:
> PES packet mismatch,Non-monotonous DTS in output stream,Timestamps are unset in a packet for stream

To my knowledge, TSDuck does not report such errors. There must be a confusion with some other tools.

udhayNuk, 2022-11-29:
@lelegard 
thank you for your reply
tsp -v -m --synchronous-log -I ip 233.1.1.1:1001 -P pcrbitrate -o file -O ip 233.1.1.1:0134 2>log
where any error instead of that while using that command
when iam running that command line after 24 hours my video was struked so i requested you what are the errors 
any command was missed for my command line please solve
kindly solve my doubt
thank you in advance

udhayNuk, 2022-11-30:
@lelegard 
please reply me
tsp -v -m --debug=1 -I ip 233.1.1.119:1001 -P pcrbitrate -P regulate --pcr-synchronous -P scrambler --cw $KEY --dvb-cissa 25 --log-protocol=1 -O ip 233.1.1.1:0134 2>log

what are the errors occured during audio and video synchronization
tell me that errors
during running my command line what are the errors occured i want that in output log.

lelegard, 2022-11-30:
Please read the user's manual. I am willing to help people who invest some effort into reading the documentation and understand the standards. 

- Already mentioned above, do not use the plugin `regulate` with `ip`, which is an already regulated input.
- In `scrambler`, the option `--log-protocol` applies to the DVB SimulCrypt protocol with the ECMG. You do not use any ECMG, you use a fixed key.
- Nobody understand what _"what are the errors occured during audio and video synchronization"_ means in that context. TSDuck is not a MUX, it does nothing to synchronize the audio and video and does not analyze this synchronization. No such error is reported by TSDuck.

udhayNuk, 2022-11-30:
@lelegard 
thank you for your reply 
i will correct my command line

udhayNuk, 2022-12-01:
@lelegard 
please reply me dude
tsp -v -m --debug=1 -I ip 233.1.1.119:1001 -P pcrbitrate -P scrambler --cw $KEY --dvb-cissa 25 --log-protocol=1 -O ip 233.1.1.1:0134 2>log
command for how to get pcr-sync log in this command line
please clear my issue 
Thank you in advance

lelegard, 2022-12-01:
> command for how to get pcr-sync log in this command line

Again, and for the last time, I have no idea of what you mean by "get pcr-sync log".

> please reply me dude

This is an free and open source project which is supported only on spare time and good will. Nobody is committed to reply, especially to repeated unclear questions.

udhayNuk, 2022-12-01:
@lelegard 
ok sorry dude
another doubt please reply me
 tsp -v -m --debug -I ip 233.1.1.119:1001 -P pcrbitrate -P aes --cbc --key 4d7145d5cef6dd22512a1abf3522fbb7 -O ip 235.1.1.1:1011

iam using this command line to scramble but my video was playing
using aes pluging how to scramble and descramble
thank you in advance