tsduck/tsduck #1135: Open multiple ECM streams within single ECM_channel_id

kamari-swami, 2022-11-30:
Hi there, 

Here is what is happening:

```
[root@rhel9dev release-x86_64-rhel9dev]# tsp -I file astra-10847V-2020-08-24.ts --infinite -P scrambler PORTADA --ecmg 192.168.1.62:2222 -d 5 --super-cas-id 0x07001001 --access-criteria 0123456789 > /dev/null &
[1] 139485
[root@rhel9dev release-x86_64-rhel9dev]#
[root@rhel9dev release-x86_64-rhel9dev]# tsp -I file astra-10847V-2020-08-24.ts --infinite -P scrambler PORTADA --ecmg 192.168.1.62:2222 -d 5 --super-cas-id 0x07001001 --access-criteria 9876543210 --stream-id 2 -i 2 > /dev/null &
[2] 139666
[root@rhel9dev release-x86_64-rhel9dev]# * Error: scrambler: unexpected response from ECMG (expected channel_status):
    channel_error (ECMG<=>SCS)
    protocol_version = 0x02
    message_type = 0x0005
    ECM_channel_id = 0x0001
    error_status = 0x0013 (ECM_channel_id value already in use)
```

My tsecmg is running on server 192.168.1.62 and tsp with scrambler plugin is running on 192.168.1.16.

I opened new channel (default channel_id) and default stream_id.
When i try to open new stream_id within same channel_id i get above error.
As per DVB Simulcrypt protocol:
1. There can be multiple streams (with unique stream_id) within a given channel_id --> This is giving error.
2. There can be only one channel_id on TCP connection --> This is not giving error, i am able to open multiple channel_id with same tsecmg server e.g.: below command works fine:
[```
root@rhel9dev release-x86_64-rhel9dev]# tsp -I file astra-10847V-2020-08-24.ts --infinite -P scrambler CONTRATO --ecmg 192.168.1.62:2222 -d 5 --super-cas-id 0x07001001 --access-criteria 9876543210 --channel-id 3 > /dev/null &
[3] 144915
[root@rhel9dev release-x86_64-rhel9dev]#
[root@rhel9dev release-x86_64-rhel9dev]# tsp -I file astra-10847V-2020-08-24.ts --infinite -P scrambler CONTRATO --ecmg 192.168.1.62:2222 -d 5 --super-cas-id 0x07001001 --access-criteria 9876543210 --channel-id 4 > /dev/null &
[4] 144994
[root@rhel9dev release-x86_64-rhel9dev]# tsp -I file astra-10847V-2020-08-24.ts --infinite -P scrambler CONTRATO --ecmg 192.168.1.62:2222 -d 5 --super-cas-id 0x07001001 --access-criteria 9876543210 --channel-id 5 > /dev/null &
[5] 145025
```

Am i missing something here?

I want to have multiple streams within a given channel (just like commercial MUX/SCS will do..).

Thanks, kamari

lelegard, 2022-11-30:
> I want to have multiple streams within a given channel (just like commercial MUX/SCS will do..).

There is a misunderstanding here.

A _channel_ identifies one TCP connection. Each TCP connection shall have only one unique channel_id. Two simultaneous TCP connections on one ECMG cannot have the same channel_id. This is the case you run into. Two TCP connections from two clients connect to the same ECMG and declare the same channel_id. This is an error and the second one is rejected with error status 0x0013. This is the normal behaviour and I doubt than any commercial ECMG allows that. And if one allows that, this is a bug in the ECMG.

You have to understand that rejecting two simultaneous TCP connections with the same channel_id is important for the redundancy in some configurations. This is true in all DVB SimulCrypt protocols. The protocol for which it is even more important is EMMG/PDG <=> MUX. Typically, two identical redundant clients (one being able to take over in case of failure of the other one) use the same channel_id (on purpose). The first one is accepted and does the job. The second one connects and is rejected with "channel_id value already in use". Then, it periodically tries to reconnect again and again, endlessly. If, at some point, the connection succeeds, this means that the main client crashed or stopped and the new connection acts as a replacement. The failed clients will restart, try to reconnect, fail with "channel_id value already in use" because the second one took over. The first one naturally becomes a backup, endlessly trying to connect with the channel_id until it is accepted again. And so on...

Now, the standard specifies that you may have several _streams_ inside one channel. This means that one client can multiplex several streams inside the _same TCP connection_. The stream_id is used to identify each stream of data inside the same channel, the same TCP connection.

In short:
- Distinct TCP connections must have distinct channel_id.
- Distinct streams inside the same TCP connection must have distinct stream_id.
- Two streams inside two distinct TCP connections may have the same stream_id. They are still distinct streams because a stream is a subcomponent of a channel.

You do the exact opposite:
- Distinct connections have the same channel_id.
- Two streams from distinct connections have distinct stream_id.

Typically, a commercial SCS is a central equipment and may manage the encryption of many services using one single TCP connection to the ECMG, and consequently one channel_id. Each service, or more precisely each _scrambling group_, is handled in one stream_id.

TSDuck is not a commercial equipment. This is a lab toolkit to test and debug equipment integration or other demo or test situations. The `scrambler` plugin can scramble only one service and directly acts as an SCS, without going through a central SCS system. In such test configuration, each `scrambler` plugin is an individual SCS able to manage only one service. If several of them connect to the same ECMG (whether it is the test tool `tsecmg` or a commercial ECMG from an renown CAS vendor), they are considered as distinct clients and they must use distinct channel_id values.

kamari-swami, 2022-11-30:
Hi Thierry,

---------------first point-------------------
ts-101197 states:

> There is always one (and only one) channel per TCP connection.

It is different (and/or even incorrect) to say: 

> Each TCP connection shall have only one **_unique_** channel_id. Two simultaneous TCP connections on one ECMG cannot have the same channel_id

The standard limits one (and only one - even if other channels are having different) channel_id per TCP socket connection. Since the tool allows to open two channel (with different channel_id), this is what i referred to in my original comment (apologies if it was not clear enough).

---------------second point-----------------
Thanks for this explanation:

> TSDuck is not a commercial equipment. This is a lab toolkit to test and debug equipment integration or other demo or test situations. The scrambler plugin can scramble only one service and directly [...]

I truly want to use TSDuck to test and debug my ECMG to check whether it can handle the ECM load (as i envision it to support) without opening multiple channel_id.

Is there a way (or work-around) that you can suggest on how can i test / general 8000 ECM streams?
is there a way i can create channel class and add/attach streams to the channel_class separately?

kamari-swami, 2022-11-30:
Every new channel connection consumes around ~15% of my 6-core VM which limits me to open 6 channel_id's.
Do you think if i use a shorter .ts input file, i will consume less cpu?

lelegard, 2022-11-30:
> Every new channel connection consumes around ~15% of my 6-core VM 

Do you mean scrambling load or ECMG load?

Just testing on a Ubuntu VM running on a core i7 laptop. Using a 3.3 Mb/s service: scrambling using DVB-CSA2 uses 24% of the CPU, scrambling using DVB-CISSA (AES) uses 4% of the CPU. This is the expected ratio. DVB-CSA is notoriously (and purposely) slow in software while AES is quite fast.

Note that the commands you mention in the top post are not realistic on a scrambling standpoint. You use an unregulated input, a file. So, the source has an "infinite" bandwidth, only limited by the I/O and CPU capability. I would even anticipate 100% CPU load per service. To test in real conditions, you have to add `-P regulate` before the scrambler.

> Is there a way (or work-around) that you can suggest on how can i test / general 8000 ECM streams?

The load on the ECMG is very low compared to the scrambling load. The scrambling load is thousands of times higher than the ECMG load. And the higher the bitrate of the service, the higher scrambling load of course.

If you want to push the ECMG load to the limit using real scrambling, you will create an unmanageable scrambling load. In your case, scrambling 8000 average services with DVB-CISSA would require 400 to 500 i7 cores.

So, pushing an ECMG to that limit is not feasible using real scrambling, neither tsp nor physical commercial scramblers ($$$).

For that kind of test, you need to develop a special test application where you open one or more connections to the ECMG with many streams inside each connection. In the real world, a scrambler needs one ECM every 10 seconds per service. With 8000 services, you need to request a global load of 800 ECM/second. Of course, for that test, you request the ECM's but you never use the returned ECM's and your never scramble anything. On an ECMG perspective, only the ECM request load is important, not what you do with the returned ECM's.

So, you need to develop a C++ application using the TSDuck library.

Simple solution: Use the class `ts::ECMGClient` to connect to the ECMG. Its method `generateECM()` is easier to use. It synchronously requests an ECM. See the source code for `tsgenecm` for a simple example. However, this class is simple and limited to one stream per channel. So, you need to instantiate 8000 times this class. Of course, you may distribute the load over multiple instances of the test application, running on several systems if necessary.

Better solution: Use the class `ts::tlv::Connection` and TLV messages. You have to manipulate all ECMG<=>SCS messages yourself but you can open as many streams as you want per channel / connection. More generic but less easy.

kamari-swami, 2022-11-30:
Hi Thierry,
Awesome and thank you for `-P regulate` option. My workload has almost gone to 0% from ~20% for single instance of tsp for scrambling 1 service. When i run 7 scrambling instances, now i only need ~7% of CPU -- this is great. 

I will try to go for 2nd method `ts::tlv::Connection` as it is closer to my requirement - multiple streams within same channel_id.
if i fail to do so, then my plan-B is to create 8000 channels to simulate the load on ECMG side.

Thanks a ton for all your help!

lelegard, 2022-11-30:
> plan-B is to create 8000 channels to simulate the load on ECMG side

In that case I would recommend to use super-low bitrate services, less that 1kb/s. That will significantly reduce the scrambling load. And of course use DVB-CISSA, not DVB-CSA.

closed 2022-11-30 by kamari-swami