tsduck/tsduck #1136: [Q] how to descrambled the scrambled content in vlc player in Android Box ??

udhayNuk, 2022-12-01:
@lelegard 
I have scrambled a stream in ATIS/ISDA method using tsduck 
tsp -v -m --debug -I ip 233.1.1.119:1001 -P pcrbitrate -P scrambler --cw 0524584d02e65865bdcfb404f700e3be --atis-idsa tune 6 music -O ip -l 127.0.0.1 233.1.1.1:1234

So can you suggest method how to descrambled the scrambled content in vlc player in Android Box ??

please reply me 
Thank you in advance

lelegard, 2022-12-01:
> So can you suggest method how to descrambled the scrambled content in vlc player in Android Box ??

I have no idea. You know the scrambling algorithm and key, then it becomes a VLC / Android question.

udhayNuk, 2022-12-01:
@lelegard 
thank you for replying
Is there any simple encryption solution available for multicast streaming for udp to udp 
Without using scrambler  parameter ??

udhayNuk, 2022-12-05:
@lelegard 

> @lelegard thank you for replying Is there any simple encryption solution available for multicast streaming for udp to udp Without using scrambler parameter ??
@lelegard 
please reply me