tsduck/tsduck #1137: [Q] Not getting ECM_response with external SCS client

kamari-swami, 2022-12-02:
Hi There,
I am writing a python based SCR that does the following:
1. Open a channel
2. Open a stream
3. Send CW_provision
4. receive ecm_response and do nothing with it.
5. Send new cw_provision message

I am able to successfully open the channel and stream and even send cw_provision message. However, i do not get ecm_response from tsecmg plugin (`tsecmg -v -d9`).

Below it the cw_provision message that i am sending through my python client (with no ecm_response).

```
SIMULCRYPT Protocol, Interface: ECMG <-> SCS
    Header, Length: 5 bytes
        Version: 0x02
        Message Type: CW_PROVISION (0x0201)
        [Interface: ECMG <-> SCS]
        Message Length: 61 bytes
    Message containing TLV parameters, Length: 61 (bytes)
        Parameter: Type=ECM_CHANNEL_ID, Value Length=2 (bytes), Value=0x0000
            Parameter Type: ECM_CHANNEL_ID (0x000e)
            Parameter Length: 2 bytes
            ECM channel ID: 0
        Parameter: Type=ECM_STREAM_ID, Value Length=2 (bytes), Value=0x0001
            Parameter Type: ECM_STREAM_ID (0x000f)
            Parameter Length: 2 bytes
            ECM stream ID: 1
        Parameter: Type=CP_NUMBER, Value Length=2 (bytes), Value=0x0003
            Parameter Type: CP_NUMBER (0x0012)
            Parameter Length: 2 bytes
            CP number: 3
        Parameter: Type=CP_DURATION, Value Length=2 (bytes), Value=0x07d0
            Parameter Type: CP_DURATION (0x0013)
            Parameter Length: 2 bytes
            CP duration: 2000 (200000 ms)
        Parameter: Type=ACCESS_CRITERIA, Value Length=5 (bytes), Value=0x9876543210
            Parameter Type: ACCESS_CRITERIA (0x000d)
            Parameter Length: 5 bytes
            Access criteria: 9876543210
        Parameter: Type=CP_CW_COMBINATION, Value Length=10 (bytes), Value=0x00032364706c1650b5c4
            Parameter Type: CP_CW_COMBINATION (0x0014)
            Parameter Length: 10 bytes
            CP CW combination: 00032364706c1650b5c4
        Parameter: Type=CP_CW_COMBINATION, Value Length=10 (bytes), Value=0x00047012c9b7f7f21a19
            Parameter Type: CP_CW_COMBINATION (0x0014)
            Parameter Length: 10 bytes
            CP CW combination: 00047012c9b7f7f21a19
```
After receiving 3 such cw_provision messages (without getting any response), tsecmg gives 'CHANNEL_ERROR' (which is very strange) on 4th cw_provision and closes the channel. The error that i receive is:

```
SIMULCRYPT Protocol, Interface: ECMG <-> SCS
    Header, Length: 5 bytes
        Version: 0x02
        Message Type: CHANNEL_ERROR (0x0005)
        [Interface: ECMG <-> SCS]
        Message Length: 18 bytes
    Message containing TLV parameters, Length: 18 (bytes)
        Parameter: Type=ECM_CHANNEL_ID, Value Length=2 (bytes), Value=0x0000
            Parameter Type: ECM_CHANNEL_ID (0x000e)
            Parameter Length: 2 bytes
            ECM channel ID: 0
        Parameter: Type=ERROR_STATUS, Value Length=2 (bytes), Value=0x0002
            Parameter Type: ERROR_STATUS (0x7000)
            Parameter Length: 2 bytes
            Error status: Unsupported protocol version (2)
        Parameter: Type=ERROR_INFORMATION, Value Length=2 (bytes), Value=0x0000
            Parameter Type: ERROR_INFORMATION (0x7001)
            Parameter Length: 2 bytes
            Error information: 0000
```
If i use tsp scrambler plugin, i can get ECM responses.

1. Is there anything wrong with the above cw_provision message format (i have check multiple times and i couldn't figure out what is wrong with the format above)?
2. Can tsecmg connect with any external SCS (i believe yes,) and if so, why it is not responding with ecm_response for my cw_provision messages?
3. Any pointers where i should look at? As i said, channel_setup and stream_set messages are processed by tsecmg successfully.

lelegard, 2022-12-02:
The error status is explicit:
~~~
 Error status: Unsupported protocol version (2)
~~~
However this is a bit strange. The current protocol version in the latest DVB SimulCrypt spec is 3. However, there is no functional difference between versions 2 and 3 in the ECMG <=> SCS protocol. By experience, many ECMG still use version 2 and refuse version 3. This is why TSDuck uses 2 by default but there is an option to set it to 3 when necessary.

You mention that this ECMG works with TSDuck. Did you specify version 3 of the protocol?

Try to update your Python code with version 3.

Otherwise, there could be a bug in your Python code, the messages have an issue and the ECMG has problems to recover and reports an incorrect error. Try to compare the binary dump of TSDuck messages and your messages using Wireshark. Manually analyze the binary dumps if the SimulCrypt dissector does not show any difference. Sometimes, the devil is in the tiny details.

lelegard, 2022-12-02:
Do you use a commercial ECMG or TSDuck tsecmg test tool?

kamari-swami, 2022-12-02:
Hi Thierry,
1. i am using tsecmg from TSDuck. No commercial ECMG (yet) available for testing.
2. When i use tsecmg (from TSDuck) with `tsp -I scrambler` plugin, i use default protocol version (which, as per documentation, is 2. In my wireshark hex dump, i can see protocol v2 when working with scrambler plugin.
3. I have compared HEX dumps and both cw_provision messages (one from scrambler plugin of TSDuck and one from my python code) and both are same (except for CW's which are generated randomly ...

I will try the following:
1. Compare binary dumps to see if there is any difference.
2. I will try to use working hex-dump (generated using scrambler plugin) into my code and see it this works...
will update you shortly.

kamari-swami, 2022-12-02:
Hi Thierry,
It worked!
Here is what happened:
1. I re-checked all my messages (channel setup, stream setup etc..) generated by my python tool. channel_setup was perfect. stream_setup had an extra byte appended to the end (circled below):
[image]
2. Extra byte is ignored and stream_status message is sent back by tsecmg to my python client.
3. Since this is an extra byte, tsecmg is still waiting for remaining bytes to be sent and stream_setup step was not concluded.
4. When tsecmg received cw_provision, it sees them as continuation to stream_setup message and returns channel_error as the message is not stream_setup.

I checked the code, fixed that extra byte, now everything works as charm....

lelegard, 2022-12-02:
> I checked the code, fixed that extra byte, now everything works as charm....

Good.

In fact, because TCP is a stream-oriented transport, the extra 0x60 was considered as the first byte of the next message, which is the protocol version number. And 0x60 is clearly not a valid version number. The confusing part of the message is "Error status: Unsupported protocol version (2)". Here, "2" is the error status value for "Unsupported protocol version", this is not the invalid version number.

kamari-swami, 2022-12-02:
Yes. Agreed. 0x60 is not correct version. Since this is for testing purposes only, i guess, this seems to be trivial.
Also, i could make the (python) application that could open multiple streams (within a channel) and push multiple ECM requests....

Finally, i tested two channels (two instances of tsecmg on port 2222 and 2223), each having 30 streams.

Thanks a lot for all the help!

closed 2022-12-02 by kamari-swami