tsduck/tsduck #1138: [Q] Property way to remux PES from 2 sources to one output MPEG-TS.

vilkoivshchi, 2022-12-07:
Good evening! 
I have a task of reduce bitrate from multicast input and pass scte35 pid to output. Input multicast is 224.1.2.100:5000 and have overall bitrate 12 Mbit:
```
Input #0, mpegts, from 'udp://224.1.2.100:5000':
  Duration: N/A, start: 52758.541689, bitrate: N/A
  Program 1000
    Metadata:
      service_name    :
      service_provider:
  Stream #0:0[0x3e9]: Video: h264 (High) ([27][0][0][0] / 0x001B), yuv420p(tv, bt709, top first), 1920x1080 [SAR 1:1 DAR 16:9], 25 fps, 25 tbr, 90k tbn
  Stream #0:1[0x3ea]: Audio: mp2 ([3][0][0][0] / 0x0003), 48000 Hz, stereo, fltp, 192 kb/s
  Stream #0:2[0x3ec](rus): Subtitle: dvb_teletext ([6][0][0][0] / 0x0006), 492x250
  Stream #0:3[0x3ed]: Data: scte_35
  No Program
  Stream #0:4[0x12]: Data: epg
```
I make fork with tsduck and cut pid's from PMT (it's actual config, early I've not cut pid's from PMT):
```
tsp -vv -I ip 224.1.2.100:5000 -l 10.10.10.10 -P fork "tsp -P pmt -r 0x03e9-0x3ea -P filter --pid 0x3e9 --pid 0x3ea --negate -O ip 224.1.2.101:6000 -l 10.10.10.10"  -P filter  --pid 0x3e8 --pid 0x3e9 --pid 0x3ea --pid 0x0000  -O ip 224.1.2.102:6000 -l 10.10.10.10
```
Now I have two multicasts: one with video/audio and another one with SCTE35 + Teletext PID's.
Then I transcoding video using FFMpeg:
```
ffmpeg -hide_banner -hwaccel qsv -i "udp://224.1.2.102:6000?localaddr=10.10.10.10&fifo_size=3000000"  -c:v h264_qsv -b:v 5M -filter:v fps=25 -g 50 -c:a mp2 -b:a 192k -ac 2 -bufsize 10M -f mpegts "udp://224.1.2.104:6000?pkt_size=1316&localaddr=10.10.10.10"
```
And now I have multicast with 5Mbit video instead 12Mbit. So far, so good. Problems begin whet I try to mux this stream and stream with SCTE35 pid to one output:
```
tsp -I ip 224.1.2.101:6000 -l 10.10.10.10  -P merge "tsp -I ip 224.1.2.104:6000 -l 10.10.10.10 -P remap -s 0x0100=0x3e9 -s 0x0101=0x3ea -P pat -r 0x1000 -P filter --pid 0x0 --pid 0x11 --pid 0x10 --pid 0x1000 --negate" -P pat -u -r 0x0010 -P pmt -a 0x03e9/27 -a 0x03ea/3  -O ip 224.1.2.103:6000 -l 10.10.10.10
``` 
Here I'm add new PES;s and define it in PMT. But in output I have CC errors and therefore unstable picture. What I'm doing wrong?

lelegard, 2022-12-07:
Hi @vilkoivshchi 

Using UDP/IP is unreliable by design. Using it to pipe into `ffmpeg` is a recipe for packet loss because of its erratic and bursty output. Which is probably what happen to you.

If you want to transcode some AV content of a service, use a classical system pipe between `tsp` and `ffmpeg`. No packet will be lost. The problem is that non-AV PID's such as the SCTE-35 PID are typically destroyed by `ffmpeg`. This is probably the reason for your complex UDP/IP configuration.

I would recommend to follow the example in section 5.2.28 of the TSDuck user's guide. It explains how to `tsp ... | ffmpeg .. | tsp ...` without losing packets and encapsulating non-AV PID's to safely travel through `ffmpeg`. In the sample case, the preserved PID's are only EIT but you can add other PID's as well in the same "tunnel PID". Reading your commands, I assume you know how to preserve and restore the corresponding signalisation.

Be sure to read the reference doc for the `encap` and `decap` plugins.

Because `ffmpeg` breaks the flow regulation and produces a bursty output, I also recommend to add a `regulate` plugin at the end, before the final `-O ip`. Also RTFM for `regulate` and `ip`. Since this is based on hypothetical problems from the `ffmpeg` output, first try without `regulate`. And if the packet reception is erratic, add it.

vilkoivshchi, 2022-12-09:
Hi @lelegard 
Thanks for your reply. I looked through the documentation on encap/decap plugins, but I couldn't figure out how to use them in my case, because I completely forgot about the pipes.
In my case I did the following:
```
tsp -I ip 224.1.2.100:5000 -l 10.10.10.10  -P encap  --pid 0x3ed  --output-pid 0x900 ^
-P pmt --remove-pid 0x03ec --remove-pid 0x03ed --add-pid 0x900/0x06  ^
-P filter --pid 0x12 --pid 0x3ed --pid 0x3ec --negate  | ^
ffmpeg -hide_banner -hwaccel qsv -f mpegts -i - -map 0:d -c:d copy -map 0:v -c:v h264_qsv -b:v 5M -r 25 -g 50 ^
-map 0:a -c:a mp2 -b:a 192k -ac 2 -bufsize 10M -mpegts_start_pid 0x900 -f mpegts - | ^
tsp -P pmt -a 0x3ed/0x86 -m 0x901/0x3e9 -m 0x902/0x3ea -P remap -s 0x901=0x3e9 ^
-s 0x902=0x3ea -P decap  --pid 0x900 -P pmt  --remove-pid 0x900 -P regulate ^
-O ip 224.1.2.101:5000 -l 10.10.10.10
```
I've cut teletext, it not necessary for me yet.

closed 2022-12-09 by vilkoivshchi