tsduck/tsduck #1139: [Q] BAT

EnoSat, 2022-12-09:
stream - https://mega.nz/file/tRkQiJoS#D6A2MoL4Hm_D3PSd-IpVV0oz3agewT8puUHPfxGOoPI
I don't know if it's a TSduck error or some error in the stream, but the complete BAT table is not extracted (partial statement)
`* BAT, TID 74 (0x4A), PID 17 (0x0011)
  Version: 23, sections: 1, total size: 577 bytes
  - Section 0:
    Bouquet Id: 1 (0x0001)
    Bouquet information:
    - Descriptor 0: Bouquet Name (0x47, 71), 9 bytes
      Name: "Antik BAT"
    Transport Stream Id: 30400 (0x76C0), Original Network Id: 366 (0x016E)
    - Descriptor 0: Service List (0x41, 65), 60 bytes
      Service id: 1151 (0x047F), Type: 0x1F (HEVC digital television service)
      Service id: 1152 (0x0480), Type: 0x1F (HEVC digital television service)
      Service id: 1153 (0x0481), Type: 0x1F (HEVC digital television service)
      Service id: 1154 (0x0482), Type: 0x1F (HEVC digital television service)
      Service id: 1155 (0x0483), Type: 0x1F (HEVC digital television service)
      Service id: 1156 (0x0484), Type: 0x1F (HEVC digital television service)
      Service id: 1157 (0x0485), Type: 0x19 (Advanced codec HD digital television service)
      Service id: 1158 (0x0486), Type: 0x1F (HEVC digital television service)
      Service id: 1159 (0x0487), Type: 0x1F (HEVC digital television service)
      Service id: 1160 (0x0488), Type: 0x1F (HEVC digital television service)
      Service id: 1161 (0x0489), Type: 0x1F (HEVC digital television service)
      Service id: 1162 (0x048A), Type: 0x1F (HEVC digital television service)
      Service id: 1163 (0x048B), Type: 0x1F (HEVC digital television service)
      Service id: 1164 (0x048C), Type: 0x1F (HEVC digital television service)
      Service id: 1165 (0x048D), Type: 0x1F (HEVC digital television service)
      Service id: 1166 (0x048E), Type: 0x1F (HEVC digital television service)
      Service id: 1167 (0x048F), Type: 0x1F (HEVC digital television service)
      Service id: 1168 (0x0490), Type: 0x1F (HEVC digital television service)
      Service id: 1171 (0x0493), Type: 0x1F (HEVC digital television service)
      Service id: 1199 (0x04AF), Type: 0x0C (Data broadcast service)
    - Descriptor 1: Private Data Specifier (0x5F, 95), 4 bytes
      Specifier: 0x00000029 (NorDig)
    - Descriptor 2: NorDig Logical Channel (V1) (0x83, 131), 76 bytes
      Service Id:  1153 (0x0481), Visible: 1, Channel number:  15
      Service Id:  1156 (0x0484), Visible: 1, Channel number:  16
      Service Id:  1154 (0x0482), Visible: 1, Channel number:  17
      Service Id:  1159 (0x0487), Visible: 1, Channel number:  86
      Service Id:  1160 (0x0488), Visible: 1, Channel number:  87
      Service Id:  1162 (0x048A), Visible: 1, Channel number:  81
      Service Id:  1164 (0x048C), Visible: 1, Channel number:  83
      Service Id:  1155 (0x0483), Visible: 1, Channel number:  84
      Service Id:  1152 (0x0480), Visible: 1, Channel number:  85
      Service Id:  1161 (0x0489), Visible: 1, Channel number:  80
      Service Id:  1171 (0x0493), Visible: 1, Channel number:  82
      Service Id:  1151 (0x047F), Visible: 1, Channel number:  32
      Service Id:  1157 (0x0485), Visible: 1, Channel number:  71
      Service Id:  1163 (0x048B), Visible: 1, Channel number:  52
      Service Id:  1158 (0x0486), Visible: 1, Channel number:  43
      Service Id:  1165 (0x048D), Visible: 1, Channel number:  18
      Service Id:  1166 (0x048E), Visible: 1, Channel number:  68
      Service Id:  1167 (0x048F), Visible: 1, Channel number:  55
      Service Id:  1168 (0x0490), Visible: 1, Channel number:   3
    Transport Stream Id: 20900 (0x51A4), Original Network Id: 366 (0x016E)
    - Descriptor 0: unknown (0x83, 131), 12 bytes
      0000:  12 5C C0 36 12 70 C0 33 12 98 F0 38        .\.6.p.3...8
    - Descriptor 1: Private Data Specifier (0x5F, 95), 4 bytes
      Specifier: 0x00000029 (NorDig)
    Transport Stream Id: 40200 (0x9D08), Original Network Id: 366 (0x016E)
    - Descriptor 0: unknown (0x83, 131), 4 bytes
      0000:  03 01 C0 4E                                ...N
    - Descriptor 1: Private Data Specifier (0x5F, 95), 4 bytes
      Specifier: 0x00000029 (NorDig)
    Transport Stream Id: 20700 (0x50DC), Original Network Id: 366 (0x016E)
    - Descriptor 0: unknown (0x83, 131), 4 bytes
      0000:  0E 74 C0 3D                                .t.=
    - Descriptor 1: Private Data Specifier (0x5F, 95), 4 bytes
      Specifier: 0x00000029 (NorDig)
    Transport Stream Id: 40600 (0x9E98), Original Network Id: 366 (0x016E)
    - Descriptor 0: unknown (0x83, 131), 40 bytes
[... 168 lines cut ...]
		 04BA C0 48
		 04BB C0 4F
		 04B9 C0 2C
		 04B2 C0 0D
		 04B7 C0 1C
		 04BD C0 15
		 04BF C0 0A
		 04BC C0 2D
C418 016E F0 30
   83 28 0262 C0 46
		 0280 C0 18
		 025E C0 42
		 026E C0 2A
		 0259 C0 4B
		 025D C0 16
		 0261 C0 1E
		 0263 C0 35
		 025A C0 0B
		 0289 C0 3A
   5F 04 00000029
C6D4 016E F0 10
   83 08 0792 C0 2F
		 079A C0 2E
   5F 04 00000029
B4 B3 F7 37`

lelegard, 2022-12-09:
Hi @EnoSat 

The BAT is incorrect in the stream. The PDS descriptor shall always precede the private descriptors. The order is correct in the first TS but incorrect in all others.

The interpretation of TSDuck is correct. See below, interspersed TSDuck interpretation and binary dump. The "Antik" bouquet (I don't know who they are) is doing a terrible job at building its signalisation.

~~~
* BAT, TID 74 (0x4A), PID 17 (0x0011)
  Version: 23, sections: 1, total size: 577 bytes
  - Section 0:
    Bouquet Id: 1 (0x0001)

00000000:  4A F2 3E 00 01 EF 00 00

    Bouquet information:
    - Descriptor 0: Bouquet Name (0x47, 71), 9 bytes
      Name: "Antik BAT"

                                   F0 0B 47 09 41 6E 74 69
00000010:  6B 20 42 41 54 F2 26

    Transport Stream Id: 30400 (0x76C0), Original Network Id: 366 (0x016E)

                                76 C0 01 6E F0 92

    - Descriptor 0: Service List (0x41, 65), 60 bytes
      ....

                                                  41 3C 04
00000020:  7F 1F 04 80 1F 04 81 1F 04 82 1F 04 83 1F 04 84
00000030:  1F 04 85 19 04 86 1F 04 87 1F 04 88 1F 04 89 1F
00000040:  04 8A 1F 04 8B 1F 04 8C 1F 04 8D 1F 04 8E 1F 04
00000050:  8F 1F 04 90 1F 04 93 1F 04 AF 0C 5F 04 00 00 00
00000060:  29 83 4C 04 81 C0 0F 04 84 C0 10 04 82 C0 11 04
00000070:  87 C0 56 04 88 C0 57 04 8A C0 51 04 8C C0 53 04
00000080:  83 C0 54 04 80 C0 55 04 89 C0 50 04 93 C0 52 04
00000090:  7F C0 20 04 85 C0 47 04 8B C0 34 04 86 C0 2B 04
000000A0:  8D C0 12 04 8E C0 44 04 8F C0 37 04 90 C0 03

    Transport Stream Id: 20900 (0x51A4), Original Network Id: 366 (0x016E)

                                                        51
000000B0:  A4 01 6E F0 14

    - Descriptor 0: unknown (0x83, 131), 12 bytes
      0000:  12 5C C0 36 12 70 C0 33 12 98 F0 38        .\.6.p.3...8

                          83 0C 12 5C C0 36 12 70 C0 33 12
000000C0:  98 F0 38

    - Descriptor 1: Private Data Specifier (0x5F, 95), 4 bytes
      Specifier: 0x00000029 (NorDig)

                    5F 04 00 00 00 29

    Transport Stream Id: 40200 (0x9D08), Original Network Id: 366 (0x016E)

                                      9D 08 01 6E F0 0C

    - Descriptor 0: unknown (0x83, 131), 4 bytes
      0000:  03 01 C0 4E                                ...N

                                                        83
000000D0:  04 03 01 C0 4E

    - Descriptor 1: Private Data Specifier (0x5F, 95), 4 bytes
      Specifier: 0x00000029 (NorDig)

                          5F 04 00 00 00 29

    Transport Stream Id: 20700 (0x50DC), Original Network Id: 366 (0x016E)
    etc....
                                            50 DC 01 6E F0
000000E0:  0C 83 04 0E 74 C0 3D 5F 04 00 00 00 29 9E 98 01
000000F0:  6E F0 30 83 28 07 D1 C0 01 07 D2 C0 02 07 D3 C0
00000100:  04 07 D4 C0 06 07 D7 C0 07 07 D9 C0 05 07 DB C0
00000110:  17 07 D5 C0 08 07 D6 C0 09 07 DC C0 0C 5F 04 00
00000120:  00 00 29 C5 A8 01 6E F0 0C 83 04 07 29 C0 30 5F
00000130:  04 00 00 00 29 9D D0 01 6E F0 3C 83 34 0A A2 C0
00000140:  22 0A 9D C0 24 0A A5 C0 25 0A 9E C0 23 0A 9F C0
00000150:  26 0A 97 C0 40 0A 98 C0 49 0A 9A C0 4D 0A 94 C0
00000160:  27 0A A0 C0 31 0A A1 C0 32 0A 9B C0 4C 0A 99 C0
00000170:  4A 5F 04 00 00 00 29 76 C1 01 6E F0 74 5F 04 00
00000180:  00 00 29 83 6C 04 B1 C0 43 04 B8 C0 14 04 B4 C0
00000190:  28 04 C0 C0 1A 04 C1 C0 21 04 C5 C0 1F 04 C6 C0
000001A0:  39 04 C7 C0 19 04 C8 C0 0E 04 C9 C0 1D 04 CA C0
000001B0:  3C 04 CB C0 41 04 CC C0 45 04 CD C0 3E 04 CF C0
000001C0:  1B 04 EA C0 29 04 B3 C0 13 04 B5 C0 3B 04 B6 C0
000001D0:  3F 04 BA C0 48 04 BB C0 4F 04 B9 C0 2C 04 B2 C0
000001E0:  0D 04 B7 C0 1C 04 BD C0 15 04 BF C0 0A 04 BC C0
000001F0:  2D C4 18 01 6E F0 30 83 28 02 62 C0 46 02 80 C0
00000200:  18 02 5E C0 42 02 6E C0 2A 02 59 C0 4B 02 5D C0
00000210:  16 02 61 C0 1E 02 63 C0 35 02 5A C0 0B 02 89 C0
00000220:  3A 5F 04 00 00 00 29 C6 D4 01 6E F0 10 83 08 07
00000230:  92 C0 2F 07 9A C0 2E 5F 04 00 00 00 29 B4 B3 F7
00000240:  37
~~~

EnoSat, 2022-12-09:
thanks for confirming
it is an official provider - https://antiksat.sk/en/

closed 2022-12-10 by EnoSat