tsduck/tsduck #1141: How to scrambler Using Multi Keys

M-Bala-2612, 2022-12-14:
@lelegard 

How to scrambler Using Multi Keys

#!/bin/bash
export STREAM_NAME=TESTING
export OUTPUT_URL="235.1.20.20:1234"
$source /home/user/SCRIPTS/TSP_Streams/TESTING/KeyGenerator
export KEY=`head -1 key` 

tsp -v -m --synchronous-log -I ip -r $INPUT_URL \
 -P scrambler --ignore-scrambled --cw $KEY --dvb-csa2 2001  \
  -P reduce --target-bitrate 2000000 -P pcrbitrate \
  -P cat -c -a "$CAS_ID/$CA_ID" \
  -P svrename --provider "$PROVIDER" \
  -P filter --negate --pid 0x1FFF \
  -O ip $OUTPUT_URL </dev/null > /dev/null 2>&1 & echo $! > $STREAM_PID_PATH1 

I will restart using this script for scrambling... it was one time scrambling .
How to the stream automatically scrambling with multi keys for every 10 minutes

M-Bala-2612, 2022-12-15:
@lelegard Thank you for your reply sir !

#!/bin/bash

if [ -f key ]; then
  rm -r key
fi
openssl rand -hex 8 >> key

I am using this script to KeyGeneration....  I want to KeyGeneration every 10 minutes 
```
tsp -v -m --synchronous-log -I ip -r $INPUT_URL
-P scrambler --ignore-scrambled --cw $KEY --dvb-csa2 2001
-P reduce --target-bitrate 2000000 -P pcrbitrate
-P cat -c -a "$CAS_ID/$CA_ID"
-P svrename --provider "$PROVIDER"
-P filter --negate --pid 0x1FFF
-O ip $OUTPUT_URL
```

if  i  Generated the key automatically in file, how to read the key automatically every 10s when in run my command line ... without stopping the stream. It possible?

 -P scrambler --ignore-scrambled `--cw $KEY` --dvb-csa2 2001

lelegard, 2022-12-15:
You need to think globally, at system level, end to end. Encrypting is not a goal in itself. Encrypting is done so that only selected people can decrypt the service. This means that you need to provide the decryption key to those selected people.

In your exercise, you generate a random key every 10s, you encrypt with this key, and then what? Nobody knows the decryption keys, nobody can decrypt the stream, you's better throw the stream to the bin instead of wasting broadcast resources.

More seriously, this is what I explained in the previous post. Changing keys requires a system view to distribute the decryption keys to selected recipients. This is why you need a CAS, conditional access system, or your can call it whatever. The DVB-defined interface to the CAS is an ECMG. When using an ECMG for a given CAS, the `scrambler` plugin will generate random keys for you every N seconds (your choice) but an ECMG is required so that those random encryption keys are not lost and securely transmitted to the selected subscribers.

Selecting random keys every N seconds without a CAS or equivalent is a non-sense. The system is unusable, it serves no purpose.

M-Bala-2612, 2022-12-30:
Thankyou sir @lelegard 

i have another one doubt pls reply me...

How to multi users plying the SRT stream

 ```
tsp -v -m --synchronous-log -I ip -r $INPUT_URL \
 -P analyze -i 30 -o report.txt  \
 -O srt --listener $SRT_IP --transtype live --messageapi ``` --multiple ```
```
It playing only one user ... If extra another users trying to play but the stream doesnot playing
first user only play the stream.... i think it unicast method .... How to output stream was multicasting

lelegard, 2022-12-30:
> how to output stream was multicasting.... any commad

This is not related to this thread of discussion. Maybe you are referring to #1150

M-Bala-2612, 2022-12-30:
Thankyou sir @lelegard

closed 2022-12-30 by M-Bala-2612