tsduck/tsduck #1143: [Q] Want to down original bitrate of  live stream video

prashant-09-tech, 2022-12-14:
Hi,
I am using ts duck tool to pull down the original bitrate of video  and I  found some command related to this but video is not working properly.

@lelegard   could you please tell me which command is used   for this ? .

prashant-09-tech, 2022-12-15:
tsp    -v  -I     ip 10201 -P pcradjust --bitrate 3000000 -O ip 127.0.0.1:10202 
 
Hi @lelegard   
I am using  the above command and as expected it pulls down  the resultant transport stream bitrate  but  when I play the transport stream  video then   I am  facing  jittering problem .
Checked by this command   : :    tsp -I ip 10202 -P until -s 8 -P analyze -O drop 

Thanks

lelegard, 2022-12-15:
> tsp -v -I ip 10201 -P pcradjust --bitrate 3000000 -O ip 127.0.0.1:10202

The plugin `pcradjust` modifies the values of the PCR. It does not change the number of bits which are received per second.

prashant-09-tech, 2022-12-16:
Hi , @lelegard 
thanks For giving time

closed 2022-12-19 by prashant-09-tech