tsduck/tsduck #1144: Real time IP analysis

przemo1983, 2022-12-16:
I am using following command to analyze IP UDP stream: 
`tsp -I ip 239.39.39.39:3939 -l 10.8.8.1 --receive-timeout 10000 -P analyze -o ip.analysis.normalized --normalized -i 2 -O drop`
Each time when new analysis is performed the analysis context is reset.

Is it possible to add a parameter that the analysis context will not be reset each time (after -i X)? 
This will be very helpful to make real time UDP stream analysis over some longer time period but with cyclic updates of errors/data. Of course the file(s) should be updated each -i X value (with correct values from beginning of analysis) to get those values for further analysis.

lelegard, 2022-12-16:
Good idea. Will do...

lelegard, 2022-12-16:
Option `--cumulative` has been added.

closed 2023-01-06 by lelegard