tsduck/tsduck #1145: [BUG] Physical Layer Scrambling (PLS) according to DVB API doesn't work

basicmaster, 2022-12-16:
#### Bug description:
Physical Layer Scrambling (PLS) cannot be used with the Digital Devices SX8, for the following reason:

The current implementation (as a result of #277) uses a triplet value for the `DTV_STREAM_ID` property, to provide the desired multistream ID as well as a PLS code. This seems to be [an older way to do it](https://tvheadend.org/issues/5556) (dvblast [does the same](https://code.videolan.org/videolan/dvblast/-/blob/master/dvblast.c#L1281)) which should probably be kept for backwards compatibility.

Meanwhile the DVB API [provides](https://git.linuxtv.org/media_tree.git/commit/include/uapi/linux/dvb/frontend.h?id=f9d79126195374c285035777b9d6abd24ceba363) the separate property `DTV_SCRAMBLING_SEQUENCE_INDEX` to set a gold code (which could be calculated from a root code, if a root code is used instead). This way is used by the mentioned card.

#### How to reproduce:
Example command for [a RAI transponder on 5°W that uses multistream and PLS](https://www.lyngsat.com/muxes/Eutelsat-5-West-B_Transalpine_11013-V.html):
```
tsp -v -I dvb -a 16 -f 11013000000 --polarity vertical --delivery-system DVB-S2 -m 8-PSK -s 35300000 --fec-inner 2/3  --pls-mode GOLD --pls-code 131070 --isi 5 -O drop
```
TSDuck doesn't lock on this transponder.

The card vendor provides [a small test tool](https://github.com/DigitalDevices/dddvb/blob/master/lib/ddzap.c) which allows to successfully lock to this transponder and to output the TS. It uses `DTV_SCRAMBLING_SEQUENCE_INDEX` for the PLS code.

#### Expected behavior:
TSDuck is able to lock on this transponder.

#### Errors and logs:
```
* dvb: using /dev/dvb/adapter16 (DVB-S, DVB-S2)
* dvb: tuned to transponder --delivery-system DVB-S2 --frequency 11,013,000,000 --symbol-rate 35,300,000 --fec-inner 2/3 --polarity vertical --modulation 8-PSK --pilots off --roll-off 0.35 --isi 5 --pls-code 131070 --pls-mode GOLD --lnb Extended --satellite-number 0
* dvb: expected bitrate from tuning parameters: 69,916,459 b/s
* Error: dvb: no input signal lock after 5000 milliseconds
```

#### Environment:
 - OS: Linux
 - OS version: Debian 11
 - TSDuck full version: 3.31-2713 (but I can see that the respective code part has not been touched since)
 - Installation type: build from Git

#### Additional information:
Maybe also the reason for #1081.

lelegard, 2022-12-16:
Hi @basicmaster 

Thanks for the detailed report and suggestion of fix. I no longer have access to satellite reception and I cannot test. This kind of fix should be adjusted according to test results. 

Would you have the possibility to fix the code and test the reception? The way you describe it, it seems a very local fix. If you can do this, then please [submit a pull request](https://tsduck.io/doxy/contributing.html).

basicmaster, 2022-12-19:
Yes, I will be able to to so (and will do). First I just have to find out the relation of root and combo codes, compared to gold codes here.

closed 2024-08-30 by basicmaster