tsduck/tsduck #1148: [BUG] tspcontrol: error connecting socket: Connection refused after stopping/restarting tsp

lmoody1, 2022-12-27:
#### Bug description:
After starting, stopping and restarting tsp I'm getting the following errors:

    Error: control commands: error binding socket to local address: Address already in use
    Error: control commands: error starting TCP server for control commands.

Now, this occurs only after executing a tspcontrol command after tsp is first started, and before it's stopped and restarted.
Note: This only seems to be an issue with linux. We're not seeing the same problem under Windows.
Another note: if we wait 60 seconds after stopping tsp and then restarting it, we're not seeing the problem.

#### How to reproduce:
in a linux session, execute the following tsp command:

    tsp  --add-start-stuffing 1 --initial-input-packets 1 --control-source ##.##.##.### --control-port #### -v -b 41209008 -I null -j -O  ip localhost:8999 -P regulate   \
    -P merge -j --no-psi-merge --ignore-conflicts  -p 0x1-0x18   -d 0x0  -d 0x20-0x1ffe  "tsp -v -b 1000000 -I file -i test_stream.ts -P regulate"  \
    -P inject Streams_PAT_Inject_File.xml --pid 0x0 --bitrate 15000  \
    -P inject Streams_PMT_Inject_File.xml --pid 0x21 --bitrate 15000   \
    -P merge  -j -t --no-pcr-restamp  "tsp -v -b 0 -I null -P regulate"

in another session on the same machine issue the following command:

    tspcontrol -t ##.##.##.###:#### list

in the first session, ctrl-c to stop tsp
after tsp is has stopped, rerun the tsp command. Tsp generates an error:

    Error: control commands: error binding socket to local address: Address already in use
    Error: control commands: error starting TCP server for control commands.

However, it does continue on to execute the muxing process. But, tspcontrol commands fail after this point.

#### Injection Files:

**Streams_PMT_Inject_File.xml:**
~~~~~
  <?xml version="1.0" ?>
  <tsduck/>
~~~~~
**Streams_PAT_Inject_File.xml:**
~~~~~
<?xml version="1.0" ?>
<tsduck>
        <PAT version="1" network_PID="0x10" transport_stream_id="21"/>
</tsduck>
~~~~~

#### Expected behavior:
tsp restarts without error

#### Environment:
 - OS:  linux
 - OS version: Debian GNU/Linux 11 (bullseye)
 - TSDuck full version: 3.32-2983
 - Installation type:  tsduck_3.32-2983.debian11_amd64.deb

#### Additional information:

lelegard, 2022-12-27:
Hi @lmoody1 

This is not a bug, this is a Linux issue indeed. After stopping tsp, the control socket remains in wait state for a few seconds. If you restart tsp using the same control port during this short interval, you get the "in use" error. 

To avoid this, use option `--control-reuse-port` to force the reuse of the port. This option is not enabled by default to avoid the mess of two active tsp at the same time on the same control port.

lmoody1, 2022-12-27:
@lelegard 

Thank you so much. That resolved my problem.

closed 2023-01-06 by lelegard