tsduck/tsduck #1150: [Q] how to multicast the output srt

udhayNuk, 2022-12-30:
@lelegard 
How to multicast the srt streams
tsp -I srt --caller 192.168.0.103:4900 -P analyze -i 10 -o output-analysis1.txt -O srt 192.168.0.10:4900 --transtype live --messageapi 

This script only unicast, how to multicast the srt, please solve my question
is it possible to multicast the srt output.
Thank you in advance

lelegard, 2022-12-30:
> is it possible to multicast the srt output using tsduck tool.

It depends on what you mean by "multicast the srt output".

- Usually, IP TV multicast is achieved on raw UDP or RTP. So, you get an SRT _input_ (`-I srt`) and you multicast it over raw UDP or RTP using `-O ip`. Please RTFM about the `ip` output plugin.
- Maybe there is a way to multicast the SRT protocol, meaning using the `srt` output plugin over multicast datagrams, but I do not know anything about it. This is a question to be asked in the https://github.com/Haivision/srt project.

udhayNuk, 2022-12-30:
@lelegard 
Thank you for your reply

closed 2022-12-30 by udhayNuk