tsduck/tsduck #1155: How the change the size of video resolution?

udhayNuk, 2023-01-03:
@lelegard
How the change the size of video resolution?
[report.txt](https://github.com/tsduck/tsduck/files/10334473/report.txt)
my video resolution was 720x576 
tsp --debug -I ip 233.1.1.119:1001 -P pcrbitrate -P reduce --target-bitrate 2000000 -P analyze -i 10 -o report.txt -O ip 235.1.1.1:2000

how to change the video resolution, any plugins available in tsduck .
please solve my question, is there any way to change the resolution of output video
thank you in advance

lelegard, 2023-01-03:
Changing the video resolution means video transcoding. I don't know how many times I will have to say that TSDuck works on the transport stream layer, not on the audio/video content. TSDuck does not perform A/V transcoding. You must use some other tool such as ffmpeg.

udhayNuk, 2023-01-03:
@lelegard 
thank you dude

udhayNuk, 2023-01-03:
@lelegard 
Thank you for your reply
i also know you told already video/audio transcoding not done in tsduck.
i want output only change the size of video
please reply me

udhayNuk, 2023-01-04:
@lelegard 
ok dude thanks for your reply

closed 2023-01-04 by udhayNuk