tsduck/tsduck #1156: [BUG] -P filter plugin doesn't seem to work after -P duplicate plug

lmoody1, 2023-01-06:
#### Bug description:
When merging a stream, I'm trying to duplicate a pid from the input stream, then filter out out the original pid and pass only the stream's new pid to the  output. I know that may seem strange, but we have conditions where we might duplicate the same pid to multiple output pids in a single output stream.

I'm using basically the "-P duplicate 0xa=0xb  -P filter -s -p 0xb" plugin structure.

It looks like the second filter isn't working, and I think both source streams with pid 0xa and 0xb are both being sent to the output stream. I'm getting a lot of "Warning: duplicate: buffer overflow, dropping packet" errors.

I've also tried adding a "-P filter -s -n -p 0xa" to specifically exclude 0xa from the output, but the results are the same.

#### How to reproduce:
Here's the tsp command I'm using:

    tsp --control-source localhost --control-port 8000 --control-reuse-port -v -b 41209008 -I null -j -O ip localhost:8999 -P regulate \
    -P merge -j -t --no-pcr-restamp "tsp -v -b 7000000 -I file -i \files\CAT_STREAM_0x0_0x21.ts -P filter -s -p 0x0-0x21 -P regulate" \
    -P merge -j -t --no-pcr-restamp "tsp -v -b 1000000 -I file -i \files\INPUT_STREAM_8E0.ts -P filter -s -p 0x8e0 -P duplicate 0x8e0=0x8e1 -P filter -s -p 0x8e1 -P regulate" 

I'm attempting to merge INPUT_STREAM_8E0.ts, filtering to pass only 0x8e0 pid then duplicating 0x8e0 to 0x8e1, then filtering to pass only 0x8e1.

I've also added "-P filter -s -n -p 0x8e0" before the "-P regulate" to specifically filter out 0x8e0 and got the same results.

#### Expected behavior:
Output stream should only include streams with the remapped pid without error.

#### Errors and logs:
Getting a continuous stream of "Warning: duplicate: buffer overflow, dropping packet" errors on the screen.

#### Environment:
 - OS: Windows
      **_- The problem is also occurring in linux servers._**
 - OS version: Windows 10 Enterprise 64 bit, version 21H2, build 19044.2364
 - TSDuck full version: 3.32-2983
 - Installation type: official binary "TSDuck-Win64-3.32-2983.exe"

lmoody1, 2023-01-06:
**_Note that this used to work (don't remember what version of tsDuck was installed at the time)._**

lelegard, 2023-01-06:
Hi @lmoody1 

Can you reproduce the problem with one single tsp command? Without the imbrication of merge plugins.

If you find packets with PID 0x0A on output, are you sure that they come from the second merged command? Maybe they come from the first merged command.

lmoody1, 2023-01-06:
Yes. Here is the tsp command I'm using:

    tsp --control-source localhost --control-port 8000 --control-reuse-port -v -b 41209008 -I file -i \files\INPUT_STREAM_8E0.ts -P filter -p 0x8e0 -P duplicate 0x8e0=0x8e1 -P filter -p 0x8e1 -O ip localhost:8999 -P regulate

lelegard, 2023-01-07:
Hi @lmoody1 

> Getting a continuous stream of "Warning: duplicate: buffer overflow, dropping packet" errors on the screen.

This means that there is not enough null packets in the stream to duplicate the PID. You need to make sure that the stream contains at least as many null packets as packets in the PID to duplicate. Otherwise, it will fail with this error, just like copying/duplicating a file into a file system where the free space is smaller that the file to copy.

> It looks like the second filter isn't working, and I think both source streams with pid 0xa and 0xb are both being sent to the output stream. 

I do not see how this is possible. I cannot reproduce your problem. I do not have your input file, but it should not make any difference.

Please try this to demonstrate that the chain works as expected:
~~~
$ tsversion
3.32-2983
$ tsp -I null 10000 \
    -P filter --every 5 --set-label 1 \
    -P craft --only-label 1 --pid 0x8e0 \
    -P count --tag step.1 \
    -P filter -p 0x8e0 -s \
    -P duplicate 0x8e0=0x8e1 \
    -P count --tag step.2 \
    -P filter -p 0x8e1 \
    -P count --tag step.3 \
    -O drop
* count: step.1: PID 2272 (0x08E0):      2,000 packets
* count: step.1: PID 8191 (0x1FFF):      8,000 packets
* count: step.2: PID 2272 (0x08E0):      2,000 packets
* count: step.2: PID 2273 (0x08E1):      2,000 packets
* count: step.2: PID 8191 (0x1FFF):      6,000 packets
* count: step.3: PID 2273 (0x08E1):      2,000 packets
~~~

The first 3 plugins generate a stream with 80% null packet and 20% PID 0x08e0. The various `count` plugins report the number of packets per PID. We can see that the last step only contains PID 0x08e1.

Please note the `-s` in `-P filter -p 0x8e0 -s`. This option converts removed packets into null packets instead of removing them. If you do not use it, there is no longer any null packet and there is no possibility to duplicate anything since there is no free space left.

Without `-s`, no packet is duplicated and the last `filter` plugin removes everything since there is no PID 0x08e1. But, in no circumstances, there will be PID 0x08e0.

lmoody1, 2023-01-10:
Let me preface my questions by stating in advance that I am not intimately familiar with streaming. I'm learning as I go.

I have the following two examples that each duplicate 0x226 to 0x227 (without any filtering...also note that the file only  has one stream with pid 0x226):

    tsp --control-local localhost --control-port 8000 -v -b 41209008 -I null -j  \
    -P mux -b 1000000 c:\files\PID_226_File.ts \
    -P duplicate 0x226=0x227 \
    -O ip localhost:8999 -P regulate

And,

    tsp --control-local localhost --control-port 8000 -v -b 41209008 -I null -j  \
    -P merge -j -t --no-pcr-restamp "tsp -v -b 1000000 -I file -i c:\files\PID_226_File.ts -P duplicate 0x226=0x227 -P regulate"  \
    -O ip localhost:8999 -P regulate

The "-P mux" plugin with the "-P duplicate" plugin doesn't cause any issues. We're not seeing any errors and we're seeing PIDs "0x226", "0x227" and "0x1fff" on the output.

However, the "-P merge" plugin with the "-P duplicate" plug causes the "* Warning: duplicate: buffer overflow, dropping packet" errors and we're only seeing output from "0x226", not "0x227" nor "0x1fff".

I guess my question is, how do these two run differently? We can't use the "-P mux" plugin option; we're pretty much locked into using "-P merge". Are there options I can set on the "-P merge" line that will emulate how the "-P mux" plugin works? 

I don't know if this helps, but here is the tsanalyze output for the file (we get the same basic results with every file we've attempted this with, regardless of the pids of the streams):
~~~
===============================================================================
|  TRANSPORT STREAM ANALYSIS REPORT                                           |
|=============================================================================|
|  Transport Stream Id: ............. Unknown  |  Services: .............. 0  |
|  Bytes: ........................ 16,723,164  |  PID's: Total: .......... 1  |
|  TS packets: ....................... 88,953  |         Clear: .......... 1  |
|     With invalid sync: .................. 0  |         Scrambled: ...... 0  |
|     With transport error: ............... 0  |         With PCR's: ..... 0  |
|     Suspect and ignored: ................ 0  |         Unreferenced: ... 1  |
|-----------------------------------------------------------------------------|
|  Transport stream bitrate, based on ....... 188 bytes/pkt    204 bytes/pkt  |
|  User-specified: ................................... None             None  |
|  Estimated based on PCR's: ...................... Unknown          Unknown  |
|  Selected reference bitrate: .................... Unknown             None  |
|-----------------------------------------------------------------------------|
|  Broadcast time: ................................................. Unknown  |
|-----------------------------------------------------------------------------|
|  Srv Id  Service Name                              Access          Bitrate  |
|                                                                             |
|  Note 1: C=Clear, S=Scrambled                                               |
|  Note 2: Unless specified otherwise, bitrates are based on 188 bytes/pkt    |
===============================================================================

===============================================================================
|  SERVICES ANALYSIS REPORT                                                   |
|=============================================================================|
|  Global PID's                                                               |
|  TS packets: 0, PID's: 0 (clear: 0, scrambled: 0)                           |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Global PID's ................................. C          Unknown  |
|   Subt.  Global PSI/SI PID's (0x00-0x1F) .............. C          Unknown  |
|=============================================================================|
|  Unreferenced PID's                                                         |
|  TS packets: 88,953, PID's: 1 (clear: 1, scrambled: 0)                      |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Unreferenced PID's ........................... C          Unknown  |
|  0x0226  Unreferenced ................................. C          Unknown  |
===============================================================================

===============================================================================
|  PIDS ANALYSIS REPORT                                                       |
|=============================================================================|
|  PID: 0x0226 (550)                                            Unreferenced  |
|-----------------------------------------------------------------------------|
|  Unreferenced PID          Transport:                Discontinuities:       |
|  Bitrate: ....... Unknown  Packets: ........ 88,953  Expected: ......... 0  |
|  Access: .......... Clear  Adapt.F.: ............ 0  Unexpect: ......... 0  |
|                            Duplicated: .......... 0  Sections:              |
|                            PCR: ................. 0  Unit start: ... 4,100  |
===============================================================================

===============================================================================
|  TABLES & SECTIONS ANALYSIS REPORT                                          |
===============================================================================
~~~

lelegard, 2023-01-11:
The problem has nothing to do with `merge`. It is only in the command of the stream to merge:
~~~
tsp -v -b 1000000 -I file -i c:\files\PID_226_File.ts -P duplicate 0x226=0x227 -P regulate
~~~

Your analysis of the file shows that it contains only one PID: 0x0226. Nothing else, no stuffing, no null packet. The description of the `duplicate` plugin mentions "duplicate PID's, reusing null packets". How could this plugin duplicate a PID when there is no null packet at all? This is like trying to copy a file in a file system which is already full with that single file.

If your goal is to modify the PID number of all packets in the TS, I would use this:

~~~
tsp -v -b 1000000 -I file -i c:\files\PID_226_File.ts -P craft --pid 0x227 -P regulate
~~~

More generally, if the TS contains several PIDs and you want to change all packets from PID 0x226 to 0x227:
~~~
tsp ... -P filter --pid 0x226 --set-label 1 -P craft --only-label 1 --pid 0x227 ...
~~~

Note that this is slighty different from `-P remap 0x226=0x227`. The above command works at packet level, all packets from 0x226 are renamed, from the beginning. The `remap` plugin works at TS level, updating the structure of the TS and its signalisation and it won't update the packets until it understands to which structure they belong to. This is why I recommend the `craft` plugi in your case.

You do not get the problem with the `mux` plugin in this command:
~~~
tsp ... -b 41209008 ... \
    -I null \
    -P mux -b 1000000 c:\files\PID_226_File.ts \
    -P duplicate 0x226=0x227 \
~~~
There is a logical explanation. In this case, the `mux` plugin reads the file one time and replaces null packets with packets from this file. However, the replacement is performed at bitrate 1 Mb/s (`-P mux -b 1000000`) while you specified that the TS bitrate was 41 Mb/s (`tsp -b 41209008`). Thus, roughly, only 1 null packet every 41 null packets is replaced by a packet from the file. This leaves plenty of free space in the TS for the `duplicate` plugin to work.

closed 2023-06-11 by github-actions[bot]