tsduck/tsduck #1160: [BUG] tsp process crash in arch-arm

Han-Suyong, 2023-01-13:
#### Bug description:

I test the sample code using plugin.
But, the crash is occurred in test.

#### How to reproduce:

tsp  -d -I file teststream01.ts -P count -O drop

#### Expected behavior:

packet count display

#### Errors and logs:

~~~
[be sure to surround verbatim log lines with "~~~" lines]
tsResidentBufferTemplate.h
template <typename T> ts::ResidentBuffer<T>::ResidentBuffer(size_t elem_count)
line 81 :     _base = new (_locked_base) T[elem_count];

#### Environment:
 - OS: Linux
 - OS version: Ubuntu 18.04 
 - TSDuck full version: 3.32-2983
 - Installation type: Both - official binary and which one, rebuilt binary and how it is used

#### Additional information:
tsResidentBufferTemplate.h
template <typename T> ts::ResidentBuffer<T>::ResidentBuffer(size_t elem_count)
line 81 :     _base = new (_locked_base) T[elem_count];

In case of typedef ResidentBuffer<TSPacketMetadata> PacketMetadataBuffer...
target arm arch - 4bytes align mode, but TSPacketMetadata (sizeof-17bytes) cannot allocation ararry memory.

lelegard, 2023-01-13:
Hi @Han-Suyong 

Thanks for the report. Could you please specify which CPU is this, arm32 or arm64?

The only Arm prebuilt binaries are arm32 on Rapsbian (Raspberry Pi) or arm64 on macOS (through Homebrew). How did you rebuild TSDuck on Ubuntu 32 or 64? Which compiler and which version? Is this a compilation error or a runtime exception?

lelegard, 2023-01-13:
Additionally, could you please post the output of the following commands;
~~~
uname -a
tsversion --version=all
~~~

micomx, 2023-01-16:
Hi lelegard,

Here are your requested info.
# uname -a
Linux 5.4.96-287 #1 SMP PREEMPT Wed Dec 7 13:37:28 UTC 2022 aarch64 GNU/Linux
# tsversion --version=all
tsversion: TSDuck - The MPEG Transport Stream Toolkit - version 3.32-2983
Built Jan 13 2023 - 11:48:57
Using GCC 9.3.0, C++ std 2011.03
System: Linux (Linux), on ARM, 32-bit, little-endian, page size: 4096 bytes
Bitrate: 64-bit floating-point value
Dektec: This version of TSDuck was compiled without Dektec support
VATek: This version of TSDuck was compiled without Dektec support
Web library: 
SRT library: This version of TSDuck was compiled without SRT support
RIST library: This version of TSDuck was compiled without RIST support

Anyway, We have workarounds for some issues already.
Could you verify them?

Runtime Issue #1
diff --git a/src/libtsduck/base/types/tsMemory.h b/src/libtsduck/base/types/tsMemory.h
index 2151059f..16a0e72f 100644
--- a/src/libtsduck/base/types/tsMemory.h
+++ b/src/libtsduck/base/types/tsMemory.h
@@ -131,6 +131,10 @@ namespace ts {
             // Simply prevent the compiler from rescheduling instructions (not a true "memory barrier" however).
             __asm__ __volatile__ ("" : : :  "memory");
 
+        #elif defined(TS_GCC) && (defined(__ARM_ARCH_7__) || defined(__ARM_ARCH_7A__) || defined(__ARM_ARCH_7R__)  || defined(__ARM_ARCH_7M__) || defined(__ARM_ARCH_7S__))
+
+            __asm__ __volatile__ ("dmb sy" : : : "memory");
+
         #elif defined(TS_GCC) && defined(TS_ARM)
 
             // For later reference, not sure this is valid.

Runtime Issue #2 (this issue)
diff --git a/src/libtsduck/dtv/transport/tsTSPacketMetadata.h b/src/libtsduck/dtv/transport/tsTSPacketMetadata.h
index 0c2312fa1..f65a9781f 100644
--- a/src/libtsduck/dtv/transport/tsTSPacketMetadata.h
+++ b/src/libtsduck/dtv/transport/tsTSPacketMetadata.h
@@ -46,7 +46,7 @@ namespace ts {
     // - The order of private fields is carefully set to avoid padding.
     // - Pragma pack to 1-byte alignment.
     //
-    #pragma pack(push, 1)
+//    #pragma pack(push, 1)
 
     //!
     //! Metadata of an MPEG-2 transport packet for tsp plugins.
@@ -337,7 +337,7 @@ namespace ts {
     };
 
     // Restore default packing after TSPacketMetadata.
-    #pragma pack(pop)
+//    #pragma pack(pop)
 
     //!
     //! Vector of packet metadata.

Additionally, I wonder why it works on Rsapsbian.
I think, it is byte align problem.
Is there any compile options for arm?
Here is command for compile.
make -C src VERBOSE=1 NOTEST=1 NODEKTEC=1 NOHIDES=1 NOVATEK=1 NOCURL=1 NOPCSC=1 NOSRT=1 NORIST=1 NOEDITLINE=1 NOGITHUB=1 NODTAPI=1 BINDIR=build 

Thanks,

lelegard, 2023-01-16:
Hi @micomx and @Han-Suyong 

Thanks for the response. There are several inconsistent information here.

1. `uname` says `aarch64` (64-bit Arm) while `tsversion` says `ARM, 32-bit`. It should say `Arm-64, 64-bit`.
2. You mentioned `Ubuntu 18.04` while `tsversion` says `Linux (Linux)`. It should say `Ubuntu (Ubuntu 18.04)`.

Did you compile and run on the same machine? What kind of machine is this? Is this a "standard" Ubuntu installation? 

You compile with `make -C src`. The `-C src` is not necessary but it should not make any difference for a simple compilation.

Could you post the output of  `make cxxmacros | grep -i -e arm -e arch` ?

Could you clone the latest state of the repo and rebuild? This should not make a difference on these low-level topics but then we will be aligned in terms of code. Please try the various commands below on the same machine as the build machine. Do you also see your runtime error on the same machine as the build machine?

For comparison, I use a MacBook with an Arm-64 CPU. An Ubuntu virtual machine is running on it. You can see the various command output below. The CPU is always identified as `Arm-64, 64-bit` by `tsversion`.

On Ubuntu 22.10 (virtual machine):
~~~
$ uname -a
Linux armubuntu 5.19.0-29-generic #30-Ubuntu SMP PREEMPT_DYNAMIC Wed Jan 4 12:32:16 UTC 2023 aarch64 aarch64 aarch64 GNU/Linux
$
$ tsversion --version=all
tsversion: TSDuck - The MPEG Transport Stream Toolkit - version 3.33-3137
Built Jan 13 2023 - 08:57:24
Using GCC 12.2.0, C++ std 2011.03
System: Ubuntu (Ubuntu 22.10), on Arm-64, 64-bit, little-endian, page size: 4096 bytes
Bitrate: 64-bit floating-point value
Dektec: This version of TSDuck was compiled without Dektec support
VATek: libvatek version 3.08
Web library: libcurl: 7.85.0, ssl: OpenSSL/3.0.5, libz: 1.2.11
SRT library: libsrt version 1.5.0
RIST library: librist version 0.2.7, API version 4.2.0
$ 
$ make cxxmacros | grep -i -e arm -e arch
#define __aarch64__ 1
#define __AARCH64_CMODEL_SMALL__ 1
#define __AARCH64EL__ 1
#define __ARM_64BIT_STATE 1
#define __ARM_ALIGN_MAX_PWR 28
#define __ARM_ALIGN_MAX_STACK_PWR 16
#define __ARM_ARCH 8
#define __ARM_ARCH_8A 1
#define __ARM_ARCH_ISA_A64 1
#define __ARM_ARCH_PROFILE 65
#define __ARM_FEATURE_CLZ 1
#define __ARM_FEATURE_FMA 1
#define __ARM_FEATURE_IDIV 1
#define __ARM_FEATURE_NUMERIC_MAXMIN 1
#define __ARM_FEATURE_UNALIGNED 1
#define __ARM_FP 14
#define __ARM_FP16_ARGS 1
#define __ARM_FP16_FORMAT_IEEE 1
#define __ARM_NEON 1
#define __ARM_PCS_AAPCS64 1
#define __ARM_SIZEOF_MINIMAL_ENUM 4
#define __ARM_SIZEOF_WCHAR_T 4
$ 
~~~

On macOS 13.1 (host machine):
~~~
$ uname -a
Darwin mactest 22.2.0 Darwin Kernel Version 22.2.0: Fri Nov 11 02:04:44 PST 2022; root:xnu-8792.61.2~4/RELEASE_ARM64_T8103 arm64
$ 
$ tsversion --version=all
tsversion: TSDuck - The MPEG Transport Stream Toolkit - version 3.33-3137
Built Jan 16 2023 - 08:44:22
Using Clang 14.0.0 (clang-1400.0.29.202), C++ std 2011.03
System: macOS (macOS 13.1, Darwin 22.2.0), on Arm-64, 64-bit, little-endian, page size: 16384 bytes
Bitrate: 64-bit floating-point value
Dektec: This version of TSDuck was compiled without Dektec support
VATek: libvatek version 3.08
Web library: libcurl: 7.85.0, ssl: SecureTransport (LibreSSL/3.3.6), libz: 1.2.11
SRT library: libsrt version 1.5.1
RIST library: librist version 0.2.7, API version 4.2.0
$ 
$ make cxxmacros | grep -i -e arm -e arch
#define __AARCH64EL__ 1
#define __AARCH64_CMODEL_SMALL__ 1
#define __AARCH64_SIMD__ 1
#define __ARM64_ARCH_8__ 1
#define __ARM_64BIT_STATE 1
#define __ARM_ACLE 200
#define __ARM_ALIGN_MAX_STACK_PWR 4
#define __ARM_ARCH 8
#define __ARM_ARCH_8_3__ 1
#define __ARM_ARCH_8_4__ 1
#define __ARM_ARCH_8_5__ 1
#define __ARM_ARCH_ISA_A64 1
#define __ARM_ARCH_PROFILE 'A'
#define __ARM_FEATURE_AES 1
#define __ARM_FEATURE_ATOMICS 1
#define __ARM_FEATURE_CLZ 1
#define __ARM_FEATURE_COMPLEX 1
#define __ARM_FEATURE_CRC32 1
#define __ARM_FEATURE_CRYPTO 1
#define __ARM_FEATURE_DIRECTED_ROUNDING 1
#define __ARM_FEATURE_DIV 1
#define __ARM_FEATURE_DOTPROD 1
#define __ARM_FEATURE_FMA 1
#define __ARM_FEATURE_FP16_FML 1
#define __ARM_FEATURE_FP16_SCALAR_ARITHMETIC 1
#define __ARM_FEATURE_FP16_VECTOR_ARITHMETIC 1
#define __ARM_FEATURE_FRINT 1
#define __ARM_FEATURE_IDIV 1
#define __ARM_FEATURE_JCVT 1
#define __ARM_FEATURE_LDREX 0xF
#define __ARM_FEATURE_NUMERIC_MAXMIN 1
#define __ARM_FEATURE_QRDMX 1
#define __ARM_FEATURE_SHA2 1
#define __ARM_FEATURE_SHA3 1
#define __ARM_FEATURE_SHA512 1
#define __ARM_FEATURE_SM3 1
#define __ARM_FEATURE_SM4 1
#define __ARM_FEATURE_UNALIGNED 1
#define __ARM_FP 0xE
#define __ARM_FP16_ARGS 1
#define __ARM_FP16_FORMAT_IEEE 1
#define __ARM_NEON 1
#define __ARM_NEON_FP 0xE
#define __ARM_NEON__ 1
#define __ARM_PCS_AAPCS64 1
#define __ARM_SIZEOF_MINIMAL_ENUM 4
#define __ARM_SIZEOF_WCHAR_T 4
#define __aarch64__ 1
#define __arm64 1
#define __arm64__ 1
$ 
~~~

> Anyway, We have workarounds for some issues already.
> Could you verify them?

I prefer to solve the configuration inconsistencies first. Applying "fixes" which were found on the basis of an inconsistent configuration is never a good idea.

micomx, 2023-01-16:
Hi @lelegard

Sorry for confusing.

It is different Build Machine and Target Board.
Build Machine is  ubuntu 18 on x86_64.
And Target Board is embeded linux on arm64.
Although Target's cpu is 64bit , but we are on develope 32bit-app.

It is used cross-compile tools(arm_32bit) to build for Target with latest official tsduck(3.32-2983).

Thanks,

micomx, 2023-01-16:
Hi @lelegard 

Here are cxxmacros's output
```

#define __arm__ 1
#define __ARM_32BIT_STATE 1
#define __ARM_ARCH 7
#define __ARM_ARCH_7A__ 1
#define __ARM_ARCH_ISA_ARM 1
#define __ARM_ARCH_ISA_THUMB 2
#define __ARM_ARCH_PROFILE 65
#define __ARM_ASM_SYNTAX_UNIFIED__ 1
#define __ARM_EABI__ 1
#define __ARMEL__ 1
#define __ARM_FEATURE_CLZ 1
#define __ARM_FEATURE_COPROC 15
#define __ARM_FEATURE_DSP 1
#define __ARM_FEATURE_LDREX 15
#define __ARM_FEATURE_QBIT 1
#define __ARM_FEATURE_SAT 1
#define __ARM_FEATURE_SIMD32 1
#define __ARM_FEATURE_UNALIGNED 1
#define __ARM_FP 12
#define __ARM_NEON 1
#define __ARM_NEON__ 1
#define __ARM_NEON_FP 4
#define __ARM_PCS 1
#define __ARM_SIZEOF_MINIMAL_ENUM 4
#define __ARM_SIZEOF_WCHAR_T 4
```

Thanks,

lelegard, 2023-01-16:
> Here are cxxmacros's output

The output describes the native compiler on the host, here a 32-bit Arm.

lelegard, 2023-01-18:
I have made a couple of modifications which should avoid the alignment issue on Arm64 when using an Arm32 binary.

Anyway, using an Arm32 binary on an Arm64 platform is clearly suboptimal. You should compile for the target architecture.

micomx, 2023-01-19:
Wow,It is great.
Thanks for your supports.

closed 2023-06-25 by github-actions[bot]